open62541/open62541 #1753: On connect : info/network	Receiving ACK message failed

danolo, 2018-04-24:
!ATTENTION!
Please read the following page carefully and provide us with all the
information requested:
https://github.com/open62541/open62541/wiki/Writing-Good-Issue-Reports

Use Github Markdown to format your text:
https://help.github.com/articles/basic-writing-and-formatting-syntax/

Fill out the sections and checklist below (add text at the end of each line).

Remove this text between the ATTENTION marks after reading it.
!ATTENTION!
--------------------------------------------------------------------------------

Description
===========

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-04-24:
Thanks for the issue report.

Since we provide an issue template, you should also try to fill that out. It is important for us to help you and queries some important info.

Especially you are missing the version which you used to test this. For the next time please use the template.
Additionally (as mentioned in the Google Group) you should set the log level to debug, to show us more info (`UA_LOGLEVEL=100`)

Your wireshark output does not show the most important part, the OPC UA package content. We would need that to further analyze the issue. Frame, ETH and IP is probably not relevant here.

danolo, 2018-04-26:
I could not find the issue template on the website.
Anyhow, I use the open62541 branch 0.3 version you indicated me.
The output with debug level 100 is 
[2018-04-26 08:55:58.165 (UTC+0200)] debug/network	Sent HEL message
[2018-04-26 08:55:58.178 (UTC+0200)] info/network	Receiving ACK message failed

the wireshark output of the transmission protocol (OpcUA) is like on the figure.
[image]

danolo, 2018-04-27:
I just reopen the issue...Anyone can help with this ?

jpfr, 2018-04-27:
This looks like a problem at the lower network levels.
Your connection is being reset before even the first packet is returned.

Maybe you device is at maximum capacity.
Can you connect with other clients? E.g. the UAExpert.

danolo, 2018-04-27:
I installed and tried uaexpert with no success as well...
Lets forget about it, it is too tricky for me.
Thanks anyway,

closed 2018-04-28 by jpfr