open62541/open62541 #1758: Add possibility to select server profile in CMake

Pro, 2018-04-26:
OPC UA supports different server and client profiles:

https://opcfoundation-onlineapplications.org/profilereporting/

Server:
* Nano Embedded Device
* Micro Embedded Device
* Embedded
* Standard
* Global Discovery Server
* Global Discovery and Certificate Mgmt

Client:
* Standard UA Client
* Global Cert. Mgmt Client

It would be a nice addition, if one can select during the build process (i.e. with CMake) which profile should be supported, and all other features are disabled (removed from the code).

With this concept we can reduce the binary size even more.

Pro, 2018-04-26:
There was already a short discussion about this in 2014: https://github.com/open62541/open62541/issues/101#issuecomment-48586643

Also related:
#1757

jpfr, 2018-04-26:
See also CMake dependent options to encode the relations between the options:
https://cmake.org/cmake/help/v3.0/module/CMakeDependentOption.html