open62541/open62541 #1760: Macros MIN and MAX shoudl be UA_MIN and UA_MAX to avoid clashing

bazza2jw, 2018-04-29:
Description
===========
open62541 includes macro definitions for MAX and MIN. This causes a clash with various other packages.
For consistency the macros should be called UA_MIN and UA_MAX.

Background Information / Reproduction Steps
===========================================
None needed

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-04-29:
Full ACK.

Can you create a PR?

HolgerJeromin, 2018-05-07:
This issue should be closed

PR #1772 (2018-05-07): replace MIN/MAX by UA_MIN/UA_MAX
fix #1760
closed 2018-05-07 by jpfr