open62541/open62541 #1761: pragmas to suppress warning when compiling with C++

bazza2jw, 2018-04-29:
Description
===========
When building with C++ there are numerous warnings given. These warnings should be disabled for GCC and MSVC.

eg:
#pragma GCC diagnostic ignored "-Wunused-parameter"

is needed in both h and c file.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-05-17:
We do build for C++ for every commit.
And the build stops at the first warning.

Maybe we should not suppress the warnings but rather fix them.

Which compiler flags should we add in Travis / appveyor to see the C++ warnings?

jpfr, 2018-05-17:
Aah. We build for C++ only on Windows. Not on travis for Linux...

teccheck, 2025-04-16:
With more recent versions, the CI also builds on Linux

closed 2025-04-16 by teccheck