open62541/open62541 #1769: UA_THREAD_LOCAL still used in variable

cabralfortiss, 2018-05-04:
On branch 0.3, in commit d783f983c7a7db9bbd7ef52a48732d3aa40ebaf0 all uses of UA_THREAD_LOCAL were removed,  but there's still one in https://github.com/open62541/open62541/blob/0.3/src/ua_types.c#L74.

I noticed this because in commit 432cc0c85e0e5cee48aa0936de4e56967a12a3c0, the configuration of VxWorks (WRS_KERNEL) was changed related to UA_THREAD_LOCAL .

If UA_THREAD_LOCAL is meant to stay, then I'd like to add again the DEFINE that made VxWorks define UA_THREAD_LOCAL  as empty as it was before.

mlang-de, 2018-06-13:
In the new release v0.3-rc2 the define is still available.

``` C
/* Thread-Local Storage
 * --------------------
 * Thread-local storage is not required by the main library functionality. It is
 * only used for some testing strategies. ``UA_THREAD_LOCAL`` is empty if the
 * feature is not available. */

#ifdef WHICH_DEFINE_COULD_BE_USED_HERE?
#if defined(__STDC_VERSION__) && __STDC_VERSION__ >= 201112L
# define UA_THREAD_LOCAL _Thread_local /* C11 */
#elif defined(__cplusplus) && __cplusplus > 199711L
# define UA_THREAD_LOCAL thread_local /* C++11 */
#elif defined(__GNUC__)
# define UA_THREAD_LOCAL __thread /* GNU extension */
#elif defined(_MSC_VER)
# define UA_THREAD_LOCAL __declspec(thread) /* MSVC extension */
#else
# define UA_THREAD_LOCAL
#endif
#else
# define UA_THREAD_LOCAL
#endif
```

> It is only used for some testing strategies.

Could the define UA_ENABLE_MULTITHREADING be a solution or which certain TESTING defines can be used for this?

cabralfortiss, 2018-06-14:
I supposse UA_ENABLE_MULTITHREADING  is the right choice. This was done and undone some times in the past.

These are the records I found:
1- September 16th 2015: UA_THREAD_LOCAL is created behind UA_ENABLE_MULTITHREADING 
2- June 24th 2016: Commit a5c41a4c1455c9ecf170fb62dfddcab52b3d4b33 took UA_ENABLE_MULTITHREADING out 
3- September 12th 2016: Issue #792 reported about the missing UA_ENABLE_MULTITHREADING 
4-  September 12th 2016: Issue fixed in 62275cf70a8f4ccbae7bfb6184066a35b19d4a9e
5- May 27th 2017: UA_ENABLE_MULTITHREADING  is again took out at aa717f20e2644ababd501f42712aa03aed2b93c6
6- August 4th 2017: Issue #1136 reporting the same problem
7- August 4th 2017: Issue fixed in 03c347408c17c130a53de08de7e337a8a923f13a
8- October 27th 2017: UA_ENABLE_MULTITHREADING  is again took out at 86d9030a67aceb495a6cff412ffd038575e523f1
9- May the 4th 2018(be with you): This Issue was reported

So can we decide once and for all how this should be done? @jpfr @Pro

cabralfortiss, 2018-06-14:
Oh, and also https://github.com/open62541/open62541/commit/432cc0c85e0e5cee48aa0936de4e56967a12a3c0#diff-415ba0f4cd9956a1addc6f24c1dc269cL30 took the special case for VXWORKS which should come back too

cabralfortiss, 2018-06-27:
Thaaaank U!

PR #1877 (2018-06-26): Remove all usages of UA_THREAD_LOCAL
@jpfr do you see a better way to handle `static pcg32_random_t UA_rng = PCG32_INITIALIZER;` not being threadsafe?

Fixes #1769
commit 0f51f666 (2018-06-26): Remove all usages of UA_THREAD_LOCAL. Fixes #1769
closed 2018-06-27 by cabralfortiss