open62541/open62541 #1770: Warning after disconnect UA_Client_runAsync()

Aleadinglight, 2018-05-04:
Here I have a simple UA_Client with Subscription:

```
UA_ClientConfig clientConfig = UA_ClientConfig_default; 
UA_Client *client = UA_Client_new(clientConfig);
ConnectToServer(client);
UA_Client_runAsync(client, 1000);

UA_Client_disconnect(client);
UA_Client_delete(client); 
```

Inside ConnectToServer is just simply create subscription and add monitor items, following the example in **tutorial_client_events.c** of open62541 github.

The problem is that when the program exit, it leaved this warning:

```
[2018-05-05 01:13:07.351 (UTC+0300)] info/userland      Create subscription succeeded, id 2739054811
[2018-05-05 01:13:07.352 (UTC+0300)] info/userland      Monitoring 'ns=3,s=AirConditioner_1.Humidity', id 2739054811
The Answer has changed!
[2018-05-05 01:13:07.854 (UTC+0300)] info/userland      60.000000

[2018-05-05 01:13:08.354 (UTC+0300)] info/client        Received a ServiceFault response
[2018-05-05 01:13:08.355 (UTC+0300)] warn/client        Received Publish Response with code BadSessionClosed
[2018-05-05 01:13:08.355 (UTC+0300)] info/client        Received a ServiceFault response
[2018-05-05 01:13:08.355 (UTC+0300)] warn/client        Received Publish Response with code BadSessionClosed
[2018-05-05 01:13:08.356 (UTC+0300)] info/client        Received a ServiceFault response
[2018-05-05 01:13:08.356 (UTC+0300)] warn/client        Received Publish Response with code BadSessionClosed
[2018-05-05 01:13:08.356 (UTC+0300)] info/client        Received a ServiceFault response
[2018-05-05 01:13:08.356 (UTC+0300)] warn/client        Received Publish Response with code BadSessionClosed
[2018-05-05 01:13:08.356 (UTC+0300)] info/client        Received a ServiceFault response
[2018-05-05 01:13:08.356 (UTC+0300)] warn/client        Received Publish Response with code BadSessionClosed
[2018-05-05 01:13:08.356 (UTC+0300)] info/client        Received a ServiceFault response
[2018-05-05 01:13:08.356 (UTC+0300)] warn/client        Received Publish Response with code BadSessionClosed
[2018-05-05 01:13:08.356 (UTC+0300)] info/client        Received a ServiceFault response
[2018-05-05 01:13:08.357 (UTC+0300)] warn/client        Received Publish Response with code BadSessionClosed
[2018-05-05 01:13:08.357 (UTC+0300)] info/client        Received a ServiceFault response
[2018-05-05 01:13:08.357 (UTC+0300)] warn/client        Received Publish Response with code BadSessionClosed
[2018-05-05 01:13:08.357 (UTC+0300)] info/client        Received a ServiceFault response
[2018-05-05 01:13:08.357 (UTC+0300)] warn/client        Received Publish Response with code BadSessionClosed
[2018-05-05 01:13:08.357 (UTC+0300)] info/client        Received a ServiceFault response
[2018-05-05 01:13:08.357 (UTC+0300)] warn/client        Received Publish Response with code BadSessionClosed
The client is disconected!
```
Can anyone give me some detail on what could have gone wrong? Since the Subscription and the Monitor item was added successfully, could there be some problem with the background process of `UA_Client_runAsync(client, 1000);` ?

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.3
 - [ ] Other OPC UA SDKs used (client or server): UaCppServer
 - [ ] Operating system: Windows

StalderT, 2018-05-07:
The behavior is normal. By default, 10 Publish Request are sent to the server and remain waiting in the server. When UA_Client_disconnect is called, the session is closed and somes servers send a Publish Response with service fault BadSessionClosed for each pending Publish Request.

I made a PR (#1771) to remove the warning ```Received Publish Response with code BadSessionClosed```. The info ```Received a ServiceFault response``` stays.

Aleadinglight, 2018-05-07:
Thanks for the response, may I ask you a little bit more on this? 

1. The `UA_Client_runAsync(client, 1000);` has a timeout of 1 second, how do you determine that 10 Requests was sent?
2. Should I better try to avoid this warning somehow?

StalderT, 2018-05-07:
>The UA_Client_runAsync(client, 1000); has a timeout of 1 second

UA_Client_runAsync will block 1 second (if no incoming response)

For subscription : timeout is treated by publishingInterval and maxKeepAliveCount
For synchronous requests : timeout is treated by config.timeout (ms)
For asynchronous requests : currently no timeout. Should be fixed if #1728 will be merged. In this case timeout will be treated by config.timeout (ms)

> how do you determine that 10 Requests was sent

Here : https://github.com/open62541/open62541/blob/d88df4281dde5744db06d6c738d0fcc6b73523e8/plugins/ua_config_default.c#L653

You can change this setting : 

```
    UA_ClientConfig config = UA_ClientConfig_default;
    config.outStandingPublishRequests = 5;
    UA_Client *client = UA_Client_new(config);
```

>Should I better try to avoid this warning somehow?

```Received Publish Response with code BadSessionClosed``` should be removed by #1771
```Received a ServiceFault response``` is an info not a warning. Infos can be hide by setting a UA_LOGLEVEL greater than 300.

jpfr, 2018-05-09:
Fixed in #1771. Thanks to @StalderT.
Please reopen if the issue persists.

PR #1771 (2018-05-09): CLIENT : remove BadSessionClosed when disconnect
fix #1770
closed 2018-05-09 by jpfr