open62541/open62541 #1773: Handling of ExtensionObject with UA_ENABLE_FULL_NS0

basyskom-jvoe, 2018-05-07:
If open62541 is built with UA_ENABLE_FULL_NS0 enabled, the additional types from the schema files (Range, EUInformation, AxisInformation, ComplexNumber, DoubleComplexNumber, XVType, etc...) are also generated.

If an array of Range is read from a server, the resulting UA_Variant contains an array of UA_ExtensionObject with decoded Range values inside.
If the value attribute contains a single Range, an UA_Variant with type index 135 (Range) containing a decoded range is returned.

The same happens for the other additional types mentioned above.

https://open62541.org/doc/current/types.html#variant states: "The standard mandates that variants contain built-in data types only. If the value is not of a builtin type, it is wrapped into an ExtensionObject."

The following data type description is generated for UA_Range
>>
    UA_TYPENAME("Range") /* .typeName */
    {0, UA_NODEIDTYPE_NUMERIC, {884}}, /* .typeId */
    sizeof(UA_Range), /* .memSize */
    UA_TYPES_RANGE, /* .typeIndex */
    2, /* .membersSize */
    false, /* .builtin */
    true, /* .pointerFree */
    true && UA_BINARY_OVERLAYABLE_FLOAT && UA_BINARY_OVERLAYABLE_FLOAT && offsetof(UA_Range, high) == (offsetof(UA_Range, low) + sizeof(UA_Double)), /* .overlayable */
    886, /* .binaryEncodingId */
    Range_members /* .members */

which marks Range as a non-builtin type.

From how I understand the specification, the scalar case should also return an UA_ExtensionObject containing a decoded Range.
Is the behavior described above expected or is this a bug?

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3
 - [ ] Other OPC UA SDKs used (client or server): 
 - [x] Operating system: Ubuntu 16.04
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Infinity95, 2018-05-09:
As far as I understood the standard, Range objects should indeed be wrapped in ExtensionObjects.
The only objects that can be encoded in Variants are specified in the standard in Part 6 (v1.03) in Table 1.

jpfr, 2018-05-09:
Hello @basyskom-jvoe ,
Hello @Infinity95 ,

Yes, For transport over the wire, any non-builtin type inside a variant is boxed inside an ExtensionObject.
This enabled the client to parse the message even if he doesn't understand the content type of the variant.

In open62541, you can put anything inside a Variant. It is wrapped and unwrapped into an ExtensionObject for transport in the background. So this "boxing" is transparently hidden from users.

basyskom-jvoe, 2018-05-09:
Hello @jpfr 

Shouldn't the behavior then be the same for scalar and array (single Range, Array of Range) instead of single Range, Array of ExtensionObject(Range)?

jpfr, 2018-05-09:
Um yeah. Uniformity is definitely a goal.

That being said, we would have to take two passes over the array.
One to check that all elements of the array are of the same type (inside their ExtensionObject).
And a second pass to decode them.

Feel free to propose something.

closed 2024-07-02 by basyskom-jvoe