open62541/open62541 #1782: UA_ServerConfig_new_basic256sha256 issuu

jhj0955, 2018-05-16:
1. I'm running the server_basic256sha256 example with self signed certificates from /tools/cert/ and try to connect from a UA Expert

2. However, the connectable communication channels are as follows : 
     None - None (uatcp-uasc-uabinary)
     Basic128Rsa15- Sign (uatcp-uasc-uabinary)
     Basic128Rsa15- Sign & Encrypt (uatcp-uasc-uabinary)
     Basic256Rsa256- Sign (uatcp-uasc-uabinary)
     Basic256Rsa256- Sign  & Encrypt (uatcp-uasc-uabinary)

3. why is None-secure channel here? What should I do to remove that channel(none)?

Checklist
=========
Please provide the following information:

 - [ o] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

schroeder-, 2018-05-16:
The plugin function UA_ServerConfig_new_basic256sha256 creates a None-secure channel. 
If you want another behavior you shoul implement your own function instead. You can use the UA_ServerConfig_new_basic256sha256 in ua_config_default.c as base.

jpfr, 2018-05-17:
Exactly. The default configuration adds an additional _none endpoint.
Modify the default configuration if required.

We might need a more elaborate configuration option via some configuration file parsing.
Having users modify the self-contained configuration setup is better than blowing up the API at this point.

closed 2018-05-17 by jpfr