open62541/open62541 #1784: Slow addition of many variables

cbelyaev, 2018-05-17:
I'm running the slightly modified `tutorial_server_variable.c` with master:
<pre>
#include <signal.h>
#include <stdio.h>
#include "open62541.h"

static void
addVariable(UA_Server *server, UA_Int32 id) {
    /* Define the attribute of the myInteger variable node */
    UA_VariableAttributes attr = UA_VariableAttributes_default;
    UA_Int32 my = id;
    UA_Variant_setScalar(&attr.value, &my, &UA_TYPES[UA_TYPES_INT32]);
    char *name = malloc(11);
    sprintf(name, "%07d", id);
    attr.displayName = UA_LOCALIZEDTEXT("en-US", name);
    attr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
    attr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;

    /* Add the variable node to the information model */
    UA_NodeId myNodeId = UA_NODEID_NUMERIC(1, id);
    UA_QualifiedName myName = UA_QUALIFIEDNAME(1, name);
    UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
    UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
    UA_Server_addVariableNode(server, myNodeId, parentNodeId,
                              parentReferenceNodeId, myName,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), attr, NULL, NULL);
}

/** It follows the main server code, making use of the above definitions. */

UA_Boolean running = true;
static void stopHandler(int sign) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "received ctrl-c");
    running = false;
}

int main(void) {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);

    for(int i = 1; i <= 30000; i++)
        addVariable(server, i);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);
    return (int)retval;
}
</pre>

Creation of 30 000 variables takes about 14 seconds on quite powerfull computer.
On ARM module it may take hours.

`grpof` says:
<pre>
Flat profile:

Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total           
 time   seconds   seconds    calls  Ts/call  Ts/call  name    
 30.63      4.35     4.35                             UA_ExpandedNodeId_equal
 27.54      8.26     3.91                             UA_Node_addReference
 18.94     10.95     2.69                             UA_String_equal
 17.96     13.50     2.55                             UA_NodeId_equal
  2.32     13.83     0.33                             UA_NodeId_isNull
  1.20     14.00     0.17                             UA_String_fromChars
  0.14     14.02     0.02                             UA_Server_editNode
  0.14     14.04     0.02                             deleteMembers_noInit
  0.11     14.06     0.02                             copy_noInit
  0.07     14.07     0.01                             AddNode_typeCheckAddRefs
  0.07     14.08     0.01                             ExpandedNodeId_deleteMembers
  0.07     14.09     0.01                             String_copy
  0.07     14.10     0.01                             UA_Array_copy
  0.07     14.11     0.01                             UA_LocalizedText_deleteMembers
  0.07     14.12     0.01                             UA_NodeId_hash
  0.07     14.13     0.01                             UA_NodeMap_getNode
  0.07     14.14     0.01                             UA_Node_setAttributes
  0.07     14.15     0.01                             addOneWayReference
  0.07     14.16     0.01                             clearSlot
  0.07     14.17     0.01                             compatibleValue
  0.07     14.18     0.01                             getNodeType
  0.07     14.19     0.01                             isNodeInTreeNoCircular
  0.07     14.20     0.01                             readValueAttribute
  0.04     14.20     0.01                             copyGuid
  0.00     14.20     0.00   120000     0.00     0.00  UA_NODEID_NUMERIC
  0.00     14.20     0.00    90000     0.00     0.00  UA_STRING
  0.00     14.20     0.00    30000     0.00     0.00  UA_LOCALIZEDTEXT
  0.00     14.20     0.00    30000     0.00     0.00  UA_QUALIFIEDNAME
  0.00     14.20     0.00    30000     0.00     0.00  UA_Server_addVariableNode
  0.00     14.20     0.00    30000     0.00     0.00  addVariable
  0.00     14.20     0.00        1     0.00     0.00  UA_LOG_INFO
  0.00     14.20     0.00        1     0.00     0.00  UA_ServerConfig_new_default
</pre>

As far as I understand, the problem is in `UA_Node_addReference`, which iterates all `references` array.

Besides the new node store plugin, are there any options?

schroeder-, 2018-05-17:
I think it's a common Problem, my commercial SDK I also use, is 10 times slower when creating the same Nodes then open62541.

jpfr, 2019-11-21:
This was resolved a few weeks ago.
We had a quadratic runtime in the number of references...

closed 2019-11-21 by jpfr