open62541/open62541 #1789: UA_Server_addPeriodicServerRegisterCallback crash in example code (server_register.c)

leoleung2000, 2018-05-18:
Description
===========
Good morning. 
When I use the example code in server_register.c 
I found it crash at function **UA_Server_addPeriodicServerRegisterCallback**
After reviewing by debugger , i found the variable **periodicCallbackId** checking will always return true even the input is NULL. 

Background Information / Reproduction Steps
===========================================
1. Use Visual Studio 2015 , Create new empty project
2. Include open62541 related files and server_register.c 
3. Add breakpoint at open62541.c:line 32094
[image]

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 0.3-rc1 (32bit Version)
 - [x] Other OPC UA SDKs used (client or server): No, just this library
 - [x] Operating system: Windows 10 Pro
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-05-18:
Thanks for your report.

So you are saying that the call to
```
    UA_StatusCode retval =
        UA_Server_addPeriodicServerRegisterCallback(server, DISCOVERY_SERVER_ENDPOINT,
                                                    10 * 60 * 1000, 500, NULL);
```
for
```
UA_Server_addPeriodicServerRegisterCallback(UA_Server *server,
                                            const char* discoveryServerUrl,
                                            UA_UInt32 intervalMs,
                                            UA_UInt32 delayFirstRegisterMs,
                                            UA_UInt64 *periodicCallbackId)
```

fails, because the last parameter `periodicCallbackId` is set to `NULL`, but the check here is not detecting the NULL:
```
    if(periodicCallbackId)
        *periodicCallbackId = cb->id;
```

Correct? If yes, than that's quite strange...

Can you do some additional tests:

Change `if(periodicCallbackId)` to `if(periodicCallbackId != NULL)`. Does that solve the issue?

Also, write a simple method, which gets a pointer to an UA_UInt64 and try to create a MCVE (https://stackoverflow.com/help/mcve) which reproduces the error (maybe even without any reference to open62541).

leoleung2000, 2018-05-29:
The description is correct. Thank you! 
For testing
1. Change checking to `if(periodicCallbackId != NULL)` , it still have the same problem.
2. Simple method gets pointer to an UA_UInt64, the pointer's value is correct 
```
UA_UInt64 hello = 1;
UA_UInt64* ptr_hello = &hello;
```
[image]
3. The ```periodicCallbackId``` remain the same value no matter i put ```NULL``` or ```ptr_hello``` (which value is 0x012ffce4) to function ```UA_Server_addPeriodicServerRegisterCallback```

Here is my sample Code which can reproduce the issue 
```
    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_String_deleteMembers(&config->applicationDescription.applicationUri);
    config->applicationDescription.applicationUri =
        UA_String_fromChars("urn:open62541.example.server_register");
    config->mdnsServerName = UA_String_fromChars("Sample Server");
    // See http://www.opcfoundation.org/UA/schemas/1.03/ServerCapabilities.csv
	config->serverCapabilitiesSize = 1;
    UA_String caps = UA_String_fromChars("LDS");
    config->serverCapabilities = &caps;

    UA_Server *server = UA_Server_new(config);

    UA_Client *clientRegister = UA_Client_new(UA_ClientConfig_default);
    // periodic server register after 10 Minutes, delay first register for 500ms
    UA_StatusCode retval =
        UA_Server_addPeriodicServerRegisterCallback(server, clientRegister, DISCOVERY_SERVER_ENDPOINT,
                                                    1 * 60 * 1000, 500, NULL);
```

leoleung2000, 2018-05-29:
I think i found out the problem. 
In example code, function input parameter as follow 
```
UA_StatusCode retval =
        UA_Server_addPeriodicServerRegisterCallback(server, clientRegister, 
                                                    DISCOVERY_SERVER_ENDPOINT,
                                                    1 * 60 * 1000, 500, NULL);
```
Comparing with @Pro example code 
```
UA_StatusCode retval =
        UA_Server_addPeriodicServerRegisterCallback(server, DISCOVERY_SERVER_ENDPOINT,
                                                    10 * 60 * 1000, 500, NULL);
```
The input parameter getting wrong starting from second parameters.
API not crash by removing ```clientRegister``` variable. 

By the way, can you update the example of server_register.c? 
As i don't know the impact after removing the ```clientRegister``` input parameter.

Pro, 2018-05-29:
Thanks for digging into it. It looks like there's a misunderstanding by mixing different versions of the source and examples.

0.3 branch it is correct to use:
```c
UA_StatusCode retval =
        UA_Server_addPeriodicServerRegisterCallback(server, DISCOVERY_SERVER_ENDPOINT,
                                                    10 * 60 * 1000, 500, NULL);
```

for current master we introduced the additional client parameter to allow to use encryption and authentication:
```c
UA_StatusCode retval =
        UA_Server_addPeriodicServerRegisterCallback(server, clientRegister, 
                                                    DISCOVERY_SERVER_ENDPOINT,
                                                    1 * 60 * 1000, 500, NULL);
```

The corresponding example codes are also correct.

Based on your first post you indicated that you are using 0.3-rc1. In your second post you are using the example code from master. Anyways it should show compilation errors if you are mixing the two.

Can you please do the following:

1. Use Version 0.3-rc1 and the example from 0.3-rc1 and test if you can reproduce
2. Do the same with the master version and master example

Then please give feedback if you can reproduce it on any of these versions.

closed 2018-05-31 by leoleung2000