open62541/open62541 #1790: Default namespace high RAM (heap) usage. How to reduce?

wrc-jasonw, 2018-05-18:
Hello, I am porting the current git HEAD (9283b07) of open62541 to a STM32F407 microcontroller. I am running FreeRTOS. I have roughly 250kb of flash (program) memory and 128kb of free RAM to allocate. My goal is to develop a very simple OPC-UA compliant server device with this stack. The issue making everything fit into memory.

I am relatively new to OPC-UA; forgive me if I use incorrect terminology. I've searched and read some of the other issues but none seemed relevant.

I have found that the baseline open62541 before namespace initialization doesn't consume much memory at all (2kb stack, 6kb heap) - which is a good sign. However when I initialize the default namespace0 provided with the server I have found that the namespace is allocating a very large amount of memory.

Specifically there are six (6) nodes that are allocated 65000 bytes each (380kB total). I would like help understanding why these default nodes are allocated so much memory [1]. In addition I would like to identify unnecessary nodes that could be removed to reduce memory usage [2].

I have attached my auto-generated namespace0: [ua_namespace0.c.txt](https://github.com/open62541/open62541/files/2016747/ua_namespace0.c.txt)
Additionally my default XML schema: [Opc.Ua.NodeSet2.Minimal.xml.txt](https://github.com/open62541/open62541/files/2016851/Opc.Ua.NodeSet2.Minimal.xml.txt)

In the C file:
- Line 1376: allocates 65000 byte to variablenode_ns_0_i_12169_EnumValueType_0_0 (->content.encoded.body)
- Line 1404: allocates 65000 bytes to variablenode_ns_0_i_12169_EnumValueType_1_0
- Line 1432: allocates 65000 bytes to variablenode_ns_0_i_12169_EnumValueType_2_0
- Line 2210: allocates 65000 bytes to variablenode_ns_0_i_11494_Argument_0_0
- Line 2247: allocates 65000 bytes to variablenode_ns_0_i_11494_Argument_1_0
- Line 3179: allocates 65000 bytes to variablenode_ns_0_i_11493_Argument_0_0

In the XML file:
- Line 561 is i=12169, the enumerated type definition.
-- If I read the XML file correctly the enumeration i=12169 only seems to consist of a few dozen bytes worth of strings. Does the compiler really need to allocate exactly 65000 bytes to the three parts of the enumerated value?
- Line 1543 is i=11494, "OutputArguments"
-- Also this datatype/variable looks like it doesn't really need that much memory
- Line 1516 is i=11493, "InputArguments"
-- Similar story.

I looked into the compiler and it looks like the 65000 byte value is hard-coded. Is the 65000 bytes required by the OPC-UA specification? If not is there a way to hand calculate the required memory of a particular node?

Thank You!
-Jason White

Checklist
=========
Please provide the following information:

 - [*] open62541 Version (9283b07):
 - [*] Other OPC UA SDKs used (server): 
 - [*] Operating system:
 - [*] Self-contained code example attached

Stasik0, 2018-05-18:
65k are here

https://github.com/open62541/open62541/blob/f777fe276ab7920f3043a642eaa45e75f9c46b95/tools/nodeset_compiler/backend_open62541_nodes.py#L223

wrc-jasonw, 2018-05-18:
Ok I see it. I suppose I will have to arbitrarily start reducing the amount of memory allocated until things break.

Stasik0, 2018-05-18:
3 lines later:

    # Reallocate the memory by swapping the 65k Bytestring for a new one

So first idea is to use only one 65k buffer.
Second is to malloc and free it.
Third is to calculate length properly.

Stasik0, 2018-05-18:
Idea one and two are probably not working - stack will be freed due to code blocks afaik

Stasik0, 2018-05-18:
The most weird idea is to compile c and run these pieces at build time and fix the sizes this way...

wrc-jasonw, 2018-05-18:
That is what I am doing: changing values in C. What I would like help with is figuring out how to identify the correct size from the debugger in C. I am unfamiliar with the internal data model.

Stasik0, 2018-05-18:
`struct {
  UA_Int64 Value;
  UA_LocalizedText DisplayName;
  UA_LocalizedText Description;
} variablenode_ns_0_i_12169_EnumValueType_0_0_struct;
variablenode_ns_0_i_12169_EnumValueType_0_0_struct.Value = (UA_Int64) 1;
variablenode_ns_0_i_12169_EnumValueType_0_0_struct.DisplayName = UA_LOCALIZEDTEXT("", "Mandatory");
variablenode_ns_0_i_12169_EnumValueType_0_0_struct.Description = UA_LOCALIZEDTEXT("", "The BrowseName must appear in all instances of the type.");
UA_ExtensionObject *variablenode_ns_0_i_12169_EnumValueType_0_0 =  UA_ExtensionObject_new();
variablenode_ns_0_i_12169_EnumValueType_0_0->encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
variablenode_ns_0_i_12169_EnumValueType_0_0->content.encoded.typeId = UA_NODEID_NUMERIC(0, 8251);
//begin - needed only for variablenode_ns_0_i_12169_EnumValueType_0_0_encOffset
UA_ByteString_allocBuffer(&variablenode_ns_0_i_12169_EnumValueType_0_0->content.encoded.body, 65000);
UA_Byte *posvariablenode_ns_0_i_12169_EnumValueType_0_0 = variablenode_ns_0_i_12169_EnumValueType_0_0->content.encoded.body.data;
const UA_Byte *endvariablenode_ns_0_i_12169_EnumValueType_0_0 = &variablenode_ns_0_i_12169_EnumValueType_0_0->content.encoded.body.data[65000];
{
retVal |= UA_encodeBinary(&variablenode_ns_0_i_12169_EnumValueType_0_0_struct.Value, &UA_TYPES[UA_TYPES_INT64], &posvariablenode_ns_0_i_12169_EnumValueType_0_0, &endvariablenode_ns_0_i_12169_EnumValueType_0_0, NULL, NULL);
retVal |= UA_encodeBinary(&variablenode_ns_0_i_12169_EnumValueType_0_0_struct.DisplayName, &UA_TYPES[UA_TYPES_LOCALIZEDTEXT], &posvariablenode_ns_0_i_12169_EnumValueType_0_0, &endvariablenode_ns_0_i_12169_EnumValueType_0_0, NULL, NULL);
retVal |= UA_encodeBinary(&variablenode_ns_0_i_12169_EnumValueType_0_0_struct.Description, &UA_TYPES[UA_TYPES_LOCALIZEDTEXT], &posvariablenode_ns_0_i_12169_EnumValueType_0_0, &endvariablenode_ns_0_i_12169_EnumValueType_0_0, NULL, NULL);
}
//end - needed only for variablenode_ns_0_i_12169_EnumValueType_0_0_encOffset
size_t variablenode_ns_0_i_12169_EnumValueType_0_0_encOffset = (uintptr_t)(posvariablenode_ns_0_i_12169_EnumValueType_0_0-variablenode_ns_0_i_12169_EnumValueType_0_0->content.encoded.body.data);
variablenode_ns_0_i_12169_EnumValueType_0_0->content.encoded.body.length = variablenode_ns_0_i_12169_EnumValueType_0_0_encOffset;
UA_Byte *variablenode_ns_0_i_12169_EnumValueType_0_0_newBody = (UA_Byte *) UA_malloc(variablenode_ns_0_i_12169_EnumValueType_0_0_encOffset);
memcpy(variablenode_ns_0_i_12169_EnumValueType_0_0_newBody, variablenode_ns_0_i_12169_EnumValueType_0_0->content.encoded.body.data, variablenode_ns_0_i_12169_EnumValueType_0_0_encOffset);
UA_Byte *variablenode_ns_0_i_12169_EnumValueType_0_0_oldBody = variablenode_ns_0_i_12169_EnumValueType_0_0->content.encoded.body.data;
variablenode_ns_0_i_12169_EnumValueType_0_0->content.encoded.body.data = variablenode_ns_0_i_12169_EnumValueType_0_0_newBody;
UA_free(variablenode_ns_0_i_12169_EnumValueType_0_0_oldBody);
`

here `variablenode_ns_0_i_12169_EnumValueType_0_0_encOffset` is the size, actually you just need to hardcode it and then get rid of

Stasik0, 2018-05-18:
i think 65k is really some upper value for a network buffer s.t. it is guaranteed to fit. Best way would be to calculate these values at compile time and not at run time... @jpfr, your thoughts on that?

wrc-jasonw, 2018-05-18:
Just an FYI: I just got open62541 to run on my microcontroller, I am clocking in at almost exactly 112kb of RAM usage (heap+stack) sitting idle with no connections. Looks like things are going to be tight.

Is everything in the default namespace required in order to pass the conformance test?

Stasik0, 2018-05-18:
i think you need the minimal one to pass CTT. There are some issue topics on how to reduce "compiled size". For RAM usage maybe you could cut off "attr.description" to get 1k less or something... 

Probably we will need a flash-backended nodestore plugin for all the static stuff...

Pro, 2018-05-19:
@wrc-jasonw did you already see this PR:
https://github.com/open62541/open62541/pull/1595

It will be integrated into master soon and adds support for different architectures, one of them is also FreeRTOS. It would be nice if you can test that an give feedback.

When implementing that PR I ran in the same issue as you mentioned, that these 65000 are too big. As a workaround I introduced an additional CMakeSetting to overwrite that hardcoded value (not in master yet):

https://github.com/open62541/open62541/commit/d8d36f2902fea69e755777052d8ae08bc1db97f2

Unfortunately this big array is currently needed for an encoding hack and will be fixed when we implement a workaround as mentioned here:
https://github.com/open62541/open62541/pull/1609#issuecomment-367124809

wrc-jasonw, 2018-05-21:
Ok. Thanks everybody!

I may not have time to evaluate #1595 but I will try to if I can.

closed 2018-05-21 by wrc-jasonw