open62541/open62541 #1795: error C2440: 'initializing': cannot convert from 'UA_ExtensionObject' to 'UA_ExtensionObject *'

diverger, 2018-05-23:
Description
===========
After build the model with UaModeler, and convert it to .c file, when compiled with VS2017, it  complains hundreds of: 

> error C2440: 'initializing': cannot convert from 'UA_ExtensionObject' to 'UA_ExtensionObject *'

The problem code is as such:

`UA_ExtensionObject *variablenode_ns_1_i_6078_Range_0_0 =  UA_ExtensionObject_new();
.....
UA_ExtensionObject *variablenode_ns_1_i_6078_variant_DataContents = *variablenode_ns_1_i_6078_Range_0_0;`

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.3-rc1
 - [ ] Other OPC UA SDKs used (client or server): UaModeler
 - [ ] Operating system: Win 7, 64bit
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-05-23:
Can you post the command line args you used for the nodeset compiler to convert the XML into c?

diverger, 2018-05-23:
> python %OPEN62541%\tools\nodeset_compiler\nodeset_compiler.py --types-array=UA_TYPES --existing %OPEN62541%/deps/ua-nodeset/Schema/Opc.Ua.NodeSet2.xml --xml uasysmonitor.xml UASysMonitor

alfred-r, 2018-06-06:
The extra '*' comes from a bug in backend_open62541_nodes.py

I corrected line 370 in file 'tools/nodeset_compiler/backend_open62541_nodes.py'
from this:
`370: generateNodeValueCode(node.value[0], instanceName, max_string_length=max_string_length) + ";")`

to this:
`370: generateNodeValueCode(node.value[0], instanceName, asIndirect=True, max_string_length=max_string_length) + ";")`

add this "**asIndirect=True,**" to the call and the unnecessary '*' is gone

But i still run into some errors when i convert the CNC companion XML file (Opc.Ua.CNC.NodeSet.xml) to C code

teccheck, 2025-04-16:
It looks to me like this is solved. Also, the mentioned models compile now

closed 2025-04-16 by teccheck