open62541/open62541 #1799: Copying and deletion of nodes with Events

jpfr, 2018-05-23:
We just got event support on master in #1739.
I ran into a couple of problems when nodes were modified and deleted that had events on them.

Probably we need to
- Handle the monitoredItemQueue of the node during the DeleteNodes service
- Ensure that the monitoredItemQueue is kept in a copy/edit/replace cycle when ``UA_ENABLE_IMMUTABLE_NODES`` is activated

These are also good cases for unit tests.

Pro, 2018-05-24:
@aDogCalledSpot can you have a look at it

aDogCalledSpot, 2018-05-25:
Could you be a bit more specific? I've created nodes, triggered some events onto them and attempted to delete them and nothing out of the ordinary is happening.

jpfr, 2018-05-25:
Crash for the events tutorial:

- Connect with UAExpert
- Add `Objects` and `Server` to an Events view
- Trigger the event a couple of times
- Disconnect the client

```
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
#1  0x00007ffff755119a in __GI_abort () at abort.c:89
#2  0x00007ffff758e300 in __libc_message (do_abort=do_abort@entry=2, fmt=fmt@entry=0x7ffff7688d30 "*** Error in `%s': %s: 0x%s ***\n") at ../sysdeps/posix/libc_fatal.c:175
#3  0x00007ffff759495e in malloc_printerr (action=3, str=0x7ffff7688da8 "double free or corruption (!prev)", ptr=<optimized out>, ar_ptr=<optimized out>) at malloc.c:5079
#4  0x00007ffff75951be in _int_free (av=0x7ffff78b9b00 <main_arena>, p=0x5555560b1190, have_lock=0) at malloc.c:3875
#5  0x0000555555a35bc3 in freeCallback (server=0x555555d4f060, data=0x5555560b11a0) at /home/jpfr/software/open62541/src/server/ua_server_worker.c:179
#6  0x0000555555a35cf0 in UA_Server_cleanupDelayedCallbacks (server=0x555555d4f060) at /home/jpfr/software/open62541/src/server/ua_server_worker.c:215
#7  0x0000555555a35f3b in UA_Server_run_iterate (server=0x555555d4f060, waitInternal=true) at /home/jpfr/software/open62541/src/server/ua_server_worker.c:381
#8  0x0000555555a36040 in UA_Server_run (server=0x555555d4f060, running=0x555555d471b0 <running>) at /home/jpfr/software/open62541/src/server/ua_server_worker.c:463
#9  0x0000555555563708 in main () at /home/jpfr/software/open62541/examples/tutorial_server_events.c:172
```

jpfr, 2018-05-29:
The mentioned issue is fixed in #1815.

closed 2021-03-16 by jpfr