open62541/open62541 #1806: RAM Usage Reduction: Autogenerated UA_DataTypeMember's should be declared const

wrc-jasonw, 2018-05-25:
Background: I am on a microcontroller running FreeRTOS. I am currently in the process of trimming RAM usage of my implementation. I figured I may as well share my findings so that others may benefit from the work.

In ua_types_generated.c and ua_transport_generated.c all variables of type "static UA_DataTypeMember" should be declared "static const UA_DataTypeMember" in order to reduce memory usage.

This yielded a "free" savings of 3172 bytes of RAM on the STM32F407.

As I understand it the datatype arrays of type "UA_DataTypeMember[...]" are constant after the server has started. Therefore they can be declared constant which allows the data to be stored in (relatively cheap) program flash rather than in (precious) RAM.

jpfr, 2018-05-25:
That's a good contribution!
Can you make a pull request from your improvements?

wrc-jasonw, 2018-05-25:
Sorry, for the time being I have only modified the auto-generated C files. The real culprit are the python auto-generator scripts.

Stasik0, 2018-05-25:
@jpfr: cannot same be done for nodes in the nodestore? Since they are immutable anyhow?

wrc-jasonw, 2018-05-25:
I would be very grateful if that is the case.

Currently (with my very limited understanding of how things work) each node seems to take about 560 bytes. My current task is identifying variables, objects, and datatype that are optional so that I can remove them. so far I have identified 55+ nodes in the default XML that I can remove which should yield a savings of ~31K of RAM.

If nodes can be moved into flash then RAM usage would be reduced ever further.

wrc-jasonw, 2018-05-29:
Thank you again everybody! This has been very helpful.

Results of my review of the default namespace0:

The XML file provided by the OPC-UA foundation contains many optional objects and datatypes that can be removed to save RAM,

In particular the OPC Binary->Opc.Ua DatatypeDictionary contains ~15kB of XML. In version 1.04 of OPC-UA DatatypeDictionary's were removed from the standard. Also OperationLimits, OperationalLimitsType, ServerDiagnosticSummery, ServerDiagnosticSummeryType and all their children nodes are all optional. The set of standard types like Byte, Float, Image, XmlElement also all can be removed if they are not used by the application.

By doing the above I observed an overall reduction of RAM usage of ~45kB.

closed 2018-05-29 by wrc-jasonw