open62541/open62541 #1820: Restrict addition of same browse name child

Pro, 2018-05-30:
The specification states:

> BrowsePath
> All targets of forward hierarchical References from a TypeDefinitionNode shall have a BrowseName that is unique within the TypeDefinitionNode. The same restriction applies to the targets of hierarchical References in forward direction from any InstanceDeclaration. This means that any InstanceDeclaration within the InstanceDeclarationHierarchy can be uniquely identified by a sequence of BrowseNames. This sequence of BrowseNames is called a BrowsePath.

Currently it is possible to add a child to a node which has the same browse path as an already existing child.
Should or do we want to add the corresponding check within AddNode or is this up to the developer?

That is important for the `TranslateBrowsePathToNodeId` call which should return the uniquely identified node.

See also:
https://stackoverflow.com/questions/34805698/opc-ua-unique-id

jpfr, 2018-05-30:
It is TranslateBrowsePath**s**ToNodeId**s**.

Yes. We can easily check one-level forward references during instantiation.

kevinherron, 2018-06-01:
This restriction only applies to type definitions and the instance declaration hierarchy, not to nodes in general. Also, as @jpfr pointed out, the `TranslateBrowsePathsToNodeIds` service explicitly mentions and handles the case where a browse path points to multiple `NodeId`s.