open62541/open62541 #1824: UA_Server_triggerEvent: "free(): invalid size, Aborted (core dumped)"

NIPE-SYSTEMS, 2018-05-31:
Description
===========

Triggering an event results in a core-dump. Valgrind reports that the sizes of `malloc` and `free` differ. (Don't know how to interpret that.) (see also https://github.com/open62541/open62541/pull/1739)

Background Information / Reproduction Steps
===========================================

```cpp
UA_NodeId nodeIdEvent;
if(UA_Server_createEvent(m_server, UA_NODEID_NUMERIC(m_namespace, 1005), &nodeIdEvent) != UA_STATUSCODE_GOOD) {
    UA_LOG_WARNING(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Failed to create event for elements addition.");
    return;
}
if(UA_Server_triggerEvent(m_server, nodeIdEvent, UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER), NULL, UA_TRUE) != UA_STATUSCODE_GOOD) {
    UA_LOG_WARNING(UA_Log_Stdout, UA_LOGCATEGORY_SERVER, "Failed to trigger event for elements addition.");
    return;
}
```

```xml
<UAObjectType NodeId="ns=1;i=1005" BrowseName="1:CncElementsChangeEventType">
    <DisplayName Locale="en">CncElementsChangeEventType</DisplayName>
    <Description Locale="en">Event type that notifies about a change of the list of elements.</Description>
    <References>
        <Reference ReferenceType="HasSubtype" IsForward="false">i=2041</Reference>
        <Reference ReferenceType="GeneratesEvent" IsForward="false">ns=1;i=5001</Reference>
        <Reference ReferenceType="HasComponent">ns=1;i=6011</Reference>
        <Reference ReferenceType="HasComponent">ns=1;i=6009</Reference>
    </References>
</UAObjectType>
<UAVariable DataType="Boolean" ParentNodeId="ns=1;i=1005" NodeId="ns=1;i=6011" BrowseName="1:IsAddition" UserAccessLevel="3" AccessLevel="3">
    <DisplayName Locale="en">IsAddition</DisplayName>
    <Description Locale="en">True if this event corresponds to an addition, false if it corresponds to a removal.</Description>
    <References>
        <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
        <Reference ReferenceType="HasModellingRule">i=78</Reference>
        <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1005</Reference>
    </References>
    <Value>
        <uax:Boolean>false</uax:Boolean>
    </Value>
</UAVariable>
<UAVariable DataType="UInt32" ParentNodeId="ns=1;i=1005" NodeId="ns=1;i=6009" BrowseName="1:Element" UserAccessLevel="3" AccessLevel="3">
    <DisplayName Locale="en">Element</DisplayName>
    <Description Locale="en">The element that has been added/removed.</Description>
    <References>
        <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
        <Reference ReferenceType="HasModellingRule">i=78</Reference>
        <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=1005</Reference>
    </References>
    <Value>
        <uax:UInt32>0</uax:UInt32>
    </Value>
</UAVariable>
```

Valgrind error log
-------------------

```
==2143== Memcheck, a memory error detector
==2143== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==2143== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==2143== Command: ./ua-server
==2143==
==2143==
==2143== Invalid free() / delete / delete[] / realloc()
==2143==    at 0x4C30D3B: free (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==2143==    by 0x13A4CE: UA_delete (open62541.c:13540)
==2143==    by 0x12E707: UA_NodeId_delete (open62541.h:17738)
==2143==    by 0x5AB778: UA_Server_triggerEvent (open62541.c:193379)
==2143==    by 0x1296D9: UaServer::elements_add(unsigned int const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, short const&) (UaServer.cpp:199)
==2143==    by 0x12BBAB: Data::elements_add(Data*, unsigned int const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, short const&) (Data.cpp:69)
==2143==    by 0x127769: main (ua-server.cpp:21)
==2143==  Address 0x6356b80 is 16 bytes inside a block of size 40 alloc'd
==2143==    at 0x4C2FB0F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==2143==    by 0x5AB2F3: getParentsNodeIteratorCallback (open62541.c:193290)
==2143==    by 0x144699: UA_Server_forEachChildNodeCall (open62541.c:32375)
==2143==    by 0x5AB3A4: getParentsNodeIteratorCallback (open62541.c:193304)
==2143==    by 0x144699: UA_Server_forEachChildNodeCall (open62541.c:32375)
==2143==    by 0x5AB3A4: getParentsNodeIteratorCallback (open62541.c:193304)
==2143==    by 0x5AB654: UA_Server_triggerEvent (open62541.c:193360)
==2143==    by 0x1296D9: UaServer::elements_add(unsigned int const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, short const&) (UaServer.cpp:199)
==2143==    by 0x12BBAB: Data::elements_add(Data*, unsigned int const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, short const&) (Data.cpp:69)
==2143==    by 0x127769: main (ua-server.cpp:21)
==2143==
==2143==
==2143==
==2143== HEAP SUMMARY:
==2143==     in use at exit: 0 bytes in 0 blocks
==2143==   total heap usage: 101,115 allocs, 101,118 frees, 232,695,156 bytes allocated
==2143== 
==2143== All heap blocks were freed -- no leaks are possible
==2143== 
==2143== For counts of detected and suppressed errors, rerun with: -v
==2143== ERROR SUMMARY: 3 errors from 1 contexts (suppressed: 0 from 0)
```

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag): 4d8f829550995ac8d7358fa69c7d9dfc69e96326
 - [x] ~~Other OPC UA SDKs used (client or server):~~ *No client involved*
 - [x] Operating system: *Ubuntu 18.04 Docker container running on Arch Linux* (`FROM ubuntu:18.04`)
 - [x] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [x] Critical issue

Pro, 2018-05-31:
@aDogCalledSpot ping

aDogCalledSpot, 2018-05-31:
This will be fixed when https://github.com/open62541/open62541/pull/1816 is merged.

The bug is that in `ua_subscription_events.c` on line 455, `UA_NodeId_delete` is being used instead of `UA_NodeId_deleteMembers`. Changing this should fix the problem.

NIPE-SYSTEMS, 2018-05-31:
Thanks @aDogCalledSpot for the quick answer!

I can confirm that changing `UA_NodeId_delete` to `UA_NodeId_deleteMembers` fixes this issue.

I've patched my local version. Any plans on when https://github.com/open62541/open62541/pull/1816 will be merged? (within days/weeks/months?)

Thanks for the great library!

Pro, 2018-05-31:
#1816 will be merged in the coming days when the last comments are addressed

closed 2018-05-31 by Pro

jpfr, 2018-05-31:
> ==2124==    at 0x5D32D3: findOccupiedSlot (open62541.c:209223)

That's the nodestore. Maybe the NodeId is not initialized.

We'll look into it.

Please give @Pro and @aDogCalledSpot a few days to install more unit tests for events.
This is a bleeding edge feature. And a few kinks remain.