open62541/open62541 #1825: usertokenpolicy problem

WoutvdBerg, 2018-06-01:
Description
===========
Trying to connect to a OFS (OPC factory server) with a UA wrapper from schneider electric, with open62541 but a simple connect with client_connect fails with: no suitable usertokenpolicy found.

Ran 1 Wireshark capture on the OFS server and 1 on an online accessible server to find the difference.

Background Information / Reproduction Steps
===========================================
Both servers support no encrypted connection. Also anynomous connect but it looks like the OFS want to encrypt the connect procedure of a anonymous login on a unencrypted connection with a basic256 encryption. 
UAExpert succeeds to connect to both servers ( but UAExpert supports the clientside user encryption).

Am I doing something wrong or is this a wrong implementation in OFS?

Need it to work, doesn't have to be by the book for me.

run1: OFS, doesn't work with open62541
run2: opcuaserver.com:48010, is working with open62541
[wiresharkruns.zip](https://github.com/open62541/open62541/files/2062062/wiresharkruns.zip)

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-06-01:
Duplicate of #1818 and #934.

Your server requires password encryption even for non-encrypted SecureChannels.

There's no support from our end atm.
Until somebody steps up and does the work. 😄 

So with this server, use anonymous or switch to an encrypted SecureChannel.

closed 2018-06-01 by jpfr

WoutvdBerg, 2018-06-01:
Understood, but I am using anonymous. That's why I think there is a mistake somewhere (probably OFS) because it requires password encryption on anonymous login, which has no password. It can be seen in the wireshark capture run1.

There aren't settings menu in the OFS where this can be setup, so it's hardcoded in their software.

jpfr, 2018-06-01:
Yep. In the wireshark dump, it says 

UserTokenType: Anonymous
SecurityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#Basic256

So it wants an encrypted UserToken. Even though the anonymous Login is used.

WoutvdBerg, 2018-06-01:
So this can be solved if I use a endpoint with encryption that is supported in open62541 with anonymous login?

What is the advantage of encrypting a usertoken if a server is accessible with anonymous login?

jpfr, 2018-06-01:
> So this can be solved if I use a endpoint with encryption that is supported in open62541 with anonymous login?

Yes. At least the EndpointDescription in the Wireshark dump says so.

> What is the advantage of encrypting a usertoken if a server is accessible with anonymous login?

I'm not sure that much though was put into the server configuration.