open62541/open62541 #1836: Unable to set logger when creating server

NIPE-SYSTEMS, 2018-06-09:
Consider the following code (from the tutorial examples):

```cpp
UA_Boolean running = true;
UA_ServerConfig *config = UA_ServerConfig_new_default();
config->logger = &my_fancy_logging_function;
UA_Server *server = UA_Server_new(config);
UA_Server_run(server, &running);
// ...
```

I want to set a custom logging function (`my_fancy_logging_function`).

This is not possible since `UA_ServerConfig_new_default()` already uses `UA_Log_Stdout` to initialize the network layer. So setting `config->logger` after `UA_ServerConfig_new_default()` doesn't have any effect.

Is this a known flaw? Does this exceed the "default" config and so I need generating my own config manually?

References: https://github.com/open62541/open62541/issues/1139, https://github.com/open62541/open62541/pull/1566, https://github.com/open62541/open62541/pull/1641, @janitza-thbe, @jpfr, @Pro

NIPE-SYSTEMS, 2018-06-09:
Thanks @Stasik0 for the quick answer!

Your first tip will not work since `ServerNetworkLayerTCP` is private (only defined internally, when using amalgamation) and the layer is stored as `void *handle` in `struct UA_ServerNetworkLayer`. So casting is needed but not possible because of the private data type. (Why is it `void *`? Because other plugins might use other data types? Then casting might be a bad idea...)

When following your second tip, I need to implement all parts manually that are private in `UA_ServerConfig_new_default()`. This means I need to duplicate:

* `createDefaultConfig()`
* `addDefaultNetworkLayers()`
* `createSecurityPolicyNoneEndpoint()`

Well, I can duplicate them (78 [sloc](https://en.wikipedia.org/wiki/Source_lines_of_code)), that is possible.

The architectural question would be whether to make these three private functions public to save some redundancy (and sloc).

Another option would be not to copy fields from the `struct UA_ServerConfig` but reference them with e.g. a pointer. When changing the server config afterwards, all child configs are also updated. (I don't know if that collides with other things that require an immutable config by copying of the fields.)

My special use-case (currently) is a basic server with a custom TCP-port and custom logging. In the future I might use one of the security policies.

Pro, 2018-11-06:
Duplicate of #2132

closed 2018-11-06 by Pro