open62541/open62541 #1838: client-side deadband

WoutvdBerg, 2018-06-10:
Is deadband filtering supported in the open62541-client?

jpfr, 2018-06-11:
Yes.
You can add the necessary filter description when the MonitoredItem is created.

https://github.com/open62541/open62541/blob/master/include/ua_client_subscriptions.h#L145

closed 2018-06-11 by jpfr

tuxroger, 2019-05-22:
I have open a new issue #2753 since I am unable to find how the client monitored items can be configured with deadband values not to receive to many notifications (temperature sensors and pressure sensors may fluctuate and have noise) .
Thanks for your time!
Roger.

Pro, 2019-05-27:
A full example for the deadband filter in client mode can be found in our unit tests:

https://github.com/open62541/open62541/blob/573f0f085280d25250d83ce9b6ae7817a40526fc/tests/server/check_monitoreditem_filter.c#L315

Look through that file to find more examples.