open62541/open62541 #1839: Custom event returns empty variant

NIPE-SYSTEMS, 2018-06-11:
I'm trying to send a custom event (event type `ns=2;i=1005`, for nodeset see below) from an open62541-server to an open62541-client. It seems that when I'm sending a populated event to the client, the client gets some empty variants instead of the populated data. It seems that this only effects fields of the custom event type. Fields of e.g. the `BaseEventType` seem to work fine.

I might have done it wrong. After hours of debugging I haven't found the issue.

The detailed informations below consist of (hopefully) the whole code that is needed for triggering and receiving the custom event:

1. open62541 client: subscribing to the event
2. open62541 client: subscription callback
3. open62541 server: triggering the event
4. and: the nodeset of the custom event type

### 1. open62541 client: subscribing to the event

```cpp
#define UA_CLIENT_SELECT_CLAUSES_AMOUNT 3

uint16_t m_namespace = 2; // via UA_Client_NamespaceGetIndex

UA_MonitoredItemCreateRequest request = UA_MonitoredItemCreateRequest_default(UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER));
request.itemToMonitor.attributeId = UA_ATTRIBUTEID_EVENTNOTIFIER;
request.requestedParameters.samplingInterval = 100;
UA_EventFilter_init(&m_filter);
m_filter.selectClauses = (UA_SimpleAttributeOperand*)UA_Array_new(UA_CLIENT_SELECT_CLAUSES_AMOUNT, &UA_TYPES[UA_TYPES_SIMPLEATTRIBUTEOPERAND]);
if(!m_filter.selectClauses) {
    m_logger.error("Failed to allocate select clauses for subscription.");
    return false;
}
for(size_t i = 0; i < UA_CLIENT_SELECT_CLAUSES_AMOUNT; ++i) {
    UA_SimpleAttributeOperand_init(&m_filter.selectClauses[i]);
}

// ---------------------------------------------------------------------
// SELECT CLAUSE 1 ("2:IsAddition" from ns=2;i=1005)

m_filter.selectClauses[0].typeDefinitionId = UA_NODEID_NUMERIC(m_namespace, 1005);
m_filter.selectClauses[0].browsePathSize = 1;
m_filter.selectClauses[0].browsePath = (UA_QualifiedName*)UA_Array_new(m_filter.selectClauses[0].browsePathSize, &UA_TYPES[UA_TYPES_QUALIFIEDNAME]);
if(!m_filter.selectClauses[0].browsePath) {
    m_logger.error("Failed to allocate browsepath for subscription.");
    UA_SimpleAttributeOperand_delete(m_filter.selectClauses);
    return false;
}
m_filter.selectClauses[0].attributeId = UA_ATTRIBUTEID_VALUE;
m_filter.selectClauses[0].browsePath[0] = UA_QUALIFIEDNAME_ALLOC(m_namespace, "IsAddition");

// ---------------------------------------------------------------------
// SELECT CLAUSE 2 ("2:Element" from ns=2;i=1005)

m_filter.selectClauses[1].typeDefinitionId = UA_NODEID_NUMERIC(m_namespace, 1005);
m_filter.selectClauses[1].browsePathSize = 1;
m_filter.selectClauses[1].browsePath = (UA_QualifiedName*)UA_Array_new(m_filter.selectClauses[1].browsePathSize, &UA_TYPES[UA_TYPES_QUALIFIEDNAME]);
if(!m_filter.selectClauses[1].browsePath) {
    m_logger.error("Failed to allocate browsepath for subscription.");
    UA_SimpleAttributeOperand_delete(m_filter.selectClauses);
    return false;
}
m_filter.selectClauses[1].attributeId = UA_ATTRIBUTEID_VALUE;
m_filter.selectClauses[1].browsePath[0] = UA_QUALIFIEDNAME_ALLOC(m_namespace, "Element");

// ---------------------------------------------------------------------
// SELECT CLAUSE 3 ("0:Severity" from built-in base event type)

m_filter.selectClauses[2].typeDefinitionId = UA_NODEID_NUMERIC(0, UA_NS0ID_BASEEVENTTYPE);
m_filter.selectClauses[2].browsePathSize = 1;
m_filter.selectClauses[2].browsePath = (UA_QualifiedName*)UA_Array_new(m_filter.selectClauses[2].browsePathSize, &UA_TYPES[UA_TYPES_QUALIFIEDNAME]);
if(!m_filter.selectClauses[2].browsePath) {
    m_logger.error("Failed to allocate browsepath for subscription.");
    UA_SimpleAttributeOperand_delete(m_filter.selectClauses);
    return false;
}
[... 111 lines cut ...]
    <Value>
        <uax:Boolean>false</uax:Boolean>
    </Value>
</UAVariable>
<UAVariable DataType="UInt32" ParentNodeId="ns=2;i=1005" NodeId="ns=2;i=6009" BrowseName="1:Element" UserAccessLevel="3" AccessLevel="3">
    <DisplayName Locale="en">Element</DisplayName>
    <Description Locale="en">The element that has been added/removed.</Description>
    <References>
        <Reference ReferenceType="HasTypeDefinition">i=63</Reference>
        <Reference ReferenceType="HasModellingRule">i=78</Reference>
        <Reference ReferenceType="HasComponent" IsForward="false">ns=2;i=1005</Reference>
    </References>
    <Value>
        <uax:UInt32>0</uax:UInt32>
    </Value>
</UAVariable>
```

I'm leaving out the logs of open62541 with `#define UA_LOGLEVEL 100`, there is nothing interesting, no errors. Just publish requests and the normal notification sending. Do you need Wireshark dumps?

You told me in https://github.com/open62541/open62541/issues/1824#issuecomment-393624451 that events are a bleeding edge feature. Is the feature still not usable? Am I too early again? :wink:

My open62541 HEAD is currently on fd304ae6452b398a49ff5028ae2458fbcfe8aadb.

Thanks in advance!

aDogCalledSpot, 2018-06-11:
The null values occur when an EventFilter has fields which aren't covered by the event. We could fix this by skipping fields which don't apply and using a realloc at the end. Should not be a lot effort, but unfortunately, I'm hard pressed for time this week, so marking it as experimental might be a good idea.

NIPE-SYSTEMS, 2018-06-12:
@aDogCalledSpot For me it would be interesting if I made a mistake in my code. Or do you need time to investigate this question? Maybe I can fix my code in parallel to the potential library-fix if you see a weird usage by my code. Thanks!

jpfr, 2018-06-12:
On the current master branch, using UAExpert results in some empty fields.
But not all fields empty!

With UAExpert, generating events yields the following trace output:

```
13:23:59.028Z|4|33D0* [uastack] OpcUa_Channel_ResponseAvailable: CallResponse
13:23:59.061Z|4|33D0* [uastack] OpcUa_Channel_ResponseAvailable: PublishResponse
13:23:59.071Z|4|20C4* [uastack] OpcUa_Channel_BeginInvokeService: called for PublishRequest!
13:23:59.262Z|2|33D0* [uastack] OpcUa_Channel_ResponseAvailable: Decoding failed! (0x80B00000)
13:23:59.273Z|1|3010* Error: Publish failed [status=0x80b00000]
```

The error code translates to
```
#define UA_STATUSCODE_BADENDOFSTREAM 0x80B00000 /* Cannot move beyond end of the stream. */
```

NIPE-SYSTEMS, 2018-06-16:
A moment ago, I discovered that I use `UA_Server_writeObjectProperty_scalar()` for writing the child nodes of the event node. In contrast, my `ChangeEventType` references the child nodes via `HasComponent`. The documentation of `UA_Server_writeObjectProperty_scalar()` states that it can modify child nodes that are referenced via `HasProperty`.

This might explain why the default fields of the event type (e.g. `Severity` via `HasProperty`) are populated but my special fields not (e.g. `IsAddition` via `HasComponent`).

I will let you know whether a corrected reference type fixes this issue.

NIPE-SYSTEMS, 2018-06-16:
Question: Should I use the correct functions for writing OR adjust the reference types of my child fields?

`BaseEventType` contains several properties (`HasProperty`) but an event type of another major companion standard contains variables (`HasComponent`).

(In the meantime I will search for other companion standards to see how they are defining events. I will then update this comment.)

NIPE-SYSTEMS, 2018-06-16:
@jpfr How are you using UaExpert to receive events? I've dragged the `Server` object into the `Event View` document and selected simply all events.

When I click on the `Refresh` icon button of the `Events` area, I get the following error:

```
Event Plugin: Call ConditionRefresh returned BadInternalError for ServerId Existing App.
```

I'm expecting events to appear in the `Events` area but there is nothing. :cry:

[image]

aDogCalledSpot, 2018-06-20:
I have double-checked the specification on this and it states in Part 4, 7.17.3 (page 133):

> A null value is returned in the corresponding event field in the Publish response if the selected field is not part of the Event or an error was returned in the selectClauseResults of the EventFilterResult.

So I would say this behaviour is desired.

NIPE-SYSTEMS, 2018-06-20:
Thanks @aDogCalledSpot, I agree with you, from the perspective of open62541 the leaving-fields-empty behaviour seems to be correct. In my scenario I mistakenly used components instead of properties for describing my event but filled the data in via properties.

Currently, I've not enough time to test the events (develop a open62541 client that receives these events). I'm okay with closing this issue (and reopen it myself if my server doesn't work).

closed 2018-06-25 by Pro