open62541/open62541 #1840: BADDECODINGERROR due to decoding binary of Boolean message

HamidMapna, 2018-06-11:
Hi.
I'm using open62541 (0.3-rc1 committed version) on ppc be (PowerPC BigEndian) target architecture, QNX OS and QNX momentics IDE . When in client side i try to create subscription request with `UA_Client_Subscriptions(...)`, i get a BADDECODINGERROR. After tracing the error source in server side code, i noticed the error is result of decoding of a Boolean message in `DECODE_BINARY(Boolean){}` method in open62541.c source code.
In this method and in my case, the `ctx->pos = 1208942710, ctx->end = 1208942712, sizeof(bool)=4 `. So the condition `if(ctx->pos + sizeof(bool) > ctx->end)` results to return UA_STATUSCODE_BADCODINGERROR.

Appreciate any hint.

jpfr, 2018-06-11:
Hello @HamidMapna ,

can you provide a Wireshark dump of the network traffic?
We can fix if we can reproduce.

HamidMapna, 2018-06-12:
Thank you. sorry for late reply. 
I tried too much to find a packet relating to OPC UA in Wireshark, but didn't succeed. could you help me what filter should i apply in Wireshark to capture an interested OPC UA packet?

Stasik0, 2018-06-12:
https://github.com/open62541/open62541/wiki/Capturing-traffic-with-Wireshark

HamidMapna, 2018-06-12:
Thank you.
I made too much efforts to get opcua traffic packets from wireshark. But i didn't succeed. I set port49380 for opcua in wireshark through Edit->Preferences->Protocols->OPCUA. Also set this port  number in client connection settings. No opcua packet captured. You can see in following picture.

 [image]

HamidMapna, 2018-06-13:
Anyway i had read somewhere that i must turn off security policy and message encryption. How can i do that in client or server side? Is it true that wireshark cannot capture encrypted packets on secure channels?

mlang-de, 2018-06-13:
It's look like you capture your real ethernet interface "Realtek PCIe GBE...".
Your server runs on the localhost, therefore you must install npcap as described in the link from Stasik0.
After install npcap (https://github.com/nmap/npcap/releases) you have a new "npcap loopback Adapter". With this you can sniff your localhost traffic over Wireshark.
But beware npcap can cause some funny behaviour like Wireshark doesn't shows any Interfaces anymore or some other programs uses WinPcap bring up error because WinPcap 4.3 is not installed. 
My best result I get which this install order:
- remove WinPcap 4.3
- restart computer
- Install npcap with WinPcap compatible mode (is an check button during the Installation)
- restart computer
- if wireshark does not Show any Interfaces, reinstall npcap and reboot.
- good luck

- an alternativ could be RawCap (http://www.netresec.com/?page=RawCap) which works on commandline. Also mentoined in the link from Stasik0.

Here an screenshot from the required npcap loopback Adapter.
[image]

>  Is it true that wireshark cannot capture encrypted packets on secure channels? 

As far as I know you can capture the traffic, but the payload is encrypted and so not readable.

HamidMapna, 2018-06-25:
Finally i found the source of my problem. The problem issued when requesting for subscription in client side (UA_Client_Subscriptions_create (...) method). But in QNX momentics IDE and QNX neutrino RTOS `sizeof(bool)=4` , while in windows OS and visual studio `sizeof(bool)=1`. 
I solved my problem by adding following lines in the for loop of decodeBinaryInternal(...) method and forced to set the size of bool to 1.
`if((type->typeName == "CreateSubscriptionRequest") && (membertype->typeName == "Boolean")
memSize = 1;
else
memSize = membertype->memSize;
`
also changed the following line in DECODE_BINARY(Boolean) method.
`if(ctx->pos + 1  > ctx->end)
return UA_STATUSCODE_BADENCODDINGERROR;`

But i don't know what is the side effects of these changes in next requests of client side.
 Is there any better and fundamental way to force measure the size of bool type be 1 in all OS and compilers.

Thanks in advance

Pro, 2018-06-25:
Thanks for digging further into this!

As a summary:
On QNX the size of `bool` is 4 bytes, while on the majority of systems (including Windows, Linux) it is 1 byte.

As a quick-fix you can try to change the typedef of UA_Boolean to e.g. use `uint8_t` in here:
https://github.com/open62541/open62541/blob/master/include/ua_types.h#L51

I also created a PR (#1871) which makes sure the typedef is used everywhere.

This will also be a TODO for the architecture PR (#1595) where you can then overwrite the definition of bool for specific architectures.

@jpfr should we introduce some type checking on server initialization, where we check if the size of the base datatypes is the expected one, e.g. 

```c
if (sizeof(UA_Boolean) != UA_TYPES[UA_TYPES_BOOLEAN]->memSize) {
// return error code here
}
```

jpfr, 2018-06-25:
If the binary encoding is fixed for all bool lengths, maybe we don't even need that check.
Anyway, `UA_TYPES[UA_TYPES_BOOLEAN]->memSize` is initialized to `sizeof(UA_Boolean)` in the generated typedescription. And it has to. Because we need to be able to step between structure members.

jpfr, 2018-06-26:
Thanks again @HamidMapna!

The fix is in #1874.
We then throw an assertion if overlayed integer encoding is enabled but booleans are the wrong size.
Manually setting UA_BINARY_OVERLAYABLE_INTEGER == 0 solves the problem on those platforms.

commit ee949d3b (2018-06-26): Types: Ensure booleans are 1 byte long for overlayed encoding

Fix the default encoding (no overlays) for longer bool definitions.

This fixes #1840. Thanks to @HamidMapna for uncovering the root cause
for issues on QNX.
PR #1874 (2018-07-02): Types: Ensure booleans are 1 byte long for overlayed encoding
Fix the default encoding (no overlays) for longer bool definitions.

This fixes #1840. Thanks to @HamidMapna for uncovering the root cause
for issues on QNX.
commit 8917135e (2018-07-02): Types: Ensure booleans are 1 byte long for overlayed encoding

Fix the default encoding (no overlays) for longer bool definitions.

This fixes #1840. Thanks to @HamidMapna for uncovering the root cause
for issues on QNX.
closed 2018-07-02 by jpfr