open62541/open62541 #1842: BADDECODINGERROR in namespace0 initialization

jonasgreen88, 2018-06-12:
Description
===========
I have upgraded to the release candidate of version 0.3 and getting a error in the namespace0 initialization. The namespace0.c have been generated using the default xml file in the cmake process. The error occurs when initializing the OutputArguments node (ns=0,i=11494). The node is intialized in the function function_ua_namespace0_107_begin() and it is the call to UA_Server_addNode_begin() that returns UA_STATUSCODE_BADDECODINGERROR (0x80070000).

In function_ua_namespace0_107_begin() the attribute value is intialized by writing to the encoded struct of UA_ExtensionObject variables. I have tracked the error code to be generated in the call chain UA_Server_addNode_begin() -> Operation_addNode_begin() -> AddNode_raw() -> UA_Node_setAttributes() -> copyVariableNodeAttributes() -> copyCommonVariableAttributes() -> UA_decodeBinary() -> decodeBinaryInternal() -> Array_decodeBinary(). The length variable gets the value 536885516 when decoding. https://github.com/open62541/open62541/blob/a5089596dad0225a06e9fe8fef2626f427d51630/src/ua_types_encoding_binary.c#L511

Have anyone seen a similar issue?

Below is the generated function from namespace0.c that encodes the data that fails to be decoded when adding the node:
```
/* OutputArguments - ns=0;i=11494 */

static UA_StatusCode function_ua_namespace0_107_begin(UA_Server *server, UA_UInt16* ns) {

UA_StatusCode retVal = UA_STATUSCODE_GOOD;
UA_VariableAttributes attr = UA_VariableAttributes_default;
attr.minimumSamplingInterval = 0.000000;
attr.userAccessLevel = 1;
attr.accessLevel = 1;
attr.valueRank = 1;
attr.arrayDimensionsSize = 1;
attr.arrayDimensions = (UA_UInt32 *)UA_Array_new(1, &UA_TYPES[UA_TYPES_UINT32]);
attr.arrayDimensions[0] = 0;
attr.dataType = UA_NODEID_NUMERIC(ns[0], 296);

struct {
  UA_String Name;
  UA_NodeId DataType;
  UA_Int32 ValueRank;
  UA_Int32 ArrayDimensionsSize;
  UA_UInt32 *ArrayDimensions;
  UA_LocalizedText Description;
} variablenode_ns_0_i_11494_Argument_0_0_struct;
variablenode_ns_0_i_11494_Argument_0_0_struct.Name = UA_STRING("ServerHandles");
variablenode_ns_0_i_11494_Argument_0_0_struct.DataType = UA_NODEID_NUMERIC(ns[0], 7);
variablenode_ns_0_i_11494_Argument_0_0_struct.ValueRank = (UA_Int32) 1;
variablenode_ns_0_i_11494_Argument_0_0_struct.ArrayDimensionsSize = 1;
variablenode_ns_0_i_11494_Argument_0_0_struct.ArrayDimensions = (UA_UInt32*) UA_malloc(sizeof(UA_UInt32));
variablenode_ns_0_i_11494_Argument_0_0_struct.ArrayDimensions[0]  = (UA_UInt32) 0;
variablenode_ns_0_i_11494_Argument_0_0_struct.Description = UA_LOCALIZEDTEXT("", "");
UA_ExtensionObject *variablenode_ns_0_i_11494_Argument_0_0 =  UA_ExtensionObject_new();
variablenode_ns_0_i_11494_Argument_0_0->encoding = UA_EXTENSIONOBJECT_ENCODED_BYTESTRING;
variablenode_ns_0_i_11494_Argument_0_0->content.encoded.typeId = UA_NODEID_NUMERIC(0, 298);
UA_ByteString_allocBuffer(&variablenode_ns_0_i_11494_Argument_0_0->content.encoded.body, 65000);
UA_Byte *posvariablenode_ns_0_i_11494_Argument_0_0 = variablenode_ns_0_i_11494_Argument_0_0->content.encoded.body.data;
const UA_Byte *endvariablenode_ns_0_i_11494_Argument_0_0 = &variablenode_ns_0_i_11494_Argument_0_0->content.encoded.body.data[65000];
{
retVal |= UA_encodeBinary(&variablenode_ns_0_i_11494_Argument_0_0_struct.Name, &UA_TYPES[UA_TYPES_STRING], &posvariablenode_ns_0_i_11494_Argument_0_0, &endvariablenode_ns_0_i_11494_Argument_0_0, NULL, NULL);
retVal |= UA_encodeBinary(&variablenode_ns_0_i_11494_Argument_0_0_struct.DataType, &UA_TYPES[UA_TYPES_NODEID], &posvariablenode_ns_0_i_11494_Argument_0_0, &endvariablenode_ns_0_i_11494_Argument_0_0, NULL, NULL);
retVal |= UA_encodeBinary(&variablenode_ns_0_i_11494_Argument_0_0_struct.ValueRank, &UA_TYPES[UA_TYPES_INT32], &posvariablenode_ns_0_i_11494_Argument_0_0, &endvariablenode_ns_0_i_11494_Argument_0_0, NULL, NULL);
retVal |= UA_encodeBinary(&variablenode_ns_0_i_11494_Argument_0_0_struct.ArrayDimensions[0], &UA_TYPES[UA_TYPES_UINT32], &posvariablenode_ns_0_i_11494_Argument_0_0, &endvariablenode_ns_0_i_11494_Argument_0_0, NULL, NULL);
retVal |= UA_encodeBinary(&variablenode_ns_0_i_11494_Argument_0_0_struct.Description, &UA_TYPES[UA_TYPES_LOCALIZEDTEXT], &posvariablenode_ns_0_i_11494_Argument_0_0, &endvariablenode_ns_0_i_11494_Argument_0_0, NULL, NULL);
}
size_t variablenode_ns_0_i_11494_Argument_0_0_encOffset = (uintptr_t)(posvariablenode_ns_0_i_11494_Argument_0_0-variablenode_ns_0_i_11494_Argument_0_0->content.encoded.body.data);
variablenode_ns_0_i_11494_Argument_0_0->content.encoded.body.length = variablenode_ns_0_i_11494_Argument_0_0_encOffset;
UA_Byte *variablenode_ns_0_i_11494_Argument_0_0_newBody = (UA_Byte *) UA_malloc(variablenode_ns_0_i_11494_Argument_0_0_encOffset);
memcpy(variablenode_ns_0_i_11494_Argument_0_0_newBody, variablenode_ns_0_i_11494_Argument_0_0->content.encoded.body.data, variablenode_ns_0_i_11494_Argument_0_0_encOffset);
UA_Byte *variablenode_ns_0_i_11494_Argument_0_0_oldBody = variablenode_ns_0_i_11494_Argument_0_0->content.encoded.body.data;
variablenode_ns_0_i_11494_Argument_0_0->content.encoded.body.data = variablenode_ns_0_i_11494_Argument_0_0_newBody;
UA_free(variablenode_ns_0_i_11494_Argument_0_0_oldBody);

struct {
  UA_String Name;
  UA_NodeId DataType;
  UA_Int32 ValueRank;
  UA_Int32 ArrayDimensionsSize;
  UA_UInt32 *ArrayDimensions;
  UA_LocalizedText Description;
} variablenode_ns_0_i_11494_Argument_1_0_struct;
variablenode_ns_0_i_11494_Argument_1_0_struct.Name = UA_STRING("ClientHandles");
[... 43 lines cut ...]

UA_free(variablenode_ns_0_i_11494_Argument_0_0_struct.ArrayDimensions);
UA_ExtensionObject_delete(variablenode_ns_0_i_11494_Argument_0_0);

UA_free(variablenode_ns_0_i_11494_Argument_1_0_struct.ArrayDimensions);
UA_ExtensionObject_delete(variablenode_ns_0_i_11494_Argument_1_0);
retVal |= UA_Server_addReference(server, UA_NODEID_NUMERIC(ns[0], 11494), UA_NODEID_NUMERIC(ns[0], 46), UA_EXPANDEDNODEID_NUMERIC(ns[0], 11492), false);
return retVal;
}
```

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): d88df4281dde5744db06d6c738d0fcc6b73523e8
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system: Proprietary embedded OS
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-06-12:
- Does this result in the entire server startup to fail?
- Is the same behaviour present on master?

@ichrispa This seems to come from the nodeset compiler.
Any ideas for a fix before we do a deep dive?

We should extract the relevant node value and make a separate unit test for the nodeset compiler.

jpfr, 2018-07-07:
Did #1874 resolve this issue as well?

jonasgreen88, 2018-07-09:
Hi, I didn't know about that fix. Will it be merged to 0.3 branch?

jpfr, 2018-07-10:
That's the plan.

jonasgreen88, 2018-07-19:
@jpfr: I merged the fix #1874 into my project which currently uses tag 0.3-rc2 of open62541 without success. I still see this issue. Is there anymore fixes on the master branch that I might need?

closed 2023-09-18 by jonasgreen88