open62541/open62541 #1844: close session after certain time??

bnadem, 2018-06-12:
create 3 subscirption:
and ervery subscription has monitored item,

the subscription settting is :
publishin ginterval : 500
lifetimecount 30000
maxkeeplive count 10000
publish ennabled :1

and sampling interval is 1000

after certain time die session is closed from client,

i dont understand why, can you help me please

B
[closesessionpcapng.txt](https://github.com/open62541/open62541/files/2094141/closesessionpcapng.txt)

Description
===========

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (ersion majo 0 ersion minor 3):
 - [ ] Other OPC UA SDKs used (node opce ua server): 
 - [ ] Operating system: ubuntu
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached attasched
[closesessionpcapng.txt](https://github.com/open62541/open62541/files/2094149/closesessionpcapng.txt)

 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-06-12:
Are you using the v0.3-rc1?
Then an update to the 0.3 branch will fix the problem.

A v0.3-rc2 will follow in the next days.

bnadem, 2018-06-12:
i have only open62541.h and open62541.c

StalderT, 2018-11-04:
Can easily be reproduced with the example: [1844test.txt](https://github.com/open62541/open62541/files/2545852/1844test.txt)

No issue if the client create only one subscritpion.

The securechannel can not be renewed. It fails here : https://github.com/open62541/open62541/blob/0aa816ec159309cf76472168498f3ffdf2662ee1/src/ua_securechannel.c#L956

I think we have a blocker for 0.3

@jpfr can it be the same issue as https://github.com/open62541/open62541/issues/1930 https://github.com/open62541/open62541/pull/1940 ?

jpfr, 2021-07-06:
The fixed were merged long ago.
Reopen if required.

closed 2021-07-06 by jpfr