open62541/open62541 #1848: User Certificate Authenticate supported?

ggokka, 2018-06-13:
I've just found these code in the client.c example source file.

/* anonymous connect would be: retval = UA_Client_connect(client, "opc.tcp://localhost:4840"); */
    retval = UA_Client_connect_username(client, "opc.tcp://localhost:4840", "user1", "password");
    if(retval != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return (int)retval;
    }
https://github.com/open62541/open62541/blob/master/examples/client.c

There are no code through a user certificate.
Is this mean open62541 library doesn't implement user certificate authentication?

jpfr, 2018-06-13:
- The client currently only supports anonymous and username/password
- The server can deal with any kind of authentication mechanism.
But you might have to make adjustments to the access control plugin:
https://github.com/open62541/open62541/blob/master/include/ua_plugin_access_control.h

I will rename the issue and keep it open as a reminder that the certificate authentication is still open for the client.

jpfr, 2019-02-15:
Implemented on master.
The token can be manually set in the client config.

closed 2019-02-15 by jpfr