open62541/open62541 #1853: server_ctt.exe not startable without parameters

Stasik0, 2018-06-17:
Description
===========

Downloaded that win64 version from the front page. Starting server_ctt.exe yields an error:

./server_ctt.exe
[2018-06-17 11:40:06.623 (UTC+0200)] fatal/userland     Missing argument for the server certificate

I guess that it is not what first-time users are expecting

Checklist
=========
 - [ x] open62541 Version (release number or git tag): 0.3-rc2

jpfr, 2018-06-18:
I don't think we shoud include server_ctt in the release.
The information model is pretty idiosyncratic and done only for the sake of the CTT.

How about we make a "showcase" server and include that...

Imho it would be more important to include .dll (.lib) than prepackaged binaries.
Our users want a library, not a fixed server.

Stasik0, 2018-06-18:
Well, users want also a demoserver to try out with uaexpert

PR #1911 (2018-07-10): server_ctt starts without defined certificate
Before the server would just exist.  This fixes #1853.
commit adaa64f3 (2018-07-10): server_ctt starts without defined certificate

Before the server would just exist.  This fixes #1853.
closed 2018-07-10 by jpfr