open62541/open62541 #1855: BadTcpMessageTypeInvalid with 1k node's

SimarglFox, 2018-06-19:
Try to create OpcUaServer, as client use UaExpert. Then i create 500 node's all fine, but then i do 1k nodes or more have error:
Could not connect to server: BadTcpMessageTypeInvalid

Initializing node's code:
`            char nodeName[varsIter.opcUaName.length() + 1];
            memcpy(nodeName, varsIter.opcUaName.c_str(), sizeof(nodeName));
            nodeName[varsIter.opcUaName.length()] = '\0';

            nodeInformation nodeForWork;
            std::unordered_map<int, isagrafVars::isagrafType>::iterator typeIter;
            typeIter = isaTypes.find(varsIter.typeNumber);
            if(typeIter == isaTypes.end())
            {
                break;
            }
            UA_VariableAttributes varAttr = UA_VariableAttributes_default;

            if(typeIter->second.name == "DINT")
            {
                varAttr.dataType = UA_TYPES[UA_TYPES_INT32].typeId;
                nodeForWork.dataTypeId = UA_TYPES_INT32;
            }
            else if(typeIter->second.name == "SINT")
            {
                varAttr.dataType = UA_TYPES[UA_TYPES_SBYTE].typeId;
                nodeForWork.dataTypeId = UA_TYPES_SBYTE;
            }
            else if(typeIter->second.name == "BOOL")
            {
                varAttr.dataType = UA_TYPES[UA_TYPES_BOOLEAN].typeId;
                nodeForWork.dataTypeId = UA_TYPES_BOOLEAN;
            }
            else if(typeIter->second.name == "UDINT")
            {
                varAttr.dataType = UA_TYPES[UA_TYPES_UINT32].typeId;
                nodeForWork.dataTypeId = UA_TYPES_UINT32;
            }
            else if(typeIter->second.name == "USINT")
            {
                varAttr.dataType = UA_TYPES[UA_TYPES_BYTE].typeId;
                nodeForWork.dataTypeId = UA_TYPES_BYTE;
            }
            else if(typeIter->second.name == "UINT")
            {
                varAttr.dataType = UA_TYPES[UA_TYPES_UINT16].typeId;
                nodeForWork.dataTypeId = UA_TYPES_UINT16;
            }
            else if(typeIter->second.name == "REAL")
            {
                varAttr.dataType = UA_TYPES[UA_TYPES_FLOAT].typeId;
                nodeForWork.dataTypeId = UA_TYPES_FLOAT;
            }
            else if(typeIter->second.name == "INT")
            {
                varAttr.dataType = UA_TYPES[UA_TYPES_INT16].typeId;
                nodeForWork.dataTypeId = UA_TYPES_INT16;
            }
            else if(typeIter->second.name == "ULINT")
            {
                varAttr.dataType = UA_TYPES[UA_TYPES_UINT64].typeId;
                nodeForWork.dataTypeId = UA_TYPES_UINT64;
            }
            else if(typeIter->second.name == "LINT")
            {
                varAttr.dataType = UA_TYPES[UA_TYPES_INT64].typeId;
                nodeForWork.dataTypeId = UA_TYPES_INT16;
            }
            //что-то пошло не так, данный тип мы не знаем
            else
            {
                continue;
            }

            varAttr.description = UA_LOCALIZEDTEXT_ALLOC("en-US", nodeName);
            varAttr.displayName = UA_LOCALIZEDTEXT_ALLOC("en-US", nodeName);
            counter++;

            UA_NodeId nodeId = UA_NODEID_STRING(1, nodeName);
            UA_QualifiedName nodeIdName = UA_QUALIFIEDNAME(1, nodeName);
            UA_NodeId parentNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
            UA_NodeId parentReferenceNodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
            UA_Server_addVariableNode(server, nodeId, parentNodeId,
                                      parentReferenceNodeId, nodeIdName,
                                      UA_NODEID_NULL, varAttr, NULL, NULL);`

maybe i do somethink wrong?

jpfr, 2018-07-04:
Are you on a constrained device with little memory?

SimarglFox, 2018-07-04:
Nope, i have 2GB RAM. So i can understand what happened. Problem in MULTITHREADING, how i understand by default it's on. Then i receive a hello packet from client (or discovery), worker rewrite 16byte of packet header and in process chunk, i have trash in head of packet. I go to call stack, and see what before packet go to worker all fine, when worker call processChunk, it's breaked. In my situation flag with  MULTITHREADING=OFF help to resolve my problem. I try it on 1core ARM processor with armhf support.

jpfr, 2018-07-04:
Multithreading is marked as experimental and not for productive use.
Please protect your sever with a semaphore.

Please reopen if a question remains.

closed 2018-07-04 by jpfr