open62541/open62541 #1856: 'BadEncodingError' during Basic256sha256 encrypting in open62541

LucaPolese, 2018-06-19:
Description
===========
> _Mon. 18th June 2018_
I was trying to connect a Server (both made by myself and downloaded from open62541 GitHub link -> **server_basic256sha256.c**) and a client (**Prosys** or **UA_Expert**).
The issue is that when I try to connect my client either goes into an infinite loop or, using a different client, send me back an error: _BadEncodingError_.
I supposed that is a library bug, because I've only copied Git code and I pasted it in mine. So as a double-check I downloaded the test source, as mentioned above.
I hope that someone can help me.

> _Tue. 19th June 2018_
I tryed to Debug the open62541.c code : i found that the the error comes from 
static status NodeId_encodeBinaryWithEncodingMask(UA_NodeId const *src, u8 encoding, Ctx *ctx) {} function.

Background Information / Reproduction Steps
===========================================
**Error in the command prompt**
```lucapolese@lucapolese-Inspiron-5559:~$ ./256server.sh
[2018-06-19 09:54:50.112 (UTC+0000)] warn/userland	No CA trust-list provided. Any remote certificate will be accepted.
[2018-06-19 09:54:50.133 (UTC+0000)] info/network	TCP network layer listening on opc.tcp://lucapolese-Inspiron-5559:4840/
[2018-06-19 09:54:56.124 (UTC+0000)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2018-06-19 09:54:56.128 (UTC+0000)] info/channel	Creating a new SecureChannel
[2018-06-19 09:54:56.128 (UTC+0000)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2018-06-19 09:54:56.128 (UTC+0000)] info/channel	Connection 5 | SecureChannel 1 | Opened SecureChannel

Ret Value Debug : 80080000

Ret Value Debug : 80080000
[2018-06-19 09:54:56.128 (UTC+0000)] info/channel	Connection 5 | SecureChannel 1 | Could not send the OPN answer with error code BadEncodingError
[2018-06-19 09:54:56.128 (UTC+0000)] info/network	Connection 5 | Processing the message failed with error BadEncodingError
[2018-06-19 09:54:56.128 (UTC+0000)] info/network	Connection 5 | Closed
[2018-06-19 09:54:59.437 (UTC+0000)] info/network	Connection 5 | New connection over TCP from 127.0.0.1
[2018-06-19 09:54:59.442 (UTC+0000)] info/channel	Creating a new SecureChannel
[2018-06-19 09:54:59.442 (UTC+0000)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2018-06-19 09:54:59.442 (UTC+0000)] info/channel	Connection 5 | SecureChannel 2 | Opened SecureChannel

Ret Value Debug : 80080000

Ret Value Debug : 80080000
[2018-06-19 09:54:59.442 (UTC+0000)] info/channel	Connection 5 | SecureChannel 2 | Could not send the OPN answer with error code BadEncodingError
[2018-06-19 09:54:59.442 (UTC+0000)] info/network	Connection 5 | Processing the message failed with error BadEncodingError
[2018-06-19 09:54:59.442 (UTC+0000)] info/network	Connection 5 | Closed
```

**Point in the code that brings me back the BadEncodingError:**
`/* For ExpandedNodeId, we prefill the encoding mask. We can return
 * UA_STATUSCODE_BADENCODINGLIMITSEXCEEDED before encoding the string, as the
 * buffer is not replaced. */
static status
NodeId_encodeBinaryWithEncodingMask(UA_NodeId const *src, u8 encoding, Ctx *ctx) {
    status ret = UA_STATUSCODE_GOOD;
    switch(src->identifierType) {
    case UA_NODEIDTYPE_NUMERIC:
        if(src->identifier.numeric > UA_UINT16_MAX || src->namespaceIndex > UA_BYTE_MAX) {
            encoding |= UA_NODEIDTYPE_NUMERIC_COMPLETE;
            ret |= ENCODE_DIRECT(&encoding, Byte);
            ret |= ENCODE_DIRECT(&src->namespaceIndex, UInt16);
            ret |= ENCODE_DIRECT(&src->identifier.numeric, UInt32);
        } else if(src->identifier.numeric > UA_BYTE_MAX || src->namespaceIndex > 0) {
            encoding |= UA_NODEIDTYPE_NUMERIC_FOURBYTE;
            ret |= ENCODE_DIRECT(&encoding, Byte);
            u8 nsindex = (u8)src->namespaceIndex;
            ret |= ENCODE_DIRECT(&nsindex, Byte);
            u16 identifier16 = (u16)src->identifier.numeric;
            /*Down here is where the error comes*/
            ret |= ENCODE_DIRECT(&identifier16, UInt16);
        } else {
            encoding |= UA_NODEIDTYPE_NUMERIC_TWOBYTE;
            ret |= ENCODE_DIRECT(&encoding, Byte);
            u8 identifier8 = (u8)src->identifier.numeric;
            ret |= ENCODE_DIRECT(&identifier8, Byte);
        }
        break;
    case UA_NODEIDTYPE_STRING:
        encoding |= UA_NODEIDTYPE_STRING;
        ret |= ENCODE_DIRECT(&encoding, Byte);
        ret |= ENCODE_DIRECT(&src->namespaceIndex, UInt16);
        if(ret != UA_STATUSCODE_GOOD)
            return ret;
        ret = ENCODE_DIRECT(&src->identifier.string, String);
        break;
    case UA_NODEIDTYPE_GUID:
        encoding |= UA_NODEIDTYPE_GUID;
        ret |= ENCODE_DIRECT(&encoding, Byte);
        ret |= ENCODE_DIRECT(&src->namespaceIndex, UInt16);
        ret |= ENCODE_DIRECT(&src->identifier.guid, Guid);
        break;
    case UA_NODEIDTYPE_BYTESTRING:
        encoding |= UA_NODEIDTYPE_BYTESTRING;
        ret |= ENCODE_DIRECT(&encoding, Byte);
        ret |= ENCODE_DIRECT(&src->namespaceIndex, UInt16);
        if(ret != UA_STATUSCODE_GOOD)
            return ret;
        ret = ENCODE_DIRECT(&src->identifier.byteString, String); /* ByteString */
        break;
    default:
        return UA_STATUSCODE_BADINTERNALERROR;
    }
    return ret;
}`

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (0.3_rc1):
 - [x] Other OPC UA SDKs used (Prosys && UA_Expert Clients): 
 - [x] Operating system: Linux
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [x] Wireshark network dump attached
 - [x] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-06-19:
Thanks for the detailed report.
This has high priority.

- Did you try with v0.3-rc2?
- You tick the box for having a wireshark dump. My browser does not show an attached file. Is this a fault on my end?

LucaPolese, 2018-06-19:
Thank you for you help.
Yes, I've already tried it, with the last stable release, but with the same results.
I've wrongly matched the Wireshark dump, my fault: I've already use it, but I haven't got any relevant information to share.
Hope you guys will solve this issue quickly.

ChHaa, 2018-06-21:
Hi,

i have some problems reproducing your errror, i just tried the main-branch example with UAExpert and everything worked fine out of the box. What exaclty is your server code ?

Infinity95, 2018-06-22:
Hi,

does the problem occur, if you run the demo application from open62541?
Please also provide a wireshark dump.
If you provide a wireshark dump, it would be of great help if you could also include the used private keys and certificates. (Please only do this if you are using throw away certificates for testing. DO NOT provide any confidential keys)
If that is not possible, please be more specific about the setup you are using open62541 in.
What exactly do you mean by 
>  I've only copied Git code and I pasted it in mine

jpfr, 2018-06-30:
The code portion that was quoted does not contain the string `badencodingerror`.
If you can reproduce, can you provide a strack trace and code location of where badencodingerror is coming from?