open62541/open62541 #1858: Linking Erros on Visual Studio 2017

Lizard00, 2018-06-20:
I get linking errors if I try to use the libary.

My installationstepps: 
1. I builded the actual version of open62541 with cmake 3.12.0. (see CMakeOutput)
[cmakeOutput.txt](https://github.com/open62541/open62541/files/2118327/cmakeOutput.txt)

2. I opened the builded solution with Visual Studio and builded the "ALL_BUILD" project (sucsessfully)
3. I builded the whole project (sucsessfully).
4. I created a new Visual Studio project and copyed the server example the .h and the .c into int. 
5. I added the link of the folder which includes the generated lib file to V++, C/C++ and Linker.

But I get some linking errors which I could not solve... 
[vsOutput.txt](https://github.com/open62541/open62541/files/2118372/vsOutput.txt)

Thank you so much for help!

basyskom-jvoe, 2018-06-20:
Have you tried adding ws2_32.lib to your project?

Lizard00, 2018-06-21:
Unfortunately yes.

Pro, 2018-06-21:
Please have a look at the appveyor config.
We are building the stack at every commit with VS2015:
https://github.com/open62541/open62541/blob/master/tools/appveyor/build.ps1

Adding the ws2_32 lib correctly should solve your issue:
https://stackoverflow.com/questions/16948064/unresolved-external-symbol-lnk2019

Lizard00, 2018-06-21:
I reinstalled my Visual Studio, created a new project and now it works. Thank you verry much!

closed 2018-06-22 by jpfr