open62541/open62541 #1859: Could not connect to server: BadTimeout

embbo, 2018-06-20:
Hi, 
I am trying to Implement OPCUA server on stm32f4 using Freertos and LWIP. when i ping the server, it works fine but when i try to connect with UaExpert(OPCUA Client) it gives error "Could not connect to server: BadTimeout". I am using hard coded ip adress 192.168.xx.xx:16664.

HolgerJeromin, 2018-06-20:
can you connect with telnet to this port?

embbo, 2018-06-21:
@HolgerJeromin No it doesn't , when i try to connect it stops or break the connection. what i found is it works till LWIP (because it pings)

Pro, 2018-06-21:
It could be a memory issue, that your server runs out of memory.

You can have a look at my sample project for an ESP32 using LWIP:

https://github.com/Pro/open62541-arduino

cabralfortiss, 2019-02-17:
Did you manage solve this by implementing the time functions? How's is this issue now related to what's been going on in #2227 ?

Pro, 2019-03-08:
Closing this to streamline the discussion in #2511

PR #2511 (2019-04-04): WIP: Cleanup of freeRTOS and LwIP
I want to use this PR to fix all issues related to freeRTOS and LwIP: 
fixes #1859 
fixes #1864 
fixes #2227 
fixes #2460. 

Users working with freeRTOS that confirmed the PR works:

- [X] @embbo (https://github.com/open62541/open62541/pull/2511#issuecomment-475769413)
- [ ] @nidhal-mars (not active in the PR)
- [ ] @FishEatingCat (not active in the PR)
- [X] @VladBraz125 (https://github.com/open62541/open62541/pull/2511#issuecomment-479844345)
- [X] @rudoq32 (https://github.com/open62541/open62541/pull/2511#issuecomment-474811430)
- [ ] @sammer-h (not active  [... cut]
closed 2019-03-08 by Pro