open62541/open62541 #1861: DataSubscription issue when the client doesn't provide enough publish requests before publishing

jonasgreen88, 2018-06-20:
Description
===========
I have found what I belive is an issue in the DataSubscription implementation. I would like to let you evaluate the logs to see what you think. 

I did a setup where a client sets up one subscription with 2 monitored items monitoring different variable nodes. The data source of the variables changes frequently, faster than the configured publishing/sampling intervals.  The queue size of the monitored items are 1. The client does only have 1 outstanding publish request and allows only 1 notification entry per publish request. The publishing interval of the subscription equals the sampling interval of the monitored items (5 seconds in the logs).

When the monitored items detect value changes (in every sample in this case) the server can only produce 1 of the changes and then have to wait for a second publish request to publish the next one. This is where the issue shows up. When the server have responded with the notification of the first monitored item and then receives a new publish request from the client, it doesn't always publish the notification of the second monitored item. See wireshark log "log-monItemQueueSize_1-publishRequests_1-maxNotifPerPubRsp_1.pcapng". The notification of the second monitored items is only published every second publish interval.

The following configuration changes solve/hide the issue:
- Set the monitored item queue size to 2 (log-monItemQueueSize_2-publishRequests_1-maxNotifPerPubRsp_1.pcapng).
- Configure the client to initially provide 2 publish requests (log-monItemQueueSize_1-publishRequests_2-maxNotifPerPubRsp_1.pcapng).
- Allow 2 notifications in one publish request (log-monItemQueueSize_1-publishRequests_2-maxNotifPerPubRsp_1.pcapng).
What is common with all these changes that solve/hide the issue are that the server have enough resource to publish all changes immediately when they occurs, it doesn't have to wait for a new publish requests from the client.

What do you guys think? I guess it is a valid configuration that should work?

Attached logs: [wireshark-logs.zip](https://github.com/open62541/open62541/files/2119138/wireshark-logs.zip)

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): d88df4281dde5744db06d6c738d0fcc6b73523e8
 - [ ] Other OPC UA SDKs used (client or server): Client lib: https://github.com/FreeOpcUa/python-opcua
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jonasgreen88, 2018-06-20:
If I setup 2 subscriptions with 1 monitored items each (same amount of monitored items as above) it doesn't matter if the client have 1 or 2 outstanding publish requests. In both cases some similar issue occur.
Attached logs: [wireshark-logs-2-subscriptions.zip](https://github.com/open62541/open62541/files/2119217/wireshark-logs-2-subscriptions.zip)

jpfr, 2018-07-02:
Solved via #1885.

closed 2018-07-02 by jpfr