open62541/open62541 #1864: porting to lwip + FreeRTOS  and STM32Fxx_HAL libraries.

nidhal-mars, 2018-06-21:
i am very interested to have an embedded OPC UA server for use with LWIP+ FreeRTOS and STM32Fxx_HAL libraries.
from which point i need to start the porting because i am new with open62541 ?

Pro, 2018-06-21:
You can have a look at my example project here:

https://github.com/Pro/open62541-arduino

I ported open62541 onto an ESP32 using LWIP.
You should also have a look at #1595

FishEatingCat, 2018-11-16:
> i am very interested to have an embedded OPC UA server for use with LWIP+ FreeRTOS and STM32Fxx_HAL libraries.
> from which point i need to start the porting because i am new with open62541 ?

Hello, I want to know if you have completed the porting of open62541 on STM32? I also recently tried to add open62541 to STM32. I am using Windows XP to write STM32 program with MDK 5. The chip I used is STM32F4. What I want to know now is how to compile open62541 for stm32. I hope you Can give me some advice.
thank you very much

jpfr, 2018-11-18:
Do you use an RTOS? We have support for FreeRTOS and a couple more.

FishEatingCat, 2018-11-30:
> Do you use an RTOS? We have support for FreeRTOS and a couple more.

Thank you for your reply ！
Yes, I use freeRTOS, but when I compile open62541 with cmake, I don't know how to compile freeRTOS and lwip files or paths into open62541.
I tried to find some open62541 files with freeRTOS on the Internet, but I was prompted to use some LWIP files, functions and some data definitions.

Pro, 2018-11-30:
@cabralfortiss can you help out?

cabralfortiss, 2018-11-30:
Hi. First of all you need to understand the layers of code you are using. 
The bottom layer would be the code for your chip, in this case STM32F4. You said you are writing already some program for it, so I assume you have all the compilers and headers for this chip (at least it's in your computer). If you had something running on the chip, you have it.
The second layer would be, in this case, the freeRTOS, the operative system running on your chip. You should have the code in your computer, which in the port.c file you should have the link to the first layer. If you have already something running freeRTOS in your chip, you're done. 
On a third layer (and maybe somwhow in parallel with the second) you have LwIP, a networking code which will run using the freeRTOS code and the code in the first layer. This link is done in some files in a folder called port. 

I assume that these three layers you have already tested, and the problem is just with open62541. This code, we could say, goes on top of all, since it will be using the code from freeRTOS and LwIP. 
So, the following is important: you compile open62541 as a library. The header files from LWIP and freeRTOS should be present in your computer, and you should point to them using the right variable in CMake. Both are open source project, so you can just clone the repos.
In Cmake you should first select the compiler when you first press configure (it's easier to use the CMake-GUI first). This compiler MUST be the same as the one you're using for layers 1 - 3. If you managed to compile for any of 1- 3 the compiler is in your computer. You will need to look for it. The best clue is to look for  in the program you are using (MDK 5, from what I understood). Somewhere there in the settings, should be a cross-compiler.
Once you have it, in CMake, choose freeRTOSLWIP as UA_ARCHITECTURE. You will get then more options. The most important are:

    UA_ARCH_EXTRA_INCLUDES which you should point to the folders where the headers files from LWIP and freeRTOS, separeted by semicolon (;).

    UA_ARCH_ADD_FLAGS, UA_ARCH_LINKER_FLAGS and UA_ARCH_REMOVE_FLAGS must be set according to the compiler you're using, so I cannot give more details now here. What I did was: in my IDE where I was compiling my normal C code for freeRTOS, I copy the flags and used them. Then when compilation fails, I had to remove some compilers flags (-Wconversion in my case) which weren't supported or were causing compilation errors.

    UA_ARCH_FREERTOS_USE_OWN_MEMORY_FUNCTIONS is a boolean that you can set to true if you don't want to use free/malloc/calloc/realloc but the ones from the external implementation of freeRTOS called vPortFree/pvPortMalloc/pvPortCalloc/pvPortRealloc

When you are done with all of these, you compile and you get a library and a header (open62541.h). These two files you will need to point in your MDK 5 tool somehow. Look in the settings, and in the build configuration should be somewhere a place to add libraries. Put the files somewhere in your project, maybe in the same folder as your main, so you won't have any problems. 

The in your code, your "normal" code, where you were writing already programs, you should be able to ```#include "open62541.h" ``` and use the functions of open62541 like, I don't know, create a server, iterate and so on. 

It's seems a little complicated, but it's worth the effort.

cabralfortiss, 2018-12-03:
> Finally, I want to compile and generate an open62541.c and open62541.h file, so I can easily add it to my other devices and make it possible for my colleagues to use it directly. The generated file will be placed in the folder of freertos and lwip, and the corresponding header file path can be set in MDK.
> Indeed, this process is quite difficult. Even I thought it was an impossible task. At that time, when I indicated at https://github.com/cmbahadir/opcua-esp32, I completed the ESP32 operation open62541 to establish OPC UA. Server, and can successfully read the corresponding value, I believe that STM32 must also be able to successfully run open62541!

I think your way is even much easier. I didn't think about that. So if you just need the .c and .h files it should be more direct. As you said, you should put these files (once generated) in your MDK project, and the .c file to the compilation and the .h file put it in a folder you have already in your project which is already incldued in the compilation. Since the freeRTOS and LwIP code is already being compiled, the include folders should be already set in your MDK tool for the project. The generated .c and .h would then find the corresponding header files from the project settings.

> For the code layer you mentioned, I can understand that the bottom layer, the second layer, and the third layer have all been tested and can be run separately. The real problem is the separate .c and .h files generated by the compilation of open62541.

> 
> The first question, I think you should talk about the freertos and lwip header files should contain all the files used in the code, right?

As said, if you have compiled freeRTOS and LwIP, the headers path are already set in your project, and the generated .c and .h won't need anymore. 

> 
> The second question, is the migration of lwip to port all of it into the code? Because I didn't fully use lwip when I was using lwip, I just chose some features for porting.

I'm not sure about this, but probably if you didn't port something from LwIP to freeRTOS you should get a linking error when compiling the project with the generated .c there.

> 
> The third question, when I compile with cmake, the Configure
> Specify the generator for this project
> Visual Studio 15 2017 ARM
> Optional toolset to use (argument to -T)
>        Use default native compilers
>     √ Identify native compilers
>        Specify toolchain file for cross-compiling
>        Specify optiocs for cross-conmpiling
> My computer is installed visual studio 2017 Community, my chip is ARM architecture, under F:\work\CodeSource\OPCUA\open62541-master-12-3\tools\cmake, I did not find the .cmake file to compile arm. So the second item I chose, configured as Compilers
> C C:/Keil_v5/ARM/ARMCC/bin/armcc.exe
> C++ C:/Keil_v5/ARM/ARMCC/bin/armcc.exe
> Fortran
> In mdk, C/C++ has armcc compiled.
> 

I'm not sure here, but I think that technically you could use the native compiler, since you just want ot generate the .c and .h files. You would need to activate UA_ENABLE_AMALGAMATION, for UA_ARCHTECTURE select freeRTOSLWIP and UA_ARCH_FREERTOS_USE_OWN_MEMORY_FUNCTIONS set to true if you have self implementation of pvPortMalloc and family.

> The fourth question, for the UA_ARCH_ADD_FLAGS, UA_ARCH_LINKER_FLAGS and UA_ARCH_REMOVE_FLAGS you mentioned, I didn't have any values ​​for the cmake compilation. I wonder if it will affect the compilation result?
> 

If you won't compile them directly, but add the .c and .h files to your project, then you don't need to do anything with these flags.

> The fifth problem, the value of UA_ARCH_FREERTOS_USE_OWN_MEMORY_FUNCTIONS corresponds to TRUE, and I have completed the corresponding code of vPortFree / pvPortMalloc / pvPortCalloc / pvPortRealloc in freertos.
> 

That's fine. If set to true, it will use the standard malloc, free and family. 

> 
> Once again, express sincere gratitude!

So in conclusion, to generate open62541.c and .h files to later compile them in a IDE where you have already a project with freeRTOS and LwIP: 
  
- use any compiler in CMake (not sure about this)
- activate UA_ENABLE_AMALGAMATION, for UA_ARCHTECTURE select freeRTOSLWIP and UA_ARCH_FREERTOS_USE_OWN_MEMORY_FUNCTIONS set to true if you have self implementation of pvPortMalloc and family.
- Click generate in CMake.
- Add the files to your project in your IDE. Tell your IDE that open62541.c must be compiled.
- Add the code to use the open62541 stack (create server and so on)
- Compile your project as before. 
- If it compiles, you should be able to see your opcua server.

FishEatingCat, 2018-12-04:
Hi
After cmake compiled and generated open62541.c and open62541.h, I added it to my project and compiled a lot of errors and warnings. There is a detailed compilation process at https://github.com/FishEatingCat/STM32F4-freertos-lwip-open62541/blob/master/CompileLog.txt.
1.ssize_t is undefined, I added #define ssize_t int in the open header file to solve this problem.
2. UA_ARCHITECTURE_FREERTOSLWIP is undefined. I have selected freertosLWIP when compiling open62541 using cmake. I defined UA_ARCHITECTURE_FREERTOSLWIP myself in the header file. Then the compile prompt <sys/time.h> won't open. After the query, I changed it to <../sys/time.h> to open it, but there is no gmtime_r() function.
3. struct timespec, struct addrinfo is not defined, the search in the project can not be found. The struct addrinfo is in lwip/nerdb.h, but it is the prompt undefined. It can be used normally in other files.
4.AI_PASSIVE is undefined

These are the problems I have discovered for the time being, some of them are solved, and some can't be solved.
Thank you very much for your reply!

cabralfortiss, 2018-12-04:
> Hi
> After cmake compiled and generated open62541.c and open62541.h, I added it to my project and compiled a lot of errors and warnings. There is a detailed compilation process at [...]

I forgot about the UA_ARCHITECTURE_FREERTOSLWIP flag. Yes you should pass this to your compilation somehow

It's very strange that <sys/time.h> is not present. This depends on your compiler so I cannot help with this. This is probably the root of the problem for struct timespec and gmtime_r. You should find function somewhere in the header files of your compiler. 

> 3. struct timespec, struct addrinfo is not defined, the search in the project can not be found. The struct addrinfo is in lwip/nerdb.h, but it is the prompt undefined. It can be used normally in other files.
> 4.AI_PASSIVE is undefined

struct addrinfo and AI_PASSIVE  should be there. As you said, is in lwip/netdb.h which is actually included in open62541.h. 

> 
> These are the problems I have discovered for the time being, some of them are solved, and some can't be solved.
> Thank you very much for your reply!

And all other erross I see are because lwip/netdb.h which is behind a ```#if LWIP_DNS && LWIP_SOCKET ``` so in your file lwipopts.h you should define these to 1

FishEatingCat, 2018-12-06:
Hi
There is still a fatal error in open62541: _Error: L6218E: Undefined symbol time (referred from open62541.o)__. I can't find any definitions about time in open62541.c and open62541.h.
In addition, I did not find _int gettimeofday(struct timeval *tv, struct timezone *tz), struct tm *gmtime_r(const time_t *timep, struct tm *result) and int access(char *name, int type)_ in freertos Function, can you give me some advice? I have updated the progress of the new project to github.

VladBraz125, 2019-02-18:
Did you have a some solution? I have the same problem with cmake compiled and generated open62541.c and open62541.h.

cabralfortiss, 2019-02-18:
mmm, maybe UA_DateTime_now and UA_DateTime_localTimeUtcOffset should be reimplemented to use xTaskGetTickCount() and you'll have the time probably set to 1970 everytime your hardware starts, and avoid using gettimeofday and gmtime_r which are not part of freeRTOS. 

Regarding access, you could disable UA_ENABLE_DISCOVERY, or keep it enabled, but disable UA_ENABLE_DISCOVERY_SEMAPHORE altought I don't what consequences will bring this to your cases

rudoq32, 2019-02-20:
Hello guys,
I am trying a similar thing with NUCLEO F767ZI. I would prefer to compile a static library instead of having open62541.c/.h.
I have already read some other threads like #1595. I cannot say that I am wiser after having it read...
Anyway following what @cabralfortiss wrote I did:
[image]
in Specify the generator for this project I selected "Visual Studio 15 2017"
and as native compilers I selected:
[image]
(these are used in my project in System Workbench for STM32)
Then in VS I tried to build ALL_BUILD and for sure I got error that cannot open file: ua_lwip.h.
Am I following the right direction? Do you know what I forgot?

rudoq32, 2019-02-21:
Hello guys,
I am trying for already couple of weeks compile the open62541 as static library for my NUCLEO F767ZI. I read #1595 and followed @cabralfortiss notes.
I chose option "Specify native compilers" (I am not sure about the Fortran)
[image]
I fed UA config with all includes and flags my project in System Workbench for STM32 is haveing:
[image]

I generated and opened a project. After building ALL_BUILD I am having like 1236 errors, mainly:

Severity	Code	Description	Project	File	Line	Suppression State
Error (active)	E1696	cannot open source file "unistd.h"	open62541-object	C:\Projects\open62541-master\arch\common\ua_freeRTOS.h	17	

Severity	Code	Description	Project	File	Line	Suppression State
Error (active)	E0757	variable "uint32_t" is not a type name	open62541-object	C:\Users\a2user\STM32Cube\Repository\STM32Cube_FW_F7_V1.14.0\Drivers\CMSIS\Device\ST\STM32F7xx\Include\stm32f767xx.h	217	

Severity	Code	Description	Project	File	Line	Suppression State
Error	C1189	#error:  Unknown compiler	open62541-object	C:\Users\a2user\STM32Cube\Repository\STM32Cube_FW_F7_V1.14.0\Drivers\CMSIS\Include\core_cm7.h	118	

Basically I have 2  question:
Am I following right way to compile open62541.a library?
By a chance do you see something obvious what I am doing wrong?

Many thanks Rudo

FishEatingCat, 2019-02-25:
> Hello guys,
> I am trying for already couple of weeks compile the open62541 as static library for my NUCLEO F767ZI. I read #1595 and followed @cabralfortiss notes.
> I chose option "Specify native compilers" (I am not sure about the Fortran)
> [image]
> I fed UA config with all includes and flags my project in System Workbench for STM32 is haveing:
> [image]
> 
> I generated and opened a project. After building ALL_BUILD I am having like 1236 errors, mainly:
> 
> Severity	Code	Description	Project	File	Line	Suppression State
> Error (active)	E1696	cannot open source file "unistd.h"	open62541-object	C:\Projects\open62541-master\arch\common\ua_freeRTOS.h	17
> 
> Severity	Code	Description	Project	File	Line	Suppression State
> Error (active)	E0757	variable "uint32_t" is not a type name	open62541-object	C:\Users\a2user\STM32Cube\Repository\STM32Cube_FW_F7_V1.14.0\Drivers\CMSIS\Device\ST\STM32F7xx\Include\stm32f767xx.h	217
> 
> Severity	Code	Description	Project	File	Line	Suppression State
> Error	C1189	#error: Unknown compiler	open62541-object	C:\Users\a2user\STM32Cube\Repository\STM32Cube_FW_F7_V1.14.0\Drivers\CMSIS\Include\core_cm7.h	118
> 
> Basically I have 2 question:
> Am I following right way to compile open62541.a library?
> By a chance do you see something obvious what I am doing wrong?
> 
> Many thanks Rudo
> 
> > Added:
> > I tried also build open62541 .c/.h. I got an error that compiler is not defined. So I defined **GNUC** as compiler.
> > Now I am having like 2775 errors, mostly like:
> > Severity	Code	Description	Project	File	Line	Suppression State
> > Error	C2143	syntax error: missing ')' before '('	open62541-object	c:\users\a2user\stm32cube\repository\stm32cube_fw_f7_v1.14.0\drivers\cmsis\include\cmsis_gcc.h	58
> > `/**`
> > `  \brief   Disable IRQ Interrupts`
> > `  \details Disables IRQ interrupts by setting the I-bit in the CPSR.`
> > `  Can only be executed in Privileged modes.`
> > ` */`
> > `__attribute__( ( always_inline ) ) __STATIC_INLINE void __disable_irq(void)`
> > `{`
> > `  __ASM volatile ("cpsid i" : : : "memory");`
> > `}`

Sorry to see it now.
You can view[#2227](https://github.com/open62541/open62541/issues/2227),here recorded in detail the process of compiling open62541.

FishEatingCat, 2019-02-25:
>

FishEatingCat, 2019-02-25:
Sorry to see it now.
You can view[#2227](https://github.com/open62541/open62541/issues/2227),here recorded the process of compiling open62541.

rudoq32, 2019-02-27:
Hi @FishEatingCat , I reviewed #2227... the initial stage of this thread is "it build fine". Unfortunately I am having problems even to get in that stage. On Friday hopefully I will meet a guy who can explain me how to compile the project.

cabralfortiss, 2019-02-28:
Hi all. I just pushed a patch to my repository and opened a PR. I added some documentation of what is important to make it work. 

I'll kindly ask everyone (@nidhal-mars, @FishEatingCat, @rudoq32 @VladBraz125) here to tets it and if it fails, keep pushing fixes to the PR until we have a nice working version for all.

rudoq32, 2019-03-02:
Hi @cabralfortiss,
I am still struggling with compiling the project.
If you have few minutes, I have couple of questions...
1) Can I use for this purpose cmake-gui?
2) Can I set "Visual Studio 15 2017 ARM" as generator for this project?
3) Can I check "Specify native compiler"?
3.1) Can I set C compiler as "C:/Ac6/SystemWorkbench/plugins/fr.ac6.mcu.externaltools.arm-none.win32_1.17.0.201812190825/tools/compiler/bin/arm-none-eabi-gcc.exe"
3.2) Can I set C++ compiler as "C:/Ac6/SystemWorkbench/plugins/fr.ac6.mcu.externaltools.arm-none.win32_1.17.0.201812190825/tools/compiler/bin/arm-none-eabi-g++.exe"
3.3) Can I leave Fortran compiler empty?
4) Can I leave Optional toolset to use empty?

Is there something else to make my VS to use my specified compilers?
Thanks much

cabralfortiss, 2019-03-03:
> I am still struggling with compiling the project.
What kind of errors are you having now? The unistd.h problem should be gone by now. 
Have you compiled any code for your board? If yes, how did you do it?
> 
>     1. Can I use for this purpose cmake-gui?
Yes, you can
> 
>     2. Can I set "Visual Studio 15 2017 ARM" as generator for this project?
I'm not sure because I never tried it. I assume it shouldn't be a problem, but with windows tools that's not always the case.  
> 3. Can I check "Specify native compiler"?
>        3.1) Can I set C compiler as [...]
Yes to all. I saw yor picture from your comment before and it looks fine. The only thing I don't see is the "Target Root" which is where you usually set the path that helps finding the incldues for the compiler.

> 
>     4. Can I leave Optional toolset to use empty?
> 
I don't know what do you mean with this. I cannot find this option.
> 
> Is there something else to make my VS to use my specified compilers?
I'd recommend the following:
1- If you have and IDE to compile projects for your hardware, use it. Add the generated .c/.h files from open62541 and compile there.
2- If that's not possible, I'd recommend to try make-based tools. My git installation in windows has a bash terminal that I use. My mingw-w64 compiler has a make.exe which I use even with cross-compilers. I'm not saying that VS won't work, but I think you'll have it harder doing like that.

rudoq32, 2019-03-06:
Hi @cabralfortiss ,
Thank you for your answer. I gave up of idea to build a static library, I found it too complex. I was able to create amalgamated .c/.h with MS VS, but at this point I finished with VS.
I added amalgamated files to my project in Atollic TrueSTUDIO, and after adding some flags I can compile it successfully.
Unfortunately I still cannot test your #2511, because nucleoF767ZI freertos doesn't have gettimeofday (always returns 0) so I need to work on this little bit.

cabralfortiss, 2019-03-06:
#2511 doesn't have gettimeofday anymore. If you pull the branch from the PR, then your amalgamated files shouldn't have gettimeofday anymore.

VladBraz125, 2019-03-06:
Hi **rudoq32**, Witch flags did you added? Can you make a Wiki for Cmake Step by Step? I have no success with generating and compiling with MS VS.

cabralfortiss, 2019-03-06:
To generate the .c/.h files with freeRTOS you just need to do the same procedure in CMake as your normal compilation for windows. 

1. Select your usual version of VS as "generator for this project" and select the "Use default native compilers".
2. In CMake then select freertosLWIP as UA_ARCHITECTURE instead of default win32. Enable also UA_AMALGAMATION. 
3. Open the project in VS, try to build everything. It will fail, but the .c/.h files will appear in the folder where the VS solution is
4. These files should then be added to your project (like Atollic TrueSTUDIO)
5. In the project add the project variable UA_ARCHITECTURE_FREERTOSLWIP (depends on the IDE how to do this)

VladBraz125, 2019-03-06:
I made everything as you sade and receive the error as in begin "../Application/Inc/open62541.h(180): error:  #5: cannot open source input file "unistd.h": No such file or directory  #include <unistd.h> // read, write, close".

cabralfortiss, 2019-03-07:
That means that you are not using the code from #2511. The unistd.h file was removed from the architecture from freeRTOS. You need to work with this branch: https://github.com/cabralfortiss/open62541/tree/fixFreeRTOS

```
$ git remote add cabralfortiss https://github.com/cabralfortiss/open62541
$ git fetch cabralfortiss 
$ git checkout fixFreeRTOS
```

rudoq32, 2019-03-07:
Got the #2511, thanks for git explanation... 
@VladBraz125, I am not familiar with Wiki, but...
1) in root folder modify CMakeLists.txt 
* add cmake_policy(SET CMP0054 OLD) at second line -  for me this works
* set set(UA_ARCHITECTURE "freertosLWIP" CACHE STRING "Architecture to build open62541 on")
* set option(UA_ENABLE_AMALGAMATION "Concatenate the library to a single file open62541.h/.c" ON)
2) in  \arch folder modify CMakeLists.txt (I am not sure if it is necessary)
* SET(UA_ARCH_ADD_FLAGS "-DUA_ARCHITECTURE_FREERTOSLWIP;-DUA_FREERTOS" CACHE STRING "Flags to be added to compilation that depends on the architecure")
* SET(UA_ARCH_REMOVE_FLAGS "-U_WIN32" CACHE STRING "Flags to be removed from compilation that depends on the architecure")

3) In CMAKE-gui set Visual Studio 15 2017 ARM, hit 2 times "Configure", then "Generate" and "Open Project".

4) Build the project in VS. You will get errors e.g. missing headers, ... **BUT the open62541.h/c are generated.** These files I included in my Atollic IDE solution (seems to be more stable and faster as System Workbench).  

In CubeMX: 

* You need to create in CubeMX program with LWIP and FreeRTOS with options (in Code Generator)  "Copy all used libraries... "  and "Generate peripheral ..."

In Atollic:

* Add C Compiler symbols UA_FREERTOS, UA_ARCHITECTURE_FREERTOSLWIP
* Set  -std=gnu99

That's were I finished so far.

cabralfortiss, 2019-03-07:
@rudoq32 did you compile and it works?

> 
>     1. in root folder modify CMakeLists.txt
> 
> 
>     * set set(UA_ARCHITECTURE "freertosLWIP" CACHE STRING "Architecture to build open62541 on")
> 
>     * set option(UA_ENABLE_AMALGAMATION "Concatenate the library to a single file open62541.h/.c" ON)
>     1. in  \arch folder modify CMakeLists.txt (I am not sure if it is necessary)
> 
> 
>     * SET(UA_ARCH_ADD_FLAGS "-DUA_ARCHITECTURE_FREERTOSLWIP;-DUA_FREERTOS" CACHE STRING "Flags to be added to compilation that depends on the architecure")
> 
>     * SET(UA_ARCH_REMOVE_FLAGS "-U_WIN32" CACHE STRING "Flags to be removed from compilation that depends on the architecure")

You don't need to change the CMakeList file, but set this in the cmake-gui. 

It should work without the UA_FREERTOS variable

rudoq32, 2019-03-07:
@cabralfortiss , nopit I cannot compile it (in Atollic). With your PR I am getting 
`..\Src\open62541.c:49936:27: error: 'UA_Connection {aka struct UA_Connection}' has no member named` `'lwip_close'; did you mean 'close'?`
`                 connection->close(connection);`
with Master version I can compile it. It seems that lwip defines close as lwip_close

cabralfortiss, 2019-03-07:
did you define LWIP_COMPAT_SOCKETS to 0 in lwipopts.h?

Remember to define as in https://github.com/cabralfortiss/open62541/blob/fixFreeRTOS/arch/common/ua_lwip.h#L12

rudoq32, 2019-03-07:
no, I had it as 1. I aligned configuration mentioned in comment.
It was built successfully. 
Program fails at
`status = UA_Client_readValueAttribute(client, UA_NODEID_STRING(1, "the.answer"), &value);`
I need to figure out where exactly...

VladBraz125, 2019-03-07:
..\Application\Src\open62541.c(46881): warning:  #1293-D: assignment in condition
      while((pre = UA_Session_dequeuePublishReq(session))) {
..\Application\Src\open62541.c(49929): error:  #20: identifier "ssize_t" is undefined
          ssize_t n = 0;
..\Application\Src\open62541.c(49961): error:  #70: incomplete type is not allowed
          struct timeval tmptv = {(long int)(timeout_usec / 1000000),
..\Application\Src\open62541.c(49992): error:  #20: identifier "ssize_t" is undefined
      ssize_t ret = UA_recv(connection->sockfd, (char*)&response->data[offset],
..\Application\Src\open62541.c(50277): error:  #70: incomplete type is not allowed
      struct timeval tmptv = {0, timeout * 1000};
..\Application\Src\open62541.c(50552): error:  #70: incomplete type is not allowed
          struct timeval tmptv = { (long int) (timeout_usec / 1000000),
..\Application\Src\open62541.c(50804): error:  #70: incomplete type is not allowed
              struct timeval tmptv = {(long int) (timeout_usec / 1000000),

cabralfortiss, 2019-03-07:
> ..\Application\Src\open62541.c(46881): warning: #1293-D: assignment in condition
> while((pre = UA_Session_dequeuePublishReq(session))) {
> ..\Application\Src\open62541.c(49929): error: #20: identifier [...]

Try adding 

#define ssize_t int

before including open62541.h in your project.

> ..\Application\Src\open62541.c(50277): error: #70: incomplete type is not allowed
> struct timeval tmptv = {0, timeout * 1000};
> ..\Application\Src\open62541.c(50552): error: #70: incomplete type is [...]

Try setting LWIP_TIMEVAL_PRIVATE to 1 (or just don't set it to 0 and it will have a default 1)

VladBraz125, 2019-03-07:
..\Application\Src\open62541.c(46881): warning:  #1293-D: assignment in condition
      while((pre = UA_Session_dequeuePublishReq(session))) {
..\Application\Src\open62541.c(49929): error:  #20: identifier "ssize_t" is undefined
          ssize_t n = 0;
..\Application\Src\open62541.c(49961): error:  #70: incomplete type is not allowed
          struct timeval tmptv = {(long int)(timeout_usec / 1000000),
..\Application\Src\open62541.c(49992): error:  #20: identifier "ssize_t" is undefined
      ssize_t ret = UA_recv(connection->sockfd, (char*)&response->data[offset],
..\Application\Src\open62541.c(50277): error:  #70: incomplete type is not allowed
      struct timeval tmptv = {0, timeout * 1000};
..\Application\Src\open62541.c(50552): error:  #70: incomplete type is not allowed
          struct timeval tmptv = { (long int) (timeout_usec / 1000000),
..\Application\Src\open62541.c(50804): error:  #70: incomplete type is not allowed
              struct timeval tmptv = {(long int) (timeout_usec / 1000000),
..\Application\Src\open62541.c: 14 warnings, 6 errors

cabralfortiss, 2019-03-07:
can you explain at least something and not just paste the same output?

VladBraz125, 2019-03-07:
It sims so ssize_t is not a definded, and timeval tmptv is not allowed struct. :)

cabralfortiss, 2019-03-07:
did you try https://github.com/open62541/open62541/issues/1864#issuecomment-470444645?

rudoq32, 2019-03-07:
@VladBraz125 
in my project the ssize_t  is defined in <project>\Middlewares\Third_Party\LwIP\src\netif\ppp\utils.c
are you using CubeMX to generate the project?
I am suggesting to use it and also modify all flags there. It automatically enable/disable other necessary flags for LWIP middleware.
[image]
(for first project generation use  Copy all used...)

cabralfortiss, 2019-03-07:
@rudoq32 how's your server running? 

where do you have `status = UA_Client_readValueAttribute(client, UA_NODEID_STRING(1, "the.answer"), &value);` ?

I'd recommend to use the same example as in #2511 to test and see if you can connect to it using UaExpert so we know that the basic example runs.

VladBraz125, 2019-03-07:
Thanks, It solve a ssize_t problem bat not " incomplete type is not allowed
struct timeval tmptv"

cabralfortiss, 2019-03-07:
did you set LWIP_TIMEVAL_PRIVATE  to 1?

Pro, 2019-03-08:
Please continue any discussion on STM32 and freeRTOS in #2511 to have all the issues and open questions in one thread

closed 2019-03-08 by Pro

zxiang1996, 2020-03-26:
I want to use stm32 to set up a server, but I don't know how much ROM and ram I need, can you tell me

rudoq32, 2020-03-26:
Hi @zxiang1996,
I suggest you for the start to use STM32F769I-DISC1. It has 2MB FLASH memory. That would be sufficient to create a server with base address space.
I addition the STM32F769I-DISC1 has QSPI flash memory (I think 128MB) - that would be enough also for companion specifications e.g. Device, OpenPLC, etc. 
Unfortunately, so far I was not able to use this memory in "mapped mode" which could act as "one block" of memory.

MohabTamer, 2024-08-06:
Hi I want a simple code to test open62541 with freeRTOS + lwip on tm4c129x

PR #2511 (2019-04-04): WIP: Cleanup of freeRTOS and LwIP
I want to use this PR to fix all issues related to freeRTOS and LwIP: 
fixes #1859 
fixes #1864 
fixes #2227 
fixes #2460. 

Users working with freeRTOS that confirmed the PR works:

- [X] @embbo (https://github.com/open62541/open62541/pull/2511#issuecomment-475769413)
- [ ] @nidhal-mars (not active in the PR)
- [ ] @FishEatingCat (not active in the PR)
- [X] @VladBraz125 (https://github.com/open62541/open62541/pull/2511#issuecomment-479844345)
- [X] @rudoq32 (https://github.com/open62541/open62541/pull/2511#issuecomment-474811430)
- [ ] @sammer-h (not active  [... cut]