open62541/open62541 #1880: Revised Codestyle wiki page

Infinity95, 2018-06-27:
As mentioned, i revised the code style page. The new version can be found [here](https://github.com/open62541/open62541/wiki/Code-Style---NEW---draft).
IMHO we should enforce the style guide for all new contributions so we have a consistent code base.
I included a schema file for the IDE i use. Other users could add their own config files for their formatters if they use one.

Please have a look at the page and tell me if you feel something is missing and i will add it.

Pro, 2018-06-27:
Thanks!

Can we somehow enforce code style for PRs using Travis or other hooks?
We already had it using `make cppckeck` (see travis build script) but as far as I remember it didn't work well...

Infinity95, 2018-06-27:
I'll see if i find something useful.

jpfr, 2018-06-27:
:+1: For the Wiki!

Clang-Format can be jerry-rigged to a commit hook for new code changes.

https://github.com/barisione/clang-format-hooks

@Pro added a clang-format description a while back.
https://github.com/open62541/open62541/blob/master/.clang-format

I'm not sure how well that matches our current style and to what extend clang-format can be configured. There are however ways to visualize the formatting config.
https://zed0.co.uk/clang-format-configurator/

Pro, 2018-06-28:
There I also added the commit hook, see https://github.com/open62541/open62541/blob/master/tools/clang-format_precommit_hook

Anyways we should find a way to force the clang format for new commits, for the changed lines using CI.

HolgerJeromin, 2018-06-28:
Or we could add a pre build with a formatter

Pro, 2018-06-28:
@HolgerJeromin can you explain that a bit more in detail?

HolgerJeromin, 2018-06-28:
I know a team where any C++ code is automatically reformatted on each compile with a prebuild step (probably via visual studio prebuild event). 

This guarantees the same code style (an exception would be a developer who commits code without compile :-)

I can check tomorrow where and which formatter is called if this is interesting.

Infinity95, 2018-06-28:
This could be problematic since it would change all files to the correct format, which we can't do because of the already mentioned reasons. If the current codebase was already correctly formatted this would be a different matter.

HolgerJeromin, 2018-06-29:
@Infinity95 
All PRs would have to merge master (with the format) and rebuild. After that the diff should be clean again.

HolgerJeromin, 2018-06-29:
The team uses [ClangFormat](https://clang.llvm.org/docs/ClangFormat.html) in a VS prebuild event. 
For performance reasons a python script checks all files if they have been changed since the last run.

As a sidenote: VS 2017 includes [clangformat](https://blogs.msdn.microsoft.com/vcblog/2018/03/13/clangformat-support-in-visual-studio-2017-15-7-preview-1/).

Infinity95, 2018-06-29:
We could probably configure the clang-formatter call as part of the build with cmake.

jpfr, 2018-06-30:
We can add a build target `style` that calls `git clang-format --style file --diff --extensions c,h`.

This formats only the diff.
So we avoid formatting noise in the commits and branch divergence.

Pro, 2018-07-02:
There's a nice project (CC0) which automatically adds clang format as a git hook and also adds an additional cmake target:
https://github.com/kbenzie/git-cmake-format

That project uses the whole file, maybe we can fork it and change it to only reformat the diff, and make sure that the hook is only added on linux.

@Infinity95 do you have time/motivation to create a PR for that?

Infinity95, 2018-07-02:
Yeah i can do that in the later half of this month.

jpfr, 2021-07-06:
We moved the code-style discussion to the CONTRIBUTING.md.
Auto-formatting is not a focus right now.
Closing this.

closed 2021-07-06 by jpfr