open62541/open62541 #1886: How to access a node by name instead of its numeric ID?

dvucko, 2018-06-29:
I have an OPC UA server that exposes a variable named _pressure_, with a browse name `2:pressure` and a node ID shown as `ns=2;i=20`. I can successfully read its value from client by creating a `UA_NODEID_NUMERIC(2, 20)` and passing it to `UA_Client_readValueAttribute()`. However, I'm unable to read the value by providing a string-based node ID: using `UA_NODEID_STRING(2, "pressure")` doesn't work - no value is read.

What am I doing wrong? Is the possibility of using `UA_NODEID_STRING()` something that should be enabled on the server side, or am I just not providing the name correctly? Should I somehow account for the fact that the variable/node is nested within other nodes (_0:Root/0:Objects/2:PLC ..._)? Or perhaps I should use something other than `UA_NODEID_STRING()` to access the node by its name/browse path?

Thanks!

Stasik0, 2018-06-29:
You search for translateBrowsePathsToNodeIds service of opc ua.

Example of usage is here:
https://github.com/open62541/open62541/blob/befb0171d9c9ce987427a9dce095b5d7b7fd6bc6/examples/client_async.c

dvucko, 2018-06-29:
Thanks for the tip. I got it to work using `UA_Client_Service_translateBrowsePathsToNodeIds` as demonstrated in [corpus_generator.c](https://github.com/open62541/open62541/blob/47fa30d4cadaaee57ded95a01981508e7b3cb1f2/tests/fuzz/corpus_generator.c). Note that you need not specify `referenceTypeIds`. Getting node ID objects from the response is demonstrated [here](https://github.com/open62541/open62541/blob/3bbc1e7de60267997497cb7d87801a6b69a03c77/tests/server/check_services_view.c#L165).

closed 2018-06-29 by dvucko