open62541/open62541 #1902: Compliance Test Tool error

jhj0955, 2018-07-06:
open62541 Version (release number or git tag):
Git-Revision: 0.3-rc2-217-g0afd439a

It is similar to issues1875-error
I think our server needs a return code such as badcertificateUriInvalid
What needs to be corrected to fix this part?
I would really appreciate if you answer about this shortly.
Thank you.

project setting's open62541-ctt-master by https://github.com/open62541/open62541-ctt
(check the Nano Embedded Device Server Profile )

[image]

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag):
 - [ ] Other OPC UA SDKs used (client or server): 
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-07-08:
This might take a while.

The parse x509 certificate currently does not give us access to the URI field of the Subject Alternative Names.

There's an open bug for the mbedtls library we use.
https://github.com/ARMmbed/mbedtls/pull/731

If this doesn't move, we will have to add a second ASN.1 parser or switch security to a more complete lib. But openssl on embedded is a pain...

jpfr, 2018-07-09:
Fixed with a workaround in #1907.

closed 2018-07-09 by jpfr