open62541/open62541 #1909: How to handle an IP change during runtime?

mlang-de, 2018-07-09:
If during the server runtime only the IPv4 Settings change, the Server should not restart complete (nodestore should not be touched). 
So I call the follwing steps
1. UA_Server_run_shutdown()
2. get new IP
3. UA_ServerConfig_set_customHostname()
4. Create networklayer and endpoints
5. UA_Server_run_startup()
6. UA_Server_run_iterate()

This will work as long as a high number of session will be configured, because the active sessions will not be deleted. The sessions are alive for there lifetime. In a limited System with for instances 2 permitted sessions after the first IP Change only one Client can connect.
For now I do not found any public function to delete all available session. 

Could make some function like the follwing sense to implement?
`void UA_Server_SessionManager_deleteMembers(UA_Server* server) {
  UA_SessionManager_deleteMembers(&server->sessionManager);
  server->sessionManager.currentSessionCount = 0;
}`

Do I forget something?!

jpfr, 2018-07-09:
Do you want clients not to recover the old sessions over the new connections?
Then you are on the right track.

We could also add the session-cleanup into UA_Server_run_shutdown.

We currently are in similar discussions with regards to the lifetime of SecureChannels and Sessions if we have changes to the certificate and trust-lists coming in from the Global Discovery Server.

mlang-de, 2018-07-10:
> Do you want clients not to recover the old sessions over the new connections?

How could this be realized? How can I inform the Client about the changed IP?
Work this only with a Discovery Server in the network? If yes, what's about networks without Discovery Servers? I only know the TransferSubsription Service but this is not what we talk about here.

> We could also add the session-cleanup into UA_Server_run_shutdown.

This would ok for me at the moment, but if there is a possibilty to inform the Client we maybe need a public function to delete all sessions on request from the application side and in addition to handle both Scenarios?

ChHaa, 2018-07-11:
Imho there is no way to inform the Client about the IP change without some form of (local) discovery. After that you should be able to use ActivateSession on a newly created SecureChannel to resume the old session.

mlang-de, 2018-07-12:
@jpfr I tend to a public function to delete the sessions on request and not block the other use case. What's your opion?

closed 2021-02-03 by mlang-de