open62541/open62541 #1912: Asynchronous connect with inaccessible server

alajovic, 2018-07-10:
I am encountering an issue with the asynchronous client connect in cases where the server is not accessible at the time of the call to UA_Client_connect_async(), but becomes accessible a short while later. My aim is to keep attempting to connect until the server appears and the connection is established.

I have tried the following scenario:

1) call UA_Client_connect_async(...)
2) call UA_Client_run_iterate(client, 0) periodically
...
3) server becomes accessible
...
4) still calling UA_Client_run_iterate(client, 0) periodically
5) but the client never connects

The first few calls to `UA_Client_run_iterate()` returned `UA_STATUSCODE_GOOD`, then the status changed to `UA_STATUSCODE_BADCONNECTIONCLOSED` and I started seeing the output

```
error/network      No connection to server.
```

on the console. The client never connected.

Since this was not working as expected, I tried calling the pair

```
UA_Client_connect_async(...)
UA_Client_run_iterate(client, 0)
```

in a loop. This actually works if the server becomes accessible after several such calls; but if the loop with the pair of functions executes more than a few dozen times, the process eventually crashes with SIGSEGV in `UA_Connection_receiveChunksNonBlocking()`.

I can provide more data, including backtraces, valgrind reports, and possibly a minimal example, but I would first like to know whether I am missing something obvious.

Context:

open62541 0.4 development version, master from about two weeks ago
CentOS 7.2
Server that I'm trying to connect to is written in Java (Eclipse Milo)

jpfr, 2018-07-10:
> I can provide more data, including backtraces, valgrind reports, and possibly a minimal example, but I would first like to know whether I am missing something obvious.

No. A crash is a bug we want to solve.
The best is if we can reproduce the issue on our local boxes.

alajovic, 2018-07-11:
Minimal example that exhibits the problem:

CMakeLists.txt
```
cmake_minimum_required(VERSION 3.0)

include(ExternalProject)

ExternalProject_Add(
  external_open62541
  GIT_REPOSITORY "https://github.com/open62541/open62541.git"
  GIT_TAG "e602432001f57a1d1d10bbd495a750a84b646ea2"
  INSTALL_COMMAND ""
)

ExternalProject_Get_Property(external_open62541 binary_dir)
set(OPEN62541_LIBRARY open62541)
set(OPEN62541_LIBRARY_PATH ${binary_dir}/bin/${CMAKE_FIND_LIBRARY_PREFIXES}${OPEN62541_LIBRARY}.a)
add_library(${OPEN62541_LIBRARY} UNKNOWN IMPORTED)
set_property(TARGET ${OPEN62541_LIBRARY} PROPERTY IMPORTED_LOCATION ${OPEN62541_LIBRARY_PATH})
set_property(TARGET ${OPEN62541_LIBRARY} PROPERTY INTERFACE_INCLUDE_DIRECTORIES ${binary_dir})
add_dependencies(${OPEN62541_LIBRARY} external_open62541)

add_executable(async_problem async_problem.c)
set_property(TARGET async_problem PROPERTY C_STANDARD 99)
target_link_libraries(async_problem ${OPEN62541_LIBRARY})
```

async_problem.c
```
#include <stdio.h>
#include <unistd.h>
#include <open62541.h>

static void
onConnect (UA_Client *client, void *userdata, UA_UInt32 requestId, void *status) {
    printf ("Async connect returned with status code %s\n",
            UA_StatusCode_name (*(UA_StatusCode *) status));
}

int main() {
    UA_Client *client = UA_Client_new (UA_ClientConfig_default);
    const char* uri = "opc.tcp://localhost:4840";

    for (unsigned int i = 0; ; i++) {
        UA_StatusCode connectStatus = UA_Client_connect_async(client, uri, onConnect, NULL);
        printf("[%u] UA_Client_connect_async -> %s\n", i, UA_StatusCode_name(connectStatus));

        UA_StatusCode iterateStatus = UA_Client_run_iterate (client, 0);
        printf("[%u] UA_Client_run_iterate -> %s\n", i, UA_StatusCode_name(iterateStatus));

        usleep(100000);
    }

    return 0;
}

```

Place the files side by side into the same directory.
```
mkdir build
cd build
cmake ..
make
./async_problem
```

alajovic, 2018-07-11:
Valgrind output:
```
==6556== Memcheck, a memory error detector
==6556== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==6556== Using Valgrind-3.10.0 and LibVEX; rerun with -h for copyright info
==6556== Command: ./async_problem
==6556== 
[0] UA_Client_connect_async -> Good
[0] UA_Client_run_iterate -> Good
[1] UA_Client_connect_async -> Good
[1] UA_Client_run_iterate -> BadConnectionClosed
[2] UA_Client_connect_async -> Good
[2] UA_Client_run_iterate -> BadConnectionClosed
[3] UA_Client_connect_async -> Good
[3] UA_Client_run_iterate -> BadConnectionClosed
[4] UA_Client_connect_async -> Good
[4] UA_Client_run_iterate -> BadConnectionClosed
[5] UA_Client_connect_async -> Good
[5] UA_Client_run_iterate -> BadConnectionClosed
[6] UA_Client_connect_async -> Good
[6] UA_Client_run_iterate -> BadConnectionClosed
[7] UA_Client_connect_async -> Good
[7] UA_Client_run_iterate -> BadConnectionClosed
[8] UA_Client_connect_async -> Good
[8] UA_Client_run_iterate -> BadConnectionClosed
[9] UA_Client_connect_async -> Good
[9] UA_Client_run_iterate -> BadConnectionClosed
[10] UA_Client_connect_async -> Good
[10] UA_Client_run_iterate -> BadConnectionClosed
[11] UA_Client_connect_async -> Good
[11] UA_Client_run_iterate -> BadConnectionClosed
[12] UA_Client_connect_async -> Good
[12] UA_Client_run_iterate -> BadConnectionClosed
[13] UA_Client_connect_async -> Good
[13] UA_Client_run_iterate -> BadConnectionClosed
[14] UA_Client_connect_async -> Good
[14] UA_Client_run_iterate -> BadConnectionClosed
[15] UA_Client_connect_async -> Good
[15] UA_Client_run_iterate -> BadConnectionClosed
[16] UA_Client_connect_async -> Good
[16] UA_Client_run_iterate -> BadConnectionClosed
[17] UA_Client_connect_async -> Good
[17] UA_Client_run_iterate -> BadConnectionClosed
[18] UA_Client_connect_async -> Good
[18] UA_Client_run_iterate -> BadConnectionClosed
[19] UA_Client_connect_async -> Good
[19] UA_Client_run_iterate -> BadConnectionClosed
[20] UA_Client_connect_async -> Good
[20] UA_Client_run_iterate -> BadConnectionClosed
[21] UA_Client_connect_async -> Good
[21] UA_Client_run_iterate -> BadConnectionClosed
[22] UA_Client_connect_async -> Good
[22] UA_Client_run_iterate -> BadConnectionClosed
[23] UA_Client_connect_async -> Good
[23] UA_Client_run_iterate -> BadConnectionClosed
[24] UA_Client_connect_async -> Good
[24] UA_Client_run_iterate -> BadConnectionClosed
[25] UA_Client_connect_async -> Good
[25] UA_Client_run_iterate -> BadConnectionClosed
[26] UA_Client_connect_async -> Good
[26] UA_Client_run_iterate -> BadConnectionClosed
[27] UA_Client_connect_async -> Good
[27] UA_Client_run_iterate -> BadConnectionClosed
[28] UA_Client_connect_async -> Good
[28] UA_Client_run_iterate -> BadConnectionClosed
[29] UA_Client_connect_async -> Good
[29] UA_Client_run_iterate -> BadConnectionClosed
[30] UA_Client_connect_async -> Good
[30] UA_Client_run_iterate -> BadConnectionClosed
[31] UA_Client_connect_async -> Good
[... 80 lines cut ...]
==6556==    by 0x51F60EF: ???
==6556==  If you believe this happened as a result of a stack
==6556==  overflow in your program's main thread (unlikely but
==6556==  possible), you can try to increase the size of the
==6556==  main thread stack using the --main-stacksize= flag.
==6556==  The main thread stack size used in this run was 8388608.
==6556== 
==6556== HEAP SUMMARY:
==6556==     in use at exit: 15,880 bytes in 218 blocks
==6556==   total heap usage: 486 allocs, 268 frees, 3,402,317 bytes allocated
==6556== 
==6556== LEAK SUMMARY:
==6556==    definitely lost: 4,240 bytes in 53 blocks
==6556==    indirectly lost: 7,420 bytes in 106 blocks
==6556==      possibly lost: 0 bytes in 0 blocks
==6556==    still reachable: 4,220 bytes in 59 blocks
==6556==         suppressed: 0 bytes in 0 blocks
==6556== Rerun with --leak-check=full to see details of leaked memory
==6556== 
==6556== For counts of detected and suppressed errors, rerun with: -v
==6556== Use --track-origins=yes to see where uninitialised values come from
==6556== ERROR SUMMARY: 419 errors from 3 contexts (suppressed: 1 from 1)
```

The problem seems to be deterministic in the sense that it segfaults at step 53 every time.

commit c9c20881 (2018-09-24): fix crash for client connect async / fix #1912
closed 2018-09-24 by jpfr