open62541/open62541 #1914: Nodeset Compiler: Generate type defintion for custom datatypes instead of using binary encoding hack

Pro, 2018-07-11:
This is a replacement for #1609:
Currently the nodeset compiler is using binary-encoded ExtensionObjects to pass custom datatypes from XML to the server initialization. This should be replaced by apropriately generated the custom datatype code and then setting the variable correspondingly. The nodeset compiler should only use public API.

Comment by @jpfr:
The ideal solution would be to not manually encode values in the generated code at all.
We have all the facilities in place to do that now.

In the generated code:
1. Print the type for a new custom structure
2. Print the type description (with macros referring to the custom structure to get the padding right)
3. Add the custom type description to the server
4. Add the variable node with our standard API

crafti5, 2018-08-30:
is it possible to use nodeset compiler with customs enumeration? I'm stuck here with

`retVal |= UA_encodeBinary(`
 
And my Visual Studio says "error C3861: 'UA_encodeBinary': identifier not found". Can I do or change something here?

Pro, 2019-01-09:
Fixed in #2337

commit 0666b4e3 (2018-07-24): Export UA_encodeBinary to allow custom datatypes until we fix #1914
closed 2019-01-09 by Pro