open62541/open62541 #1925: Compilation ERROR with C++ compilator

FranckEtienne, 2018-07-17:
--------------------------------------------------------------------------------

Description
===========
I have a compilation error when I compile open62541-master with C++ compilator :+1: 
[ 33%] Building CXX object CMakeFiles/openLoki.dir/fabrique.cpp.o
/usr/bin/c++    -I/home/idevice/OpenLoki/../open62541-master   -o CMakeFiles/openLoki.dir/fabrique.cpp.o -c /home/idevice/OpenLoki/fabrique.cpp
In file included from /home/idevice/OpenLoki/fabrique.h:6:0,
                 from /home/idevice/OpenLoki/fabrique.cpp:9:
/home/idevice/OpenLoki/../open62541-master/open62541.h:21300:5: error: ‘UA_SOCKET’ does not name a type
     UA_SOCKET sockfd;                 /* Most connectivity solutions run on
     ^
/home/idevice/OpenLoki/../open62541-master/open62541.h:25644:17: error: ‘UA_SOCKET’ was not declared in this scope
 ssize_t UA_send(UA_SOCKET sockfd, const void *buf, size_t len, int flags); //equivalent to posix send implementation
                 ^

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): Client
 - [ ] Operating system: Linux Debian 8
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

FranckEtienne, 2018-07-17:
Thank’s for help me.

It’s ok with flag -DUA_ARCHITECTURE_POSIX.
But I work without problem with Master a week ago.
Which change in master is responsible of my error ?

jpfr, 2018-07-17:
Hello all,

can we assume an architecture by default?
In CMake as well as the amalgamated single-file distribution.

```c
#ifdef _WIN32
#define UA_ARCHITECTURE_WIN32
#else
#define UA_ARCHITECTURE_POSIX.
#endif
```

Otherwise, many existing build setups will suddenly break. That's not good.

jpfr, 2018-07-18:
>  But we should only define this if there are currently no other architectures defined

👍

Pro, 2018-07-19:
@cabralfortiss do you have a proposal how we could define either `UA_ARCHITECTURE_WIN32` or `UA_ARCHITECTURE_POSIX` if no other architecture is selected, using preprocessor in ua_config.h, not only CMake?
If someone is using the amalgamated version, he always has to define one of those two, at least.

ghazivakili, 2018-07-19:
Hey, Guys, I have a similar problem with compiling my code that uses the single-file library, and I have added also the same thing that you suggested but not working and the same error happened, 

Could you please let me know is related to what?
do I need include ua_config.h? and if yes where is it?
If you know about this problem I will be appriciate if you can help me,

Regards

//error !!!!!

In file included from /home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.c:28:0:
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:21349:5: error: unknown type name ‘UA_SOCKET’
     UA_SOCKET sockfd;                 /* Most connectivity solutions run on
     ^~~~~~~~~
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25632:1: error: unknown type name ‘ssize_t’; did you mean ‘size_t’?
 ssize_t UA_send(UA_SOCKET sockfd, const void *buf, size_t len, int flags); //equivalent to posix send implementation
 ^~~~~~~
 size_t
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25632:17: error: unknown type name ‘UA_SOCKET’; did you mean ‘UA_FORMAT’?
 ssize_t UA_send(UA_SOCKET sockfd, const void *buf, size_t len, int flags); //equivalent to posix send implementation
                 ^~~~~~~~~
                 UA_FORMAT
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25636:1: error: unknown type name ‘ssize_t’; did you mean ‘size_t’?
 ssize_t sendto(UA_SOCKET sockfd, const void *buf, size_t len, int flags, const struct sockaddr *dest_addr, socklen_t addrlen); //equivalent to posix sendto implementation
 ^~~~~~~
 size_t
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25636:16: error: unknown type name ‘UA_SOCKET’; did you mean ‘UA_FORMAT’?
 ssize_t sendto(UA_SOCKET sockfd, const void *buf, size_t len, int flags, const struct sockaddr *dest_addr, socklen_t addrlen); //equivalent to posix sendto implementation
                ^~~~~~~~~
                UA_FORMAT
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25636:108: error: unknown type name ‘socklen_t’; did you mean ‘__socklen_t’?
 ssize_t sendto(UA_SOCKET sockfd, const void *buf, size_t len, int flags, const struct sockaddr *dest_addr, socklen_t addrlen); //equivalent to posix sendto implementation
                                                                                                            ^~~~~~~~~
                                                                                                            __socklen_t
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25640:15: error: unknown type name ‘UA_SOCKET’; did you mean ‘UA_FORMAT’?
 int UA_select(UA_SOCKET nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout); //equivalent to posix select implementation
               ^~~~~~~~~
               UA_FORMAT
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25640:31: error: unknown type name ‘fd_set’; did you mean ‘false’?
 int UA_select(UA_SOCKET nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout); //equivalent to posix select implementation
                               ^~~~~~
                               false
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25640:48: error: unknown type name ‘fd_set’; did you mean ‘false’?
 int UA_select(UA_SOCKET nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout); //equivalent to posix select implementation
                                                ^~~~~~
                                                false
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25640:66: error: unknown type name ‘fd_set’; did you mean ‘false’?
 int UA_select(UA_SOCKET nfds, fd_set *readfds, fd_set *writefds, fd_set *exceptfds, struct timeval *timeout); //equivalent to posix select implementation
                                                                  ^~~~~~
                                                                  false
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25644:1: error: unknown type name ‘ssize_t’; did you mean ‘size_t’?
 ssize_t UA_recv(UA_SOCKET sockfd, void *buf, size_t len, int flags); //equivalent to posix recv implementation
 ^~~~~~~
 size_t
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25644:17: error: unknown type name ‘UA_SOCKET’; did you mean ‘UA_FORMAT’?
 ssize_t UA_recv(UA_SOCKET sockfd, void *buf, size_t len, int flags); //equivalent to posix recv implementation
                 ^~~~~~~~~
                 UA_FORMAT
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25648:1: error: unknown type name ‘ssize_t’; did you mean ‘size_t’?
 ssize_t recvfrom(UA_SOCKET sockfd, void *buf, size_t len, int flags, struct sockaddr *src_addr, socklen_t *addrlen);
 ^~~~~~~
 size_t
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25648:18: error: unknown type name ‘UA_SOCKET’; did you mean ‘UA_FORMAT’?
 ssize_t recvfrom(UA_SOCKET sockfd, void *buf, size_t len, int flags, struct sockaddr *src_addr, socklen_t *addrlen);
                  ^~~~~~~~~
                  UA_FORMAT
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.h:25648:97: error: unknown type name ‘socklen_t’; did you mean ‘__socklen_t’?
 ssize_t recvfrom(UA_SOCKET sockfd, void *buf, size_t len, int flags, struct sockaddr *src_addr, socklen_t *addrlen);
                                                                                                 ^~~~~~~~~
[... cut ...]
/home/mohamad/workspace/DIIG/OpcUADIIG-Server/open62541.c:48816:8: warning: implicit declaration of function ‘UA_socket_set_blocking’ [-Wimplicit-function-declaration]
     if(UA_socket_set_blocking(clientsockfd) != UA_STATUSCODE_GOOD) {
        ^~~~~~~~~~~~~~~~~~~~~~
CMakeFiles/OpcUAServer.dir/build.make:62: recipe for target 'CMakeFiles/OpcUAServer.dir/open62541.c.o' failed
make[2]: *** [CMakeFiles/OpcUAServer.dir/open62541.c.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/OpcUAServer.dir/all' failed
make[1]: *** [CMakeFiles/OpcUAServer.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

FranckEtienne, 2018-07-20:
With CMAKE use UNIX like :
if (UNIX)
    #do something
endif (UNIX)
if (MSVC)
    #do something
endif (MSVC)

Doc CMake (https://cmake.org/cmake/help/v3.6/variable/UNIX.html):
UNIX
True for UNIX and UNIX like operating systems.
Set to true when the target system is UNIX or UNIX like (i.e. APPLE and CYGWIN).

https://cmake.org/cmake/help/v3.0/variable/MSVC.html:
MSVC
True when using Microsoft Visual C
Set to true when the compiler is some version of Microsoft Visual C.

cabralfortiss, 2018-07-20:
I think the ua_config.h should be divided in two. The top main part where all the cmakedefine commands are (nothing to do with the architectures), and the bottom part, which is below the include of the architecture, where architecture-dependant code is found.

In the top part we could have:
```
#if ${UA_NOT_DEFINED_ARCHS}
# ifdef _WIN32
#  define UA_ARCHITECTURE_WIN32
# else
#  define UA_ARCHITECTURE_POSIX.
# endif
#endif
```

In the smain CMakeLists.txt file then, from all possible architectures, the string variable UA_NOT_DEFINED_ARCHS will be created:

```
GET_PROPERTY(architectures GLOBAL PROPERTY UA_ARCHITECTURES)
list(SORT architectures)
set(UA_NOT_DEFINED_ARCHS "1 ") #to make it easier to append later the && symbol
foreach(arch ${architectures})
   string(TOUPPER ${arch} UA_ARCHITECTURE_UPPER)
   set(UA_NOT_DEFINED_ARCHS ${UA_NOT_DEFINED_ARCHS}" && !defined(UA_ARCHITECTURE_${UA_ARCHITECTURE_UPPER})")
endforeach(arch ${architectures})
```

Pro, 2018-07-20:
@FranckEtienne we already have something like this, but that does not solve the issue here, because if you do not use CMake, then it will not be set. See https://github.com/open62541/open62541/blob/master/CMakeLists.txt#L73

@cabralfortiss good idea, I will try to create a PR for that.

ghazivakili, 2018-07-23:
Guys Still I have the problem I tried what you told me, I guess those solution is for compiling the library but I don't have this problem, I can not compile my code, I wrote my code with version 2 and now after updating the library I have problem and I don't know what is UA_SOCKET ?????
I can not compile my code anymore!!!!!

Could you please let me know what I have to do ????

Pro, 2018-07-23:
@ghazivakili 
> I guess those solution is for compiling the library but I don't have this problem,

Guessing it not knowing :smile_cat: Did you already read through the issue and tried to solve it using the provided info? For the time being you need to compile your code with the additional architecture define as it is already mentioned above (https://github.com/open62541/open62541/issues/1925#issuecomment-405590231)! You can also help by solving the issue on your own and submit Pull Requests.

You should also know that the master branch is NOT a release branch. So it will definitely happen in the future that your build will be broken or not compatible, if you are using `master`. So if you are using master, you should be able to accept these issues until they are fixed. If you want a stable API you have to use the releases (current release is 0.3-rc2).

closed 2018-07-25 by Pro

Pro, 2018-07-25:
@FranckEtienne @ghazivakili can you test if current master is working for you. It should not be necessary to set the architecture if you are on Windows/Linux.

FranckEtienne, 2018-07-25:
Hi,

I compile open62541 but I have a problem (already reported) : #1931.
I can't test.

Due to instability of current I come back to 0.3-RC2.

commit d54f4c67 (2018-07-24): Fixes #1925 and clean examples
commit 1605455a (2018-07-25): Fixes #1925 and clean examples