open62541/open62541 #1928: Memory leak detected after using UA_Server_ReadValue()

SebastianGraeber, 2018-07-18:
Description
===========
Hi,

I'm developing an OPC UA server using VS 2017 Pro. The follwoing code is used:

// +++START+++
UA_ServerConfig l_UaServerConfig = UA_ServerConfig_standard;
UA_ServerNetworkLayer nl = UA_ServerNetworkLayerTCP(UA_ConnectionConfig_standard, l_dlgEdasOpcUaServerGuiDlg->m_iOpcUaPort);
l_UaServerConfig.networkLayers = &nl;
l_UaServerConfig.networkLayersSize = 1;
UA_Server *l_UaServer = UA_Server_new(l_UaServerConfig);

std::vector<UA_VariableAttributes> l_OpcWriteVarAttr(l_WriteValueCnt);
std::vector<UA_Float> l_OpcWriteValue(l_WriteValueCnt);
std::vector<UA_QualifiedName> l_OpcWriteName(l_WriteValueCnt);
std::vector<UA_NodeId> l_OpcWriteNodeValue(l_WriteValueCnt);
std::vector<UA_NodeId> l_OpcWriteNodeParent(l_WriteValueCnt);
std::vector<UA_NodeId> l_OpcWriteNodeParentRef(l_WriteValueCnt);

#pragma omp parallel for
for (int i = 0; i < 100; i++)
{
	UA_VariableAttributes_init(&l_OpcWriteVarAttr[i]);
	l_OpcWriteValue[i] = -99999.0;
	l_OpcWriteVarAttr[i].valueRank = -1;
	UA_Variant_setScalar(&l_OpcWriteVarAttr[i].value, &l_OpcWriteValue[i], &UA_TYPES[UA_TYPES_FLOAT]);
	l_OpcWriteVarAttr[i].description = UA_LOCALIZEDTEXT("de_de", l_WriteNameList.GetAt(i).GetBuffer());
	l_OpcWriteVarAttr[i].displayName = UA_LOCALIZEDTEXT("de_de", l_WriteNameList.GetAt(i).GetBuffer());
	l_OpcWriteVarAttr[i].accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
	l_OpcWriteVarAttr[i].dataType = UA_TYPES[UA_TYPES_FLOAT].typeId;
	l_OpcWriteName[i] = UA_QUALIFIEDNAME(1, l_WriteNameList.GetAt(i).GetBuffer());
	l_OpcWriteNodeValue[i] = UA_NODEID_STRING(1, l_WriteNameList.GetAt(i).GetBuffer());
	l_OpcWriteNodeParent[i] = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER);
	l_OpcWriteNodeParentRef[i] = UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES);
	UA_Server_addVariableNode(l_UaServer, l_OpcWriteNodeValue[i], l_OpcWriteNodeParent[i], l_OpcWriteNodeParentRef[i], l_OpcWriteName[i], UA_NODEID_NULL, l_OpcWriteVarAttr[i], NULL, NULL);
}
UA_Server_run_startup(l_UaServer);

while (true)
{
	//	Execute OPC server loop
	UA_Server_run_iterate(l_UaServer, false);

	//	Put data to OPC interface => OPC to EDAS
#pragma omp parallel for
	for (int i = 0; i < 100; i++)
	{
		UA_Server_readValue(l_UaServer, l_OpcWriteNodeValue[i], &l_OpcWriteVarAttr[i].value);
		l_EdasWriteValues[i] = *(UA_Float*)l_OpcWriteVarAttr[i].value.data;
	}
        // Clocking
        Sleep(100);
}
UA_Server_run_shutdown(l_UaServer);
UA_Server_delete(l_UaServer);
l_UaServer = NULL;

nl.deleteMembers(&nl);
// ---END---

The OPC UA server works as expected (transfer of values in both directions), but while executing the server, the memory increases. VS 2017 detects memory leaks after closing the application.
It seems that the following function produces the increasing memory that is never released:
VS Call Stack:

EdasOpcUaServerGui.exe!UA_Array_copy() - Line 5748	C++

What might cause this issue?

UA_Server_WriteValue() works as expected without a memory increase.

Background Information / Reproduction Steps
===========================================
I'm using Version 0.2, but the same behavoiour occurred with 0.3rc1 and 03.rc2. No other OPC UA SDKs are used.
OS: Windows 10 x64 Pro

Checklist
=========
Please provide the following information:

 - [x] open62541 Version (release number or git tag):
 - [x] Other OPC UA SDKs used (client or server): 
 - [x] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-07-19:
` UA_Server_readValue(l_UaServer, l_OpcWriteNodeValue[i], &l_OpcWriteVarAttr[i].value);`

This line returns a copy of the value in the information model.
You need to clean it up eventually with UA_Variant_deleteMembers or similar.

SebastianGraeber, 2018-07-19:
Thanks, that fixed my issue.

closed 2018-07-19 by SebastianGraeber