open62541/open62541 #1931: Error on linking

FranckEtienne, 2018-07-19:
Description
===========
I have a linking error with the last master : 
idevice@itopopcua:~/myProject$ g++ myProject.c cis.o -I. libopen62541.a libmbedtls.a libmbedx509.a libmbedcrypto.a -o myProject
/tmp/cc306c6O.o: dans la fonction « Open62541::createServer() »:
myProject.c:(.text._ZN9Open6254112createServerEv[_ZN9Open6254112createServerEv]+0x21): référence indéfinie vers « UA_ServerConfig_new_minimal »
collect2: error: ld returned 1 exit status

UA_ServerConfig_new_minimal is present in open62541.h.

On libopen62541.a, I only see:
idevice@itopopcua:~/myProject$ ar -t libopen62541.a
open62541.c.o

I have no problem with Master's branch 2 weeks ago.

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): server
 - [ ] Operating system: linux Debian 8
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
  int createServer()
  {
    printf( "createServer() =>\n" );
    UA_ServerConfig *config = UA_ServerConfig_new_default();
    m_server = UA_Server_new( config );

 - [ ] Critical issue

FranckEtienne, 2018-07-20:
I get the last Master: git clone https://github.com/open62541/open62541.git

And I have  the linking issue:
/tmp/ccDH3Udq.o: dans la fonction « UA_ServerConfig_new_default »:
myProject.c:(.text+0x233): référence indéfinie vers « UA_ServerConfig_new_minimal »
collect2: error: ld returned 1 exit status

jpfr, 2018-07-20:
Can you try with gcc instead of g++?
Maybe we have an `extern C` wrapper missing somewhere.

FranckEtienne, 2018-07-23:
My main is C++.
I try to compile my C++ main with gcc but I have errors:
/tmp/ccuSEXYc.o: dans la fonction « OPCUA::createOpcua(std::string) »:
openGaia.cpp:(.text+0x2d1): référence indéfinie vers « std::string::c_str() const »
openGaia.cpp:(.text+0x302): référence indéfinie vers « operator new(unsigned long) »
...

Pro, 2018-07-25:
From where did you get the `libopen62541.a` file?
Did you compile it yourself? If yes, which settings/command did you use for that?

FranckEtienne, 2018-07-25:
Hi, thank's for help me.

I use a libopen62541.a which I obtain :+1: 
git clone https://github.com/open62541/open62541.git open62541-current
cd open62541-current
cmake -DUA_ENABLE_AMALGAMATION=ON -DUA_ENABLE_ENCRYPTION=ON CMakeLists.txt
make
=>
idevice@itopopcua:~/openGaia/open62541/open62541-current$ ll bin
total 1404
-rw-r--r-- 1 idevice idevice 1435042 juil. 25 09:40 libopen62541.a

FranckEtienne, 2018-07-25:
It's working for me now.
I forget to copy the new open62541.h where I build the executable.

Sorry for the inconvenience.

closed 2018-07-25 by FranckEtienne