open62541/open62541 #1935: Question about ua_client_highlevel.c open62541

Boody91, 2018-07-20:
Description
===========
Hi everyone, first of all: great job on this project.

I tried to apply the solution to the problem raised in:

https://stackoverflow.com/questions/47389892/how-to-read-a-array-value-from-the-opcua-server-by-client?rq=1

and I was reviewing the code in:

https://github.com/open62541/open62541/blob/master/src/client/ua_client_highlevel.c

but I still don't succeed in extracting a nodeID from the matrix

Background Information

===========================================

I'm using the sample server.c  and this is the code that I tried to compile:

`#include <stdio.h>
#include "open62541.h"

int browser_node(UA_Client* client) {

    printf("Browsing nodes in objects folder:\n");
    UA_BrowseRequest bReq;
    UA_BrowseRequest_init(&bReq);
    bReq.requestedMaxReferencesPerNode = 0;
    bReq.nodesToBrowse = UA_BrowseDescription_new();
    bReq.nodesToBrowseSize = 1;
    bReq.nodesToBrowse[0].nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER); 
    bReq.nodesToBrowse[0].resultMask = UA_BROWSERESULTMASK_ALL; 

    UA_BrowseResponse bResp = UA_Client_Service_browse(client, bReq);

    printf("%-9s %-16s %-16s %-16s\n", "NAMESPACE", "NODEID", "BROWSE NAME", "DISPLAY NAME");
    for (size_t i = 0; i < bResp.resultsSize; ++i) {
        for (size_t j = 0; j < bResp.results[i].referencesSize; ++j) {
            UA_ReferenceDescription *ref = &(bResp.results[i].references[j]);
            if (ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_NUMERIC) {
                printf("%-9d %-16d %-16.*s %-16.*s\n", ref->nodeId.nodeId.namespaceIndex,
                    ref->nodeId.nodeId.identifier.numeric, (int)ref->browseName.name.length,
                    ref->browseName.name.data, (int)ref->displayName.text.length,
                    ref->displayName.text.data);
            }
            else if (ref->nodeId.nodeId.identifierType == UA_NODEIDTYPE_STRING) {          

                printf("%-9d %-16.*s %-16.*s %-16.*s\n", ref->nodeId.nodeId.namespaceIndex,
                    (int)ref->nodeId.nodeId.identifier.string.length,
                    ref->nodeId.nodeId.identifier.string.data,
                    (int)ref->browseName.name.length, ref->browseName.name.data,
                    (int)ref->displayName.text.length, ref->displayName.text.data);
            }
           
        }
    }
    UA_BrowseRequest_deleteMembers(&bReq);
    UA_BrowseResponse_deleteMembers(&bResp);

    printf("\n");

    return 0;

}

int browser_nodevalue(UA_Client* client) {
 
    printf("\n");
   
    UA_ReadRequest request;
    UA_ReadRequest_init(&request);
    UA_ReadValueId id;
    UA_ReadValueId_init(&id);
    id.attributeId = UA_ATTRIBUTEID_VALUE;
    id.nodeId = UA_NODEID_NUMERIC(0, UA_NS0ID_SERVER_NAMESPACEARRAY);
    request.nodesToRead = &id;
    request.nodesToReadSize = 1;

   
 
            UA_ReadResponse response = UA_Client_Service_read(client, request);

            UA_StatusCode retval = UA_STATUSCODE_GOOD;
            if (response.responseHeader.serviceResult != UA_STATUSCODE_GOOD)
                retval = response.responseHeader.serviceResult;
            else if (response.resultsSize != 1 || !response.results[0].hasValue)
                retval = UA_STATUSCODE_BADNODEATTRIBUTESINVALID;
            else if (response.results[0].value.type != &UA_TYPES[UA_TYPES_STRING])
                retval = UA_STATUSCODE_BADTYPEMISMATCH;

            if (retval != UA_STATUSCODE_GOOD) {
                UA_ReadResponse_deleteMembers(&response);
                return retval;
            }    

            retval = UA_STATUSCODE_BADNOTFOUND;
            UA_String *ns = (UA_String *)response.results[0].value.data;
            for (size_t i = 0; i < response.results[0].value.arrayLength; ++i) {
                printf("Valor es %*.s\n", (int)ns[i].length, ns[i].data);

            }
      
   
    printf("\n");
    return retval;

   
}

int main(int argc, char *argv[])
{
    
    UA_Client *client = UA_Client_new(UA_ClientConfig_default);
    UA_StatusCode status = UA_Client_connect(client, "opc.tcp://169.254.81.206:4840/");
    if (status != UA_STATUSCODE_GOOD) {
        UA_Client_delete(client);
        return status;
    }
    
    browser_node(client);
    browser_nodevalue(client);
      

    UA_Client_delete(client); 
    return status;
}`

Any suggestion would be very useful, since I don't have enough experience in C

Thanks for your time.

Checklist
=========
Please provide the following information:

 - [x ] open62541 Version (release number or git tag): version 4.0
 - [x ] Other OPC UA SDKs used (client or server): cient.c
 - [ ] Operating system:
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

jpfr, 2018-07-21:
> but I still don't succeed in extracting a nodeID from the matrix

What do you mean by that?
In the example you posted, you try to print an array of strings. Isn't that what you want to do?

By the way, if browser_nodevalue is successful, you do not clean up the response and always return UA_STATUSCODE_BADNOTFOUND.

jpfr, 2018-08-09:
We still don’t know what this matrix is.
Please reopen with the required clarification.

closed 2018-08-09 by jpfr