open62541/open62541 #1936: securityPolicyEndPoint can not disable None/None connection

7486913, 2018-07-21:
Description
===========
When I use the following function, it works very well but UaExpert still can connect to my OPC Server Successfully without any certificate. 

In UaExpert, set  **Security Policy**  as None/**Message Security Mode** as None, it will pass authority 
and connect to OPC Server.

 UA_ServerConfig* config = UA_ServerConfig_new_basic256sha256(serverPort,&certificate,&privateKey,
                                      trustlist,trustlistSize,
                                      revocationList,revocationListSize);

I note there are two times connection in logs, so  change codes in ua_securechannel.c.
UA_StatusCode
UA_SecureChannel_init(UA_SecureChannel *channel,
                      const UA_SecurityPolicy *securityPolicy,
                      const UA_ByteString *remoteCertificate) {
    if(channel == NULL || securityPolicy == NULL || remoteCertificate == NULL)
        return UA_STATUSCODE_BADINTERNALERROR;

    /* Linked lists are also initialized by zeroing out */
    memset(channel, 0, sizeof(UA_SecureChannel));
    channel->state = UA_SECURECHANNELSTATE_FRESH;
    channel->securityPolicy = securityPolicy;

    UA_StatusCode retval;
    if(channel->securityPolicy->certificateVerification != NULL) {
        retval = channel->securityPolicy->certificateVerification->
            verifyCertificate(channel->securityPolicy->certificateVerification->context, remoteCertificate);

        if(retval != UA_STATUSCODE_GOOD)
            return retval;
    } else {
        UA_LOG_WARNING(channel->securityPolicy->logger, UA_LOGCATEGORY_SECURITYPOLICY, "No PKI plugin set. "
            "Accepting all certificates");
		
    }
	//first connect
	static int time = 0;
	if (time == 0) {
		time = 1;
	}
	else{
        //second connect if remoteCertificate is still null, close connection.
		time = 0;
		if (remoteCertificate->length == 0)
			return UA_STATUSCODE_BADINTERNALERROR;
	}
	

    retval = securityPolicy->channelModule.
        newContext(securityPolicy, remoteCertificate, &channel->channelContext);
    if(retval != UA_STATUSCODE_GOOD)
        return retval;

    retval = UA_ByteString_copy(remoteCertificate, &channel->remoteCertificate);
    if(retval != UA_STATUSCODE_GOOD)
        return retval;

    UA_ByteString remoteCertificateThumbprint = {20, channel->remoteCertificateThumbprint};
    retval = securityPolicy->asymmetricModule.
        makeCertificateThumbprint(securityPolicy, &channel->remoteCertificate,
                                  &remoteCertificateThumbprint);

    return retval;
}

Now it's works. But I think it not best way to settle this, so somebody have better idea?

Background Information / Reproduction Steps
===========================================
UaExpert v1.4.4  is download from here:
https://www.unified-automation.com/downloads/opc-ua-clients.html

Checklist
=========
Please provide the following information:

 - [X] open62541 Version : 0.3.0 rc
 - [X] Operating system: Windows7 bit32
 - [X] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
[2018-07-21 12:37:02.167 (UTC-0700)] info/channel	Creating a new SecureChannel
[2018-07-21 12:37:02.167 (UTC-0700)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2018-07-21 12:37:02.167 (UTC-0700)] info/channel	Connection 844 | SecureChannel 2 | Opened SecureChannel
[2018-07-21 12:37:02.168 (UTC-0700)] info/channel	Connection 844 | SecureChannel 2 | CloseSecureChannel
[2018-07-21 12:37:02.169 (UTC-0700)] info/network	Connection 844 | Closed
[2018-07-21 12:37:04.915 (UTC-0700)] info/network	Connection 844 | New connection over TCP from 127.0.0.1
[2018-07-21 12:37:04.916 (UTC-0700)] info/channel	Creating a new SecureChannel
[2018-07-21 12:37:04.916 (UTC-0700)] warn/securitypolicy	No PKI plugin set. Accepting all certificates
[2018-07-21 12:37:04.916 (UTC-0700)] info/channel	Connection 844 | SecureChannel 3 | Opened SecureChannel
[2018-07-21 12:37:04.917 (UTC-0700)] info/session	Connection 844 | SecureChannel 3 | Session 72dbfd91-ebc2-c469-63c6-1c810520f29f | ActivateSession: Session activated
[2018-07-21 12:37:04.928 (UTC-0700)] info/session	Connection 844 | SecureChannel 3 | Session 72dbfd91-ebc2-c469-63c6-1c810520f29f | Could not create MonitoredItem with status code BadMonitoredItemFilterInvalid
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Infinity95, 2018-07-23:
The function you are using will always create a none endpoint. If you do not wish to create a None endpoint you should create your own version of the function or manually set the config.
(If modifying you should be able to just delete createSecurityPolicyNoneEndpoint(...))

7486913, 2018-07-24:
I tried your way. But it dose not work. Because SecurityPolicySha256Endpoint could not build a connection if I delete createSecurityPolicyNoneEndpoint(...).

Infinity95, 2018-07-24:
Could you be a bit more specific what errors you encountered? It might be because UAExpert first tries to connect with None in order to get the endpoints or something.

7486913, 2018-07-24:
Yes, you are right. UaExpert first tries to connect with None. If None is not work, the UaExpert will stop trying other endpoints. Even UaExpert connect with None, it still need twice connection. 
So I change the code to allow the first connection always.

jonasgreen88, 2018-12-06:
I have also been thinking about this issue lately. Some citations from the specifications to discuss:
OPC UA Part 2 (1.04), chapter 4.7 Security Profiles: "Security Profiles describe a Profile "None” that is used for testing, but if any other more secure Profiles are available this Profile is disabled by default. Profile “None” provides no security."
OPC UA Part 4 (1.04), chapter 5.4.4 GetEndpoints: "This Service shall not require message security but it may require transport layer security."

With these two statements in mind I think it should be possible to have no endpoint supporting SecurityPolicy None but it shall still be possible to use SecurityPolicy None to send the GetEndpoints request. This is not supported by open62541 for now.

This issue has also been discussed in the node-opcua project (has Wireshark logs present to visualize the issue): https://github.com/node-opcua/node-opcua/issues/382.

What do you guys think, do I interpret the specifications properly and is this something open62541 should support?

jonasgreen88, 2018-12-12:
Some more statements from chapter 5.4 Discovery Service Set:

> "Every Server shall have a DiscoveryEndpoint that Clients can access without establishing a Session. This Endpoint may or may not be the same Session Endpoint that Clients use to establish a SecureChannel. Clients read the security information necessary to establish a SecureChannel by calling the GetEndpoints Service on the DiscoveryEndpoint."

> "DiscoveryEndpoints shall not require any message security, but it may require transport layer
> security. In production systems, Administrators may disable discovery for security reasons and
> Clients shall rely on cached EndpointDescriptions. To provide support for systems with disabled
> Discovery Services Clients shall allow Administrators to manually update the EndpointDescriptions
> used to connect to a Server. Servers shall allow Administrators to disable the DiscoveryEndpoint."

I belive that we should somehow support to have no normal endpoint with SecurityPolicy None, but still allow clients to connect to the Discovery endpoint without security to perform the requests GetEndpoints, FindServers and FindServersOnNetwork.

jpfr, 2018-12-12:
> I belive that we should somehow support to have no normal endpoint with SecurityPolicy None, but still allow clients to connect to the Discovery endpoint without security to perform the requests GetEndpoints, FindServers and FindServersOnNetwork.

This can be done already.
The three services you mention only require a SecureChannel but no session.https://github.com/open62541/open62541/blob/master/src/server/ua_server_binary.c#L85

The user opening a session over an unencrypted channel can be denied indepdently.

7486913, 2018-12-13:
I had found one switch to disable this.
ua_config_default.c line:250
/* Access Control. Anonymous Login only. */
conf->accessControl = UA_AccessControl_defaul(false, usernamePasswordsSize, usernamePasswords);

The first boolean parameter is used to disable the None user connect the OPC Server without password.

jonasgreen88, 2018-12-13:
> This can be done already.
The three services you mention only require a SecureChannel but no session.
https://github.com/open62541/open62541/blob/master/src/server/ua_server_binary.c#L85
>The user opening a session over an unencrypted channel can be denied indepdently.

I can't find any way to deny a user to not create a session over a none secured channel. But I found now that the endpoint information is forwarded to the activate session and there it would be possible to deny access. I'm still not sure if this is the correct behavior. The specification doesn't say explicitly that it should be possible to create a session on a none secured channel, as it does for the disovery services mentioned earlier. The specification also says that the SecurityPolicy None should be disabled by default. I treat this as there shouldn't be any endpoint available with that security policy. Still it should be possible to perform the three discovery services.

jpfr, 2018-12-13:
> But I found now that the endpoint information is forwarded to the activate session and there it would be possible to deny access. 

Exactly. We could also deny before that by matching against the endpoints before calling the ActivateSession callback in the access control plugin,

> The specification also says that the SecurityPolicy None should be disabled by default. I treat this as there shouldn't be any endpoint available with that security policy. Still it should be possible to perform the three discovery services.

That is also my understanding.
The commercial tools all connect with #None first to discover the current endpoints.
Then they reconnect with encryption.

jonasgreen88, 2018-12-13:
> Exactly. We could also deny before that by matching against the endpoints before calling the ActivateSession callback in the access control plugin,

How do you mean I can do this? I tried to remove the endpoint with the SecurityPolicy None from the server config. In that case I was not able to browse the endpoints from UaExpert as it couldn't connect with SecurityPolicy None at all.

jpfr, 2018-12-13:
> How do you mean I can do this?

Currently the SecurityPolicies are bound to an endpoint.
With your change in #2247, we decouple the policies from the endpoints.
Than we can allow securechannels #none even if no endpoint can be selected for it.

But that requires changes on the code level...

jonasgreen88, 2018-12-14:
> Currently the SecurityPolicies are bound to an endpoint.
> With your change in #2247, we decouple the policies from the endpoints.
> Than we can allow securechannels #none even if no endpoint can be selected for it.
> 
> But that requires changes on the code level...

Yes, this is what I wanted to come to. With PR #2247, if connecting with a security policy without a matching endpoint some discovery services could be supported, but no other services.

jonasgreen88, 2018-12-18:
> Currently the SecurityPolicies are bound to an endpoint.
With your change in #2247, we decouple the policies from the endpoints.
Than we can allow securechannels #none even if no endpoint can be selected for it.

I posted a couple of questions about this topic on the OPC forum as well. The response confirmed what you posted I think:
https://opcfoundation.org/forum/opc-ua-standard/server-with-securitypolicy-none-disabled-by-default/#p1689

NoelGraf, 2025-03-05:
is fixed and can be closed.

closed 2025-03-05 by NoelGraf