open62541/open62541 #1937: 10 requests at startup [client] [subscription]

fabian727, 2018-07-23:
Hi guys,

I wrote a simple client, which subscribes to a variable. If I trace the packets. The first thing the client is requesting this variable 10 times, but I cannot find a reason why or the line of code, which generates it. Is it a problem of mine or is it wanted by the source. If so, why?

This is my Code:

```    UA_CreateSubscriptionRequest request;
	UA_CreateSubscriptionResponse response;
	UA_MonitoredItemCreateRequest monRequest;
	UA_MonitoredItemCreateResult monResponse;

    request = UA_CreateSubscriptionRequest_default();
    request.requestedPublishingInterval = 50.0; //[ms]
	response = UA_Client_Subscriptions_create(client, request, NULL, NULL, NULL);

	monRequest  = UA_MonitoredItemCreateRequest_default(variableId);
	monRequest.monitoringMode = UA_MONITORINGMODE_REPORTING;
	//UA_MONITORINGMODE_SAMPLING;
	monRequest.requestedParameters.samplingInterval = 50; //[ms]
	monRequest.monitoringMode = UA_MONITORINGMODE_REPORTING;
	monRequest.requestedParameters.queueSize = 1;

	monResponse = UA_Client_MonitoredItems_createDataChange(client, response.subscriptionId, UA_TIMESTAMPSTORETURN_BOTH, monRequest, NULL, positionChangeHandler, NULL);
```

StalderT, 2018-07-23:
The behavior is normal. By default a "background task" sends 10 PublishRequest to the server.

https://github.com/open62541/open62541/blob/f5ed77c902820e70ca9d81ac9831c5c9942e8d8a/plugins/ua_config_default.c#L692

You can change this by setting outStandingPublishRequests

```
    UA_ClientConfig config = UA_ClientConfig_default;
    config.outStandingPublishRequests = 1;
    UA_Client *client = UA_Client_new(config);
```

StalderT, 2018-07-23:
> Why does it send 10 Requests? Is it in the specification?

see

https://github.com/open62541/open62541/pull/1497#issuecomment-356228132

and

https://github.com/open62541/open62541/pull/1497#issuecomment-356024764

fabian727, 2018-07-24:
Thanks for your fast support

closed 2018-07-24 by fabian727