open62541/open62541 #1938: It isn't possible to set more than 10mn for secureChannelLifeTime

FranckEtienne, 2018-07-23:
Description
===========
There is :
 const UA_ClientConfig UA_ClientConfig_default = {
    5000, /* .timeout, 5 seconds */
    10 * 60 * 1000, /* .secureChannelLifeTime, 10 minutes */
    UA_Log_Stdout, /* .logger */
    { /* .localConnectionConfig */
        0, /* .protocolVersion */
        65535, /* .sendBufferSize, 64k per chunk */
        65535, /* .recvBufferSize, 64k per chunk */
        0, /* .maxMessageSize, 0 -> unlimited */
        0 /* .maxChunkCount, 0 -> unlimited */
    },

But if I do:
  UA_ClientConfig uaClientConfig = UA_ClientConfig_default;
    uaClientConfig.secureChannelLifeTime = UINT32_MAX; // OR other value > 10 * 60 * 1000 (10mn)

the renew meesage is always send every 10mn.

It is possible to set less !

Background Information / Reproduction Steps
===========================================

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): Master
 - [ ] Other OPC UA SDKs used (client or server): client
 - [ ] Operating system: Debian 8
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

FranckEtienne, 2018-07-23:
Is it due to this code:
    /* Set the lifetime. Lifetime 0 -> set the maximum possible */
    channel->securityToken.revisedLifetime =
        (request->requestedLifetime > cm->server->config.maxSecurityTokenLifetime) ?
        cm->server->config.maxSecurityTokenLifetime : request->requestedLifetime;
    if(channel->securityToken.revisedLifetime == 0)
        channel->securityToken.revisedLifetime = cm->server->config.maxSecurityTokenLifetime;

FranckEtienne, 2018-07-23:
It is the client which initiate the renew.

I don't understand why I have to change the server configuration (I can't modify it (on the OPCUA server)).

The renew in code :
  42415 static UA_StatusCode
  42416 openSecureChannel(UA_Client *client, UA_Boolean renew) {
...
  42443     opnSecRq.clientNonce = client->channel.localNonce;
  42444     opnSecRq.requestedLifetime = client->config.secureChannelLifeTime;

  29745     /* Set the lifetime. Lifetime 0 -> set the maximum possible */
  29746     channel->securityToken.revisedLifetime =
  29747         (request->requestedLifetime > cm->server->config.maxSecurityTokenLifetime) ?
  29748         cm->server->config.maxSecurityTokenLifetime : request->requestedLifetime;
  29749     if(channel->securityToken.revisedLifetime == 0)
  29750         channel->securityToken.revisedLifetime = cm->server->config.maxSecurityTokenLifetime;

I change the code (it is not confortable and clear to modify server code) :
  47061 static UA_ServerConfig *
  47062 createDefaultConfig(void) {
...
  47127     conf->maxSecurityTokenLifetime = 120 * 60 * 1000; /* 30 minutes */

And there are no impact, the renew message is sent every 7500s.

Infinity95, 2018-07-23:
Like i said, the Server!! only allows a lifetime up to a maximum. That is why in the response the revised lifetime is sent, which the client should use to send the renew. If not, the Channel will just time out, since it is not renewed in time.

If you for example request a lifetime of 10 years, but the server only allows a maximum lifetime of 10 minutes, it will respond with a revised lifetime of 10 minutes, which is what needs to be used to send the renew.

So if you changed the maximum time in the server config you can now increase the time on the client side.

FranckEtienne, 2018-07-23:
Ok, I understand my mistake : I use an open62541 server where the maxSecurityTokenLifetime is set also to 10mn.

It is possible to see the value of revisedLifetime sent by the server in response of openSecureChannel ?

I can't see where to find the response in UA_Client_connect interface :
     status = UA_Client_connect( m_client , endpointUrl );

If I display uaClientConfig.secureChannelLifeTime (which is greater than on the server) after the call to UA_Client_connect, it doesn't contain the revisedLifeTime.

FranckEtienne, 2018-07-25:
It is possible to see the value of revisedLifetime sent by the server in response of openSecureChannel ?

jpfr, 2018-07-25:
It is part of the response and stored in the client securechannel.
You can get to the securechannel settings via internal API.

FranckEtienne, 2018-07-25:
Not easy to found.

Here is the struct of     UA_SecureChannel channel define in UA_Client:
struct UA_SecureChannel {
    UA_SecureChannelState   state;
    UA_MessageSecurityMode  securityMode;
    UA_ChannelSecurityToken securityToken; /* the channelId is contained in the securityToken */
    UA_ChannelSecurityToken nextSecurityToken;

    /* The endpoint and context of the channel */
    const UA_SecurityPolicy *securityPolicy;
    void *channelContext; /* For interaction with the security policy */
    UA_Connection *connection;

    /* Asymmetric encryption info */
    UA_ByteString remoteCertificate;
    UA_Byte remoteCertificateThumbprint[20]; /* The thumbprint of the remote certificate */

    /* Symmetric encryption info */
    UA_ByteString remoteNonce;
    UA_ByteString localNonce;

    UA_UInt32 receiveSequenceNumber;
    UA_UInt32 sendSequenceNumber;

    LIST_HEAD(session_pointerlist, UA_SessionHeader) sessions;
    LIST_HEAD(chunk_pointerlist, MessageEntry) chunks;
};

I don't see what I search.

What is the internal API ??

Infinity95, 2018-07-25:
If you have access to the securechannel struct, you can just read the current securityToken.
It is defined like this:
```
typedef struct {
    UA_UInt32 channelId;
    UA_UInt32 tokenId;
    UA_DateTime createdAt;
    UA_UInt32 revisedLifetime;
} UA_ChannelSecurityToken;
```
The revisedLifetime attribute should contain the data you are looking for.

FranckEtienne, 2018-07-25:
But How to have access to secureChannel ?

In open62541.h, we have :
struct UA_Client;
typedef struct UA_Client UA_Client;

The UA_Client structure is define in open62541.c :
struct UA_Client {
    /* State */
    UA_ClientState state;
...
    UA_SecureChannel channel;

And I have an compiling error message when I try to print from UA_Client pointer :
    LOG_WARN( "connect(): revisedLifetime = %1" , m_client->channel.revisedLifetime );

opcuaBackend.cpp:269:59: error: invalid use of incomplete type ‘UA_Client {aka struct UA_Client}’
     LOG_WARN( "connect(): revisedLifetime = %1" , m_client->channel.revisedLifetime );

jpfr, 2018-08-09:
As said before, use cmake to build the project instead of the single file release.
Then you can include internal api.
But that can change quite often.