open62541/open62541 #1941: Implementing Certificate Authenticate for the client

jhj0955, 2018-07-25:
hi

I would like to implement user certificates on the client side out of the three authentication methods.
(anonymous, username/password and user certificate authentication)

So I modified the ua_client_connect.c and headerfile.

<pre>
activateSession(UA_Client *client) {
    UA_ActivateSessionRequest request;
    UA_ActivateSessionRequest_init(&request);
    request.requestHeader.requestHandle = ++client->requestHandle;
    request.requestHeader.timestamp = UA_DateTime_now();
    request.requestHeader.timeoutHint = 600000;

    //manual ExtensionObject encoding of the identityToken
    if(client->authenticationMethod == UA_CLIENTAUTHENTICATION_NONE) {
        UA_AnonymousIdentityToken* identityToken = UA_AnonymousIdentityToken_new();
        UA_AnonymousIdentityToken_init(identityToken);
        UA_String_copy(&client->token.policyId, &identityToken->policyId);
        request.userIdentityToken.encoding = UA_EXTENSIONOBJECT_DECODED;
        request.userIdentityToken.content.decoded.type = &UA_TYPES[UA_TYPES_ANONYMOUSIDENTITYTOKEN];
        request.userIdentityToken.content.decoded.data = identityToken;
    } else {
        UA_UserNameIdentityToken* identityToken = UA_UserNameIdentityToken_new();
        UA_UserNameIdentityToken_init(identityToken);
        UA_String_copy(&client->token.policyId, &identityToken->policyId);
        UA_String_copy(&client->username, &identityToken->userName);
        UA_String_copy(&client->password, &identityToken->password);
        request.userIdentityToken.encoding = UA_EXTENSIONOBJECT_DECODED;
        request.userIdentityToken.content.decoded.type = &UA_TYPES[UA_TYPES_USERNAMEIDENTITYTOKEN];
        request.userIdentityToken.content.decoded.data = identityToken;
    }
</pre>

First, I want to change token and client structure, So I  found UA_UserNameIdentityToken_new() , but I don't know structure of this function
what is the architectural difference between UserNameIdentityToken and X509IdentityToken ?
Thanks, I hope your answer.

ChHaa, 2018-07-25:
One is for authentication with username/password and the other for authentication with X509 certificates, see Part 4, 7.36.x

jpfr, 2018-08-08:
Ongoing discussion in #1966

closed 2018-08-08 by jpfr