open62541/open62541 #1942: client can't reconnect to a server when a server restart

FranckEtienne, 2018-07-25:
Description
===========
A client is connect to a server.
This server is killed and restarted.
The open62541 client can't reconnect to the server.

Background Information / Reproduction Steps
===========================================
Every 100 ms, I execute :
    UA_StatusCode status = UA_Client_runAsync( m_client , ( UA_Int16 ) 50 );
    if ( status != UA_STATUSCODE_GOOD )
    {
      LOG_ERROR( "runIterate(): fail, reason is  %1 ( '%2' )" , status , UA_StatusCode_name( status ) );
      LOG_INFO( "runIterate(): Try to reconnect" );
      reconnect();
    }

UA_StatusCode OPCUABackend::reconnect( void )
{
    UA_StatusCode status = UA_STATUSCODE_GOOD;
    LOG_WARN( "=> reconnect()" );

    // Reconnect with the same client
    LOG_WARN( "reconnect(): to %1" , m_endpointUrl );
    status = UA_Client_connect( m_client , m_endpointUrl );
    if ( status != UA_STATUSCODE_GOOD )
    {
      LOG_ERROR( "reconnect(): connect fail %1 = '%2'" , status , UA_StatusCode_name( status ) );
    }
    return status;
}

Traces on client 
[2018-07-25 09:05:44.703 (UTC+0200)] warn/network	Server url is invalid: opc.tcp://172.24.139.24:
[2018-07-25 09:05:45.702 (UTC+0200)] warn/network	Server url is invalid: opc.tcp://172.24.139.24:

2018-07-25 09:05:44,702 WARN  [0x000055ce05b76d60] TSVC::OPCUABackend - => runIterate( timeout=[500] )
2018-07-25 09:05:44,703 INFO  [0x000055ce05b76d60] TSVC::OPCUABackend - runIterate(): clientState = [3]
2018-07-25 09:05:44,703 ERROR [0x000055ce05b76d60] TSVC::OPCUABackend - runIterate(): fail, reason is  -2136080384 ( 'BadConnectionClosed' )
2018-07-25 09:05:44,703 INFO  [0x000055ce05b76d60] TSVC::OPCUABackend - runIterate(): Try to reconnect
2018-07-25 09:05:44,703 WARN  [0x000055ce05b76d60] TSVC::OPCUABackend - => reconnect()
2018-07-25 09:05:44,703 WARN  [0x000055ce05b76d60] TSVC::OPCUABackend - reconnect(): to opc.tcp://172.24.139.24:
2018-07-25 09:05:44,703 ERROR [0x000055ce05b76d60] TSVC::OPCUABackend - reconnect(): connect fail -2136080384 = 'BadConnectionClosed'
2018-07-25 09:05:44,703 WARN  [0x000055ce05b76d60] TSVC::OPCUABackend - <= reconnect( ret=[-2136080384=BadConnectionClosed] )
2018-07-25 09:05:44,703 WARN  [0x000055ce05b76d60] TSVC::OPCUABackend - <= runIterate( -2136080384 = 'BadConnectionClosed' )

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): 0.3-rc2
 - [ ] Other OPC UA SDKs used (client or server): Client
 - [ ] Operating system: Debian 8
 - [ ] Logs (with `UA_LOGLEVEL` set as low as necessary) attached
 - [ ] Wireshark network dump attached
 - [ ] Self-contained code example attached
 - [ ] Critical issue

Pro, 2018-07-25:
@FranckEtienne can you please use the corresponding code formatting if you post code. Either use the button in the editor or have a look at https://guides.github.com/features/mastering-markdown/

Also, did you already check the example here:
https://github.com/open62541/open62541/blob/master/examples/client_connect_loop.c

I just ran it on 0.3-rc2 and it successfully reconnects

FranckEtienne, 2018-07-25:
I compile and launch the example : It works fine on my machine.

In my case, I want to do subscription and the client could be reconnect to the server (if this last go down and up), it is the reason why I used :
`UA_StatusCode status = UA_Client_runAsync( m_client , ( UA_Int16 ) 50 );
if ( status != UA_STATUSCODE_GOOD )
{
LOG_ERROR( "runIterate(): fail, reason is %1 ( '%2' )" , status , UA_StatusCode_name( status ) );
LOG_INFO( "runIterate(): Try to reconnect" );
reconnect();
}`

I try to replace UA_Client_runASync by UA_Client_connect, but I doesn't receive the data !

What is the best pratice to do subcrption and to be able to reconnect to a server ?

StalderT, 2018-07-25:
If I understand correctly your question, the best way is to use ```stateCallback ```

You can check the example here :

https://github.com/open62541/open62541/blob/0.3/examples/client_subscription_loop.c

jpfr, 2018-07-27:
I renamed the title to "Client **does not** reconnect to a server when a server restart".
There is no one solution. There should be options to configure the agent behaviour.

- don't reconnect
- reconnect with a new securechannel and recover the old session (default)
- reconnect with a new securechannel, try torecover the session, if that does not work, create a new session and recreate the Subscriptions

@FranckEtienne You seem to require this feature.
Do you have capacity to work on an API proposal?

FranckEtienne, 2018-07-30:
Hi,

I find the problem, my ServerUrl was truncated and so the client can't reconnect to the server.

I close this issue.

closed 2018-07-30 by FranckEtienne

jpfr, 2018-07-30:
Was this a problem in our implementation that we should fix?

FranckEtienne, 2018-07-30:
No it is in my code.
Sorry for the inconvenience.