open62541/open62541 #1946: Compile Error due [help wanted]

fabian727, 2018-07-30:
Hi guys,
I get compiler errors. I can not state which effect creates to start this error. All I can state is:

1. clone repo
2. init submodules
3. create dirs for building
4. initialisation for building: cmake ..; ccmake . (enable FULL_Namespace,Encryption)
5. make -j4
6. Everything works fine. Some time goes by
7. ccmake . (for new configuration, like Encryption or Events)
8. make clean && make -j4
9. see following logging-output:

```
...
Integrating file '/home/user/open62541/arch/ua_architecture_functions.h'...done.
The size of /home/user/open62541/debug_x86/open62541.h is 1676345 Bytes.
[ 14%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_nodestore_default.c.o
[ 18%] Built target open62541-amalgamation-header
[ 18%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_config_default.c.o
[ 20%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_securitypolicy_none.c.o
[ 22%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_securitypolicy_basic128rsa15.c.o
Scanning dependencies of target open62541-object
[ 22%] Building C object CMakeFiles/open62541-plugins.dir/plugins/ua_securitypolicy_basic256sha256.c.o
[ 24%] Building C object CMakeFiles/open62541-plugins.dir/arch/ua_network_tcp.c.o
[ 26%] Building C object CMakeFiles/open62541-object.dir/src/ua_types.c.o
[ 28%] Building C object CMakeFiles/open62541-object.dir/src/ua_types_encoding_binary.c.o
[ 36%] Built target open62541-plugins
[ 36%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_types_generated.c.o
[ 38%] Building C object CMakeFiles/open62541-object.dir/src_generated/ua_transport_generated.c.o
[ 38%] Building C object CMakeFiles/open62541-object.dir/src/ua_util.c.o
[ 40%] Building C object CMakeFiles/open62541-object.dir/src/ua_connection.c.o
[ 40%] Building C object CMakeFiles/open62541-object.dir/src/ua_securechannel.c.o
[ 42%] Building C object CMakeFiles/open62541-object.dir/src/server/ua_session.c.o
/homeuser/open62541/src/ua_util.c:34:1: error: no previous prototype for ‘UA_parseEndpointUrl’ [-Werror=missing-prototypes]
 UA_parseEndpointUrl(const UA_String *endpointUrl, UA_String *outHostname,
 ^~~~~~~~~~~~~~~~~~~
/homeuser/open62541/src/ua_util.c:107:1: error: no previous prototype for ‘UA_ByteString_toBase64String’ [-Werror=missing-prototypes]
 UA_ByteString_toBase64String(const UA_ByteString *byteString, UA_String *str) {
 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/user/open62541/src/ua_util.c:128:1: error: no previous prototype for ‘UA_NodeId_toString’ [-Werror=missing-prototypes]
 UA_NodeId_toString(const UA_NodeId *nodeId, UA_String *nodeIdStr) {
 ^~~~~~~~~~~~~~~~~~
/home/user/open62541/src/ua_util.c: In function ‘UA_NodeId_toString’:
/home/user/open62541/src/ua_util.c:180:75: error: expected ‘)’ before ‘UA_PRINTF_GUID_FORMAT’
             UA_snprintf((char*)nodeIdStr->data, nodeIdStr->length, "%sg=" UA_PRINTF_GUID_FORMAT,
                                                                           ^~~~~~~~~~~~~~~~~~~~~
/home/user/open62541/src/ua_util.c:180:70: error: format ‘%s’ expects a matching ‘char *’ argument [-Werror=format=]
             UA_snprintf((char*)nodeIdStr->data, nodeIdStr->length, "%sg=" UA_PRINTF_GUID_FORMAT,
                                                                      ^
/home/user/open62541/src/ua_util.c: At top level:
cc1: error: unrecognized command line option ‘-Wno-static-in-inline’ [-Werror]
cc1: all warnings being treated as errors
CMakeFiles/open62541-object.dir/build.make:240: recipe for target 'CMakeFiles/open62541-object.dir/src/ua_util.c.o' failed
make[2]: *** [CMakeFiles/open62541-object.dir/src/ua_util.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:369: recipe for target 'CMakeFiles/open62541-object.dir/all' failed
make[1]: *** [CMakeFiles/open62541-object.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```
If I create a new subdirectory with new configuration of cmake etc the error keeps existing.
If I download the repository new and create the same changes. There are no errors, everything as it should be.

I use the up to date Version of open62541 in the master branch. I develop on Debian9, which I think does not affect the problem.

jpfr, 2018-07-30:
Strange. There definitely is a prototype defined.
The methods `UA_ByteString_toBase64String` and `UA_parseEndpointUrl` are defined here:
https://github.com/open62541/open62541/blob/master/include/ua_util.h#L65

Your PR #1948 only adds cosmetic changes.
That does not change much.

fabian727, 2018-07-31:
i know its only cosmetic. Just thought if I see it I can change it.
At the moment I am at the wrong computer, but I am very sure, that I have a second header in src/ua_util.h. i declared there the functions, than this part of the problem was solved. The snprintf still remained. I had to comment it out.

fabian727, 2018-07-31:
I just removed this file. I don't know where it came from. Now it works again

closed 2018-07-31 by fabian727