open62541/open62541 #1958: how can I make my own single file?

Hootan-Man, 2018-08-02:
Description
===========
I'm trying to port open62541 into an new platform. here is [my fork](https://github.com/Hootan-Man/open62541/tree/master/arch)

Background Information / Reproduction Steps
===========================================
I followed all the given instruction, and all the needed functionality(malloc, alloc, sockets, etc) are substituted.
now I want to compile it.
it is an embedded platform and has it's own compiler. (not a strange one)
I tried to include all the header files and .c   files.
the problem is that some of header files are resolved at compile time. 
for example "ua_config.h"
Is there any way that I can make my own single file distribution including my new architecture ?
I need to compile and test it, any advice is appreciated.

Checklist
=========
Please provide the following information:

 - [ ] open62541 Version (release number or git tag): [open62541](https://github.com/open62541/open62541)
 - [ ] Other OPC UA SDKs used (client or server): both
 - [ ] Operating system: arch linux

jpfr, 2018-08-02:
You can use CMake to build your project.
Just set UA_ENABLE_AMALGAMATION and it will generate a single-file distribution.

Just edit the /include/ua_config.h.in so that the generated result is to your liking.

Pro, 2018-08-03:
It should be enough if you set `UA_ENABLE_AMALGAMATION=ON` and then include your custom architecture in the amalgamation with `UA_ARCHITECTURE=zerynth` which should add the architecture to the list here `UA_AMALGAMATION_ARCHITECTURES`. Hope this helps?

At the end you have two files `open62541.h` and `open62541.c` which you can use to compile with the rest of your code. Also have a look at my example project for an arduino chip: https://github.com/Pro/open62541-arduino/blob/master/CMakeLists.txt#L51

Hootan-Man, 2018-08-03:
thanks, guys. it worked perfectly.
just it produces the single file after executing the `make` command.
what should I do to only have the single files, and don't compile it to binary? (is there any flag)

since it misses a bunch of sources files needed for the architecture and I end up having a long list of errors.
It saves a lot of time if I stop it at the end of the pre-processor stage, and then use my own compiler.

Pro, 2018-08-04:
Can you try to use `make open62541-amalgamation-source` and `make open62541-amalgamation-header`?

We are already setting these targets in Cmake which should only execute the amalgamation. Please let me know if it works.

Hootan-Man, 2018-08-06:
thanks, Pro.
I confirm that the mentioned commands work. :+1:

Hootan-Man, 2018-08-06:
I noticed something.
I have defined some custom function that my target platform misses.
for example:
[gch_calloc](https://github.com/Hootan-Man/open62541/blob/6d8fb799cbefd1edfe4bbd81e632044b52f9bb96/arch/zerynth/ua_architecture_function.c#L35) defined in the same file.
[mini_snprintf](https://github.com/Hootan-Man/open62541/blob/6d8fb799cbefd1edfe4bbd81e632044b52f9bb96/arch/zerynth/mini_snprintf.c#L140) defined in a separate file.

no matter if I run `make` or `make open62541-amalgamation-source` and `make open62541-amalgamation-header`

despite that `UA_ARCHITECTURE = zerynth ` and `UA_ENABLE_AMALGAMATION=ON `
the generated single file doesn't contain my functions !!!

Pro, 2018-08-06:
You need to add your custom architecture files to the array here:
https://github.com/Hootan-Man/open62541/blob/6d8fb799cbefd1edfe4bbd81e632044b52f9bb96/arch/zerynth/CMakeLists.txt#L12

And probably you need to clean, because it may be that cmake does not correctly see your file changes.

jpfr, 2018-08-08:
@pro
Can we get rid of the sprintf dependency?
As we see, not every platforms comes with an implementation...

Pro, 2018-08-13:
> Can we get rid of the sprintf dependency?

Not so easily, since it is used in many places. Maybe we can add a similar file as @Hootan-Man did in his fork to a common architecture folder which can be reused for architectures which do not support sprintf (of course only if the License is equal or more open than MPL2.0). For now I do not have a good solution for this.

jpfr, 2018-09-30:
This seems to be solved for @Hootan-Man .
Please reopen if issues remain.

closed 2018-09-30 by jpfr