open62541/open62541 #1962: BadEncodingLimitsExceeded while running server on wce7

mark1357945, 2018-08-07:
Description
===========
I have find solution for opcua on wce7 about a month. Finally I found  [ ISSUE-884](https://github.com/open62541/open62541/issues/884) successfully build & run server on wce7, and connected by opcua organization client from my win32 system.

while the opcua library is out of date now, some service wasn't implement then.
so I update open62541.c/open62541.h, change sampleServer code and fix some compile error.

the server still can run on wce7,
but if I try to connect, it would jump out an exception after activate session:
**ServiceResultException : Unexpected error processing response.
BadEncodingLimitsExceeded : MaxArrayLength 65535 < 1852994816**

Server Log: 
[ServerLog.txt](https://github.com/open62541/open62541/files/2265016/ServerLog.txt)

any infomation is appreciated :)

Background Information / Reproduction Steps
===========================================
open62541 library Version: rls v0.3-rc2  
IDE: Visual Studio 2008
original project: https://github.com/open62541/open62541/files/603652/opcuaserver.zip made by @frax2222
my project: 
[opcuaserver_rls0.3-rc2.zip](https://github.com/open62541/open62541/files/2265019/opcuaserver_rls0.3-rc2.zip)
server binary is in path"\opcuaserver\WEC7 (ARMv4I)\Debug"

jpfr, 2018-08-09:
Diagnostic info is not implemented so far.
So this looks like a memory corruption bug.

 An you try with the current master and/or run the unit tests on wce7?
I don’t have access to the platform. That makes things difficult.

crafti5, 2018-08-10:
I've got the same error under WEC7 so i tried to use files from master and after some WEC7 customisations I was able to build successful the server. The communication isn't broken anymore.

Under WEC7 there shall some changes with logging and also for the file open part in "process_RegisterServer"

mark1357945, 2018-08-10:
@jpfr 
I've try to build open62541.c/.h from master, but there's some mistake so that i can't build out my open62541.c/.h files, even i have set UA_ENABLE_AMALGAMATION, that's why I choose release v0.3 as replacement.
Next week I'll do it again and log my build step. Try to figure out why. thanks!

@crafti5 
feels good to hear that! so it seems to be a fixed bug between release v0.3 and master.
I'll works on building library from master next week, thank you.
and more appreciate if you can share the wec7-compatible open62541 library!

crafti5, 2018-08-13:
@mark1357945 sure i can upload the changed files, but at this time i have only changed the amalgated file. So i don't know if this help. For a pull request i must prepare the files and this can take some time, i think until end of the week.

Pro, 2018-08-13:
@mark1357945 and @crafti5 you can use the new architecture structure to easily add your own architecture (only in master), i.e. add a separate folder for WEC7 and modify the methods.
https://github.com/open62541/open62541/tree/master/arch

Theoretically it should only be necessary to change files within the `arch` folder, but we still need to evaluate/test this. It would be nice if we can officially add support for WEC7, thus maybe you can try to create a Pull Request with the new architecture and required changes.

KamjooBayat, 2019-02-02:
@mark1357945  Thanks . I am trying to compile . But a lot of syntax error is there  . I fixed one series , again another series appear . can u send me please your open62541.h and open62541.c file .
I used cmake and based on documentation of open62541 I make open62541.h and open62541.c  files . but many  syntax error is there .it is make me tired in last few days .

KamjooBayat, 2019-02-03:
finally fixed . I did following sequence : 
1 - get latest version from  master 
2 - make win32 target with cmake for native win32 compiler . 
3 - used generated open62541.c and open62541.h files in  VS2008 with ARM SDK .
4 - did very simple modifications which all are in \arch\wce7  folder in master 
5 - compile or Wince 5 . 

It is working fine now :)

teccheck, 2025-04-15:
Unfortunately, we don't support wce7 anymore. As there's been no recent activity, I'll close this issue

closed 2025-04-15 by teccheck