open62541/open62541 #1963: Data Source Write

marioricardojr, 2018-08-07:
Hi,
I am trying to develop a server that receives a data from a client, such as UA Expert. This server runs on a raspberry pi and this received value will be send to an i2c module. I know that the value written in UA Expert is received by the server because the operation returns GOOD. But I'm finding it difficult to convert the value written in UA_Expert to an Int_16 value. The code developed so far is below. Any help is welcome.

`#include <stdio.h>
#include <stdlib.h>
#include <time.h>
#include <stdbool.h>
#include <signal.h>
#include <sys/types.h> // open
#include <sys/stat.h>  // open
#include <fcntl.h>     // open
#include <unistd.h>    // read/write usleep
#include <inttypes.h>  // uint8_t, etc
#include <linux/i2c-dev.h> // I2C bus definitions
#include <open62541.h>

int fd;
// Note PCF8591 defaults to 0x48!
int ads_address = 0x48;
int16_t val;

uint8_t writeBuf[3];
uint8_t readBuf[2];

float myfloat;

const float VPS = 4.096 / 32768.0; //volts per step

/*
The resolution of the ADC in single ended 
mode we have 15 bit rather than 16 bit resolution, 
the 16th bit being the sign of the differential reading.
*/

static UA_StatusCode readValvePositionerTK1(UA_Server *server, const UA_NodeId *sessionId, void *sessionContext,
                                         const UA_NodeId *nodeId, void *nodeContext, UA_Boolean sourceTimeStamp,
                                         const UA_NumericRange *range, UA_DataValue *dataValue) {

    UA_Int16 level = 7631;//only test
    UA_Variant_setScalarCopy(&dataValue->value, &level,
                             &UA_TYPES[UA_TYPES_UINT16]);
    dataValue->hasValue = true;
    return UA_STATUSCODE_GOOD;
}

static UA_StatusCode writeValvePositionerTK1(UA_Server *server, const UA_NodeId *sessionId, void *sessionContext,
                                      const UA_NodeId *nodeId, void *nodeContext, const UA_NumericRange *range,
                                      const UA_DataValue *data) {
    
	UA_Int16 PositionValve;

    UA_Variant_setScalarCopy(&PositionValve, &data,&UA_TYPES[UA_Variant]);
    data->hasValue = true;
    return UA_STATUSCODE_GOOD;
    //UA_Variant_setScalarCopy(&PositionValve,&data->value,&UA_TYPES[UA_TYPES_UINT16]);
    //return UA_STATUSCODE_GOOD;
    //UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "Manually changing sensor value NOT implemented!");
    //return UA_STATUSCODE_BADINTERNALERROR;
}

static void addValvePositionerTK1DataSourceVariable(UA_Server *server) {

UA_NodeId ValvePositionerId; /* get the nodeid assigned by the server */
    UA_ObjectAttributes oAttr = UA_ObjectAttributes_default;
    oAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Posicionador Inteligente Valvula Tanque 01");
    UA_Server_addObjectNode(server, UA_NODEID_NULL,
                            UA_NODEID_NUMERIC(0, UA_NS0ID_OBJECTSFOLDER),
                            UA_NODEID_NUMERIC(0, UA_NS0ID_ORGANIZES),
                            UA_QUALIFIEDNAME(1, "Posicionador Inteligente Valvula Tanque 01"), UA_NODEID_NUMERIC(0, UA_NS0ID_BASEOBJECTTYPE),
                            oAttr, NULL, &ValvePositionerId);

    UA_VariableAttributes mnvAttr = UA_VariableAttributes_default;
    UA_String manufacturerNameValve = UA_STRING("smar");
    UA_Variant_setScalar(&mnvAttr.value, &manufacturerNameValve, &UA_TYPES[UA_TYPES_STRING]);
    mnvAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Fabricante");
    UA_Server_addVariableNode(server, UA_NODEID_NULL, ValvePositionerId,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "Fabricante"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), mnvAttr, NULL, NULL);

    UA_VariableAttributes modelvAttr = UA_VariableAttributes_default;
    UA_String modelNameValve = UA_STRING("FY301");
    UA_Variant_setScalar(&modelvAttr.value, &modelNameValve, &UA_TYPES[UA_TYPES_STRING]);
    modelvAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Modelo");
    UA_Server_addVariableNode(server, UA_NODEID_NULL, ValvePositionerId,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "Modelo"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), modelvAttr, NULL, NULL);

    UA_VariableAttributes tagAttr = UA_VariableAttributes_default;
    UA_String tag = UA_STRING("FIT-31");
    UA_Variant_setScalar(&tagAttr.value, &tag, &UA_TYPES[UA_TYPES_STRING]);
    tagAttr.displayName = UA_LOCALIZEDTEXT("en-US", "TAG");
    UA_Server_addVariableNode(server, UA_NODEID_NULL, ValvePositionerId,
                              UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT),
                              UA_QUALIFIEDNAME(1, "TAG"),
                              UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), tagAttr, NULL, NULL);
							  
							  
							  
	UA_NodeId currentNodeId = UA_NODEID_STRING(1, "valor");
    UA_QualifiedName currentName = UA_QUALIFIEDNAME(1, "valor");
	
	UA_DataSource PositionValveDataSource;
	UA_VariableAttributes vAttr = UA_VariableAttributes_default;
vAttr.accessLevel = UA_ACCESSLEVELMASK_READ | UA_ACCESSLEVELMASK_WRITE;
    PositionValveDataSource.read = readValvePositionerTK1;
    PositionValveDataSource.write = writeValvePositionerTK1;
	UA_Variant_setScalar(&vAttr.value, &PositionValveDataSource, &UA_TYPES[UA_TYPES_STRING]);
	vAttr.displayName = UA_LOCALIZEDTEXT("en-US", "Valor");
    UA_Server_addDataSourceVariableNode(server, currentNodeId, ValvePositionerId,
                                        UA_NODEID_NUMERIC(0, UA_NS0ID_HASCOMPONENT), currentName,
                                        UA_NODEID_NUMERIC(0, UA_NS0ID_BASEDATAVARIABLETYPE), vAttr,
                                        PositionValveDataSource, NULL, NULL);
}
							  
	
							  
							  

							  
							  
							  
	 
	
						  
	

UA_Boolean running = true;
static void stopHandler(int sig) {
    UA_LOG_INFO(UA_Log_Stdout, UA_LOGCATEGORY_USERLAND, "received ctrl-c");
    running = false;
}

int main() {
    signal(SIGINT, stopHandler);
    signal(SIGTERM, stopHandler);

    UA_ServerConfig *config = UA_ServerConfig_new_default();
    UA_Server *server = UA_Server_new(config);

    addValvePositionerTK1DataSourceVariable(server);

    UA_StatusCode retval = UA_Server_run(server, &running);
    UA_Server_delete(server);
    UA_ServerConfig_delete(config);

    return (int) retval;
}`

schroeder-, 2018-08-07:
Can you change the formating of your source code, it's hard to read. See: https://guides.github.com/features/mastering-markdown/

schroeder-, 2018-08-07:
Change your writeValvePositionerTK1:

```
static UA_StatusCode writeValvePositionerTK1(UA_Server *server, const UA_NodeId *sessionId, void *sessionContext,
                                             const UA_NodeId *nodeId, void *nodeContext, const UA_NumericRange *range,
                                             const UA_DataValue *data) {
    if (UA_Variant_hasScalarType(&data->value, &UA_TYPES[UA_TYPES_INT16])) {
                UA_Int16 val = *(UA_Int16 *)data->value.data;
                // Work with the value

                return UA_STATUSCODE_GOOD;
    }
    return UA_STATUSCODE_BADTYPEMISMATCH;
}
```

Also your initialization seems to set the wrong data type. You set the Variable to be a UA_String. Instead Init it as INT16:

```
UA_INT16 default = xxx; //your default value
UA_Variant_setScalar(&vAttr.value, &default, &UA_TYPES[UA_TYPES_INT16]);
```

marioricardojr, 2018-08-07:
Thanks for helping.
I am make the changes, but returns this error in UA Expert:
`14:38:17.422 | DA Plugin          | open62541-based OPC UA Appl... | Write to node 'NS1|String|valor' failed [ret = BadTypeMismatch]`

code:
[teste.txt](https://github.com/open62541/open62541/files/2267461/teste.txt)

Thanks.

schroeder-, 2018-08-08:
You can check the type of the variable like:
`printf(data->value.type->typeName); `

jpfr, 2018-11-18:
Please reopen if the issue remains.

closed 2018-11-18 by jpfr